Koenkk/zigbee2mqtt #2673: trying to add Ledvance Smart+ bulb white classic E27 dimmable

Thor2503, 2020-01-04:
Good evening,
First of all I would like to say that I like very much the Zigbee2MQTT "tool" and say thank you for all the work that has flown into the project. I came across z2m a couple of months ago when starting to look into home automation and trying things out with OpenHAB. So far all the devices which I wanted to add were found directly after pairing but today I am trying to add a new light bulb to my setup which was not found. It is a "Ledvance Smart+ Classic E27 Dimmable". This should be very similar if not identical to the bulb with corresponding characteristics from Osram, Lightyfy or Sylvania, so I first updated my system to version 1.8.0 and then started following the instructions to add new devices and included the following code in the devices.js:
'''
// Ledvance
    {
        zigbeeModel: ['A60 DIM Z3'],
        model: 'AC10786-DIM',
        vendor: 'Ledvance',
        description: 'Ledvance Smart+ bulb white Classic E27 8.5 W 2700K dimmable',
        extend: generic.light_onoff_brightness,
    },
'''
Via MQTT.fx I am able to switch the bulb on and off as well as to set the brightness but I do not get any information about the linkquality.
So my first question is what I might be missing to get the linkquality information and 
Second, as the Osram bulbs might have some additional functionality how could I check if this would be available for the Ledvance bulb as well and in the case it is how to implement this.
Hoping that somebody here can point me in the right direction.

Koenkk, 2020-01-07:
- Linkquality is only available when the bulb sends message, most of the time not the case unless reporting is enabled (https://www.zigbee2mqtt.io/information/report.html) or when doing a network scan

- try with the configuration below and try executing the commands from https://www.zigbee2mqtt.io/devices/AC10787.html#set-default-power-onoff-transition, if the behaviour of the bulb changes it is supported.
```js
{
    zigbeeModel: ['A60 DIM Z3'],
    model: 'AC10786-DIM',
    vendor: 'Ledvance',
    description: 'Ledvance Smart+ bulb white Classic E27 8.5 W 2700K dimmable',
    supports: generic.light_onoff_brightness.supports,
    toZigbee: generic.light_onoff_brightness.toZigbee.concat([tz.osram_cmds]),
    fromZigbee: generic.light_onoff_brightness.fromZigbee,
},
```

Thor2503, 2020-01-08:
thanks a lot.
after activating the reporting the linkquality is now showing up.
the second part I will try within the next days and report back

closed 2020-03-15 by stale[bot]

shirou93, 2020-11-26:
> Good evening,
> First of all I would like to say that I like very much the Zigbee2MQTT "tool" and say thank you for all the work that has flown into the project. I came across z2m a couple of months [...]

This bulp work with new zigbee2mqtt without changes in files?