Koenkk/zigbee2mqtt #2674: Update UniFlash page and FAQ for CC26X2R1

choeflake, 2020-01-04:
Today I was struggling to get a CC26X2R1 working. First issue I ran into was that used port. When the device connects, it registers /dev/ttyACM0 and /dev/ttyACM1. The ttyACM0 is for data and ttyACM1 is for debug. Previously I had another CC2531 running, so I thought that the system has ttyACM1 selected for the CC26X2R1.
Looking at /dev/serial/by-id, we see:
- usb-Texas_Instruments_XDS110__03.00.00.05__Embed_with_CMSIS-DAP_L1100BT8-if00  -> /dev/ttyACM0
- usb-Texas_Instruments_XDS110__03.00.00.05__Embed_with_CMSIS-DAP_L1100BT8-if03 .-> /dev/ttyACM1

So the device id ending with if00 is for data! 

The next problem I was experiencing was caused due to incorrect flashing. I got this error:
```
Error: SREQ '--> SYS - osalNvWrite - {"id":3,"len":1,"offset":0,"value":{"type":"Buffer","data":[2]}}' failed with status '2' (expected '0')
     at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:245:27)
     at Generator.next (<anonymous>)
     at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
```
The flash page has to add a little but essential step:

1. connect the board to a free USB port on the PC
2. open the UniFlash Tool
3. perform a manually erase of the chip, to do this go to the "Settings & Utilities" menu and set the option "Erase entire flash (You must reconnect to the target afterwards)" on the manual erase section and perform the erase.
4. close UniFlash
5. disconnect and reconnect the board to the USB PC port
6. reopen UniFlash
7. **Under the "Settings & Utilities" menu change the option inside the "Erase Settings during Program Load" section to the "All Unprotected Sectors"**
8. Load the .hex file
9. Flash the firmware
10. close UniFlash

Step 7 is essential. This is found by @ChumKiu at https://github.com/Koenkk/zigbee2mqtt/issues/1429.

Please at this step to the uniflash page at https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html.
And please write something ttyACM0 story to the FAQ or somewhere else.
This could save a lot of time for others!

nldroid, 2020-01-06:
You can edit the file yourself:

https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/information/flashing_via_uniflash.md

And if you create a pull request, it can be merged by koen.

choeflake, 2020-01-06:
Pull request is created: https://github.com/Koenkk/zigbee2mqtt.io/pull/196

Koenkk, 2020-01-08:
Closing this as https://github.com/Koenkk/zigbee2mqtt.io/pull/196 has been merged, thanks!

closed 2020-01-08 by Koenkk