Koenkk/zigbee2mqtt #2675: no status report from in-wall switch

eddso, 2020-01-04:
# Bug Report
No status report from two supported in-wall switches: Sunricher ZigBee AC in-wall switch (ZG9101SAC-HP-Switch) and Müller Licht Tint smart switch (404021).
No event even with debug level.
## What happened
- Added device to network: ok success while device supported (bought it because on supported list). 
- Control the switch over mqtt publish .../set: work.
- Control switch locally over button, switch works but no status on/off is received from z2m.
 
## What did you expect to happen
Expecting status of the switch in z2m
## How to reproduce it (minimal and precise)
Just add to network and switch on/off with frontside button of switch
## Debug Info
zigbee2mqtt version: 1.8.0 (Docker
CC253X firmware version: CC2531 with FW: 20190608

eddso, 2020-01-17:
ok, extended the Müller Licht Tint smart switch (zigbeeModel: ['tint Smart Switch'], 404021) in devices.js with:
```
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint);
        },
```
and the switch reports his events

closed 2020-01-17 by eddso

meisterlampe, 2020-01-30:
Thanks for sharing. 
Just stumbled over the same problem. Your fix also made my Tint Smart Switch report its state back.
Unfortunately there is now an error in the logs, each time, I use the switch. Any idea?

`zigbee2mqtt:error 2020-01-30 20:21:15: Failed to call 'DeviceConfigure' 'onZigbeeEvent' (TypeError: Cannot read property 'configureKey' of undefined
    at DeviceConfigure.shouldConfigure (/app/lib/extension/deviceConfigure.js:22:58)
    at DeviceConfigure.onZigbeeEvent (/app/lib/extension/deviceConfigure.js:81:18)
    at Controller.callExtensionMethod (/app/lib/controller.js:327:44))
`

meisterlampe, 2020-01-30:
What I just tried:
- rolled back the fix in devices.json --> The error was still showing up
- rolled back the line containing the switch in the database.db from a backup --> The error was gone

Now everything is working like a charm, even when I reapply your fix. Maybe the switch did get something wrong. Linkquality is very poor there.