Koenkk/zigbee2mqtt #2678: Remove| Force_remove does not remove from HA MQTT discovery

NewFolk, 2020-01-05:
Hi,

I have HA 0.103.5 in docker and 1.8.0 Zigbee2MQTT.

Today I decide to remove one lamp.
```
zigbee2mqtt/bridge/config/remove 0x086bd7fffe2184dd
zigbee2mqtt/bridge/log {"type":"device_removed","message":"0x086bd7fffe2184dd"}
zigbee2mqtt/bridge/log {"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x086bd7fffe2184dd"}}
```

But it does not leave Mqtt discovery integration and after Zigbee2MQTT restart i see this topic:
`homeassistant/sensor/0x086bd7fffe2184dd/linkquality/config`

`zigbee2mqtt/bridge/config/force_remove` does not hep me too.

But `zigbee2mqtt/bridge/networkmap `does not show me 0x086bd7fffe2184dd.
And it looks like it is some buffer special for home assistant

nldroid, 2020-01-07:
Could it be that it is still in your MQTT server?

https://community.home-assistant.io/t/easiest-way-to-remove-mqtt-device/130466

NewFolk, 2020-01-08:
Hello, thank you for your link. I got a lot of new info from it.
I deleted information by MQTT Explorer, restart HA, but it does not help. Will take a lot at HA database.

I think that delete action from "homeassistant/" should be added to "Remove" command in z2m.

Koenkk, 2020-01-09:
@NewFolk agreed and implemented, latest zigbee2mqtt dev branch will now clear the HA discovery topics on device removal.

closed 2020-01-09 by Koenkk