Koenkk/zigbee2mqtt #2679: Zigbee led driver

turboproc, 2020-01-05:
So this might be a bit off topic but Zigbee related anyway.

Looking for a led driver (220v in, 12 v out, let's say 60 watt) wich is Zigbee controllable. On / off would be sufficient, no need for a dimmer.
Application would be my garden lightning.

Any idea whether a off the shelf solution exists for this?

zladukas, 2020-01-05:
Why You don't want to use Ikea outlet(220V 16A) and/or sonoff zigbee (220V 10A) and drive any existing led transformer?

turboproc, 2020-01-05:
Thnx for your suggestion. Would be an option for indoor use. Having it outdoors, I would need something of IP44 or similar grade.

zladukas, 2020-01-05:
I'm sure You can find plenty of ip<something> boxes in nearest homedepot. I've seen some chneese drivers (for strips with dimming), but I'm realy sure they aren't IP44

closed 2020-03-12 by stale[bot]