Koenkk/zigbee2mqtt #2681: RTCGQ01LM is not able to pair

Danny-P, 2020-01-05:
I am not able to pair a Xiaomi Motion sensor RTCGQ01LM with my CC2531 stick. 
When doing the pairing as the documentation describes, I get not entry in the log and nothing happens. 
That is my zigbee2mqtt/bridge/config output: 
`{"version":"1.8.0",
  "commit":"da4d26a",
  "coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}`

I tried changing the battery and I tried another motion sensor as well. But nothing worked.

chavalaloco, 2020-01-08:
Same Issue here! 

i´m running zigbee2mqtt on a Intel NUC with debian 10 with a CC2531 Stick and zigbee2mqtt v 1.8 -dev:
here is my log output:

`info  2020-01-07 19:06:01: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)                                                                                                                                    
info  2020-01-07 19:06:01: Starting zigbee-herdsman...                                                                                                                                                             
debug 2020-01-07 19:06:01: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2$
info  2020-01-07 19:06:04: zigbee-herdsman started                                                                                                                                                                 
info  2020-01-07 19:06:04: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'                                    
debug 2020-01-07 19:06:04: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}                                                                                             
info  2020-01-07 19:06:04: Currently 16 devices are joined:                                                                                                                                                        `

i could get the following paired, 
10 pcs. MCCGQ01LM - contact Sensor
5 pcs. WXKG01LM - Switch 
1 pcs. WSDCGQ01LM - Temp Sensor 

but my stick doesn´t recognize any attempt of the Motion Sensors: RTCGQ01LM and RTCGQ11LM, also the stick doesn´t recognize the following IKEA Stuff: 

ICPSHC24-30EU-IL-1 - IKEA TRADFRI driver for wireless control (30 watt) 
E1524/E1810 - IKEA TRADFRI remote control
E1744 - IKEA SYMFONISK sound controller
E1525 - IKEA TRADFRI motion sensor

Koenkk, 2020-01-09:
Do you have any other bridge or zigbee network running?

Can you provide the herdsman debug logging when trying to pair the sensor?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

chavalaloco, 2020-01-09:
Thank you for your help! 

No i have only the CC2531 stick running, before i had the stick the devices were paired to a XIAOMI Bridge, but it isn´t running anymore... 

I will provide the logfile this evening.. 

I noticed that my stick coordinator stick is running on the default firmware, in a other post i´ve readed that maybe the source-reading fw can solve the problem, but i haven´t got the cc debugger + adapter cable right now.

closed 2020-03-16 by stale[bot]

SIM0N-F, 2020-11-16:
Hi, I have same setup and same issue, can you help me please?

So, I have juste one coordinator (CC2531), I use raspbian and Zigbee2mqtt 1.15.0.

When I want pairing my device, I dont see anything in logs.

Many thanks by advance.

chavalaloco, 2020-11-17:
which firmware do you have flashed on your coordinator? 
Do you already have any device paired yet?
Can you post your logs please?

SIM0N-F, 2020-11-17:
Hi @chavalaloco, thanks for your answer.

Yes I have some aqara devices paired. I have IKEA TRADFRI  for router, but I have test without router and I have same issue. 

Yesterday I have update my version of Zigbee2mqtt, but same issue.

For CC2531 : 
info  2020-11-16 21:39:20: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'

My logs:

################################################
head -n30 data/log/2020-11-16.21-39-12/log.txt
info  2020-11-16 21:39:16: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-11-16.21-39-12' filename: log.txt
info  2020-11-16 21:39:16: Starting Zigbee2MQTT version 1.16.1 (commit #6b32f30)
info  2020-11-16 21:39:16: Starting zigbee-herdsman...
info  2020-11-16 21:39:20: zigbee-herdsman started
info  2020-11-16 21:39:20: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
info  2020-11-16 21:39:20: Currently 10 devices are joined:
info  2020-11-16 21:39:20: 0x00158d00033a48ac (0x00158d00033a48ac): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)      
info  2020-11-16 21:39:20: 0x00158d000391ce8a (0x00158d000391ce8a): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
info  2020-11-16 21:39:20: 0x00158d00044f8211 (0x00158d00044f8211): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)      
info  2020-11-16 21:39:20: 0x00158d000392010c (0x00158d000392010c): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
info  2020-11-16 21:39:20: 0x00158d000346ae50 (0x00158d000346ae50): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2020-11-16 21:39:20: 0x00158d00044f7f42 (0x00158d00044f7f42): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)      
info  2020-11-16 21:39:20: 0x00158d0003216b2f (0x00158d0003216b2f): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2020-11-16 21:39:20: 0x00158d0003cb7b89 (0x00158d0003cb7b89): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
info  2020-11-16 21:39:20: 0x00158d000483c062 (0x00158d000483c062): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
info  2020-11-16 21:39:20: 0x680ae2fffe82be51 (0x680ae2fffe82be51): E1603/E1702 - IKEA TRADFRI control outlet (Router)
warn  2020-11-16 21:39:20: `permit_join` set to  `true` in configuration.yaml.
warn  2020-11-16 21:39:20: Allowing new devices to join.
warn  2020-11-16 21:39:20: Set `permit_join` to `false` once you joined all devices.
info  2020-11-16 21:39:20: Zigbee: allowing new devices to join.
info  2020-11-16 21:39:20: Started frontend on port 8099
info  2020-11-16 21:39:20: Connecting to MQTT server at mqtt://192.168.1.66
info  2020-11-16 21:39:21: Connected to MQTT server
...
################################################

chavalaloco, 2020-11-17:
Please try the source_routing fw for the coordinator, for me it solved the problem.
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20190619.zip