Koenkk/zigbee2mqtt #2683: Handling wifi disconnects for ESP8266 / CC2530

Z0l, 2020-01-06:
Running 1.8.0 from docker with the coordinator being an ESP8266 with a CC2530 flashed to the 1.2 coordinator firmware. I reboot my wifi routers regularly (to fix memory leaks and other weird issues that come from lack of reboots) but now that the coordinator is on wifi as well, I noticed that once the connection was lost at 3:30AM, the app didn't even try to connect back until I restarted the thing in the evening. See below log snippet with some details:

[code]
error 2020-01-05 03:30:44: Adapter disconnected, stopping
info  2020-01-05 19:42:20: Logging to console and directory: 'data/log'
info  2020-01-05 19:42:20: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
info  2020-01-05 19:42:20: Starting zigbee-herdsman...
error 2020-01-05 19:42:27: Error while starting zigbee-herdsman
info  2020-01-05 19:42:40: Logging to console and directory: 'data/log'
info  2020-01-05 19:42:41: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
info  2020-01-05 19:42:41: Starting zigbee-herdsman...
info  2020-01-05 19:42:46: zigbee-herdsman started
info  2020-01-05 19:42:46: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
[/code]

Is it possible to introduce a setting that allows connection retries when virtual serial port is used?

pepp86, 2020-01-06:
Which restart policy do you have for your docker container?

Z0l, 2020-01-07:
I didn't even know I have one until you asked :) I was following the instructions here, so turns out I don't have a restart policy: https://www.zigbee2mqtt.io/information/docker.html

Maybe this could be updated to include --restart on-failure (or one of the other flavours)?

closed 2020-03-14 by stale[bot]