Koenkk/zigbee2mqtt #2686: No network route (205) on single device but not in groups

ondoteam, 2020-01-06:
# Bug Report
## What happened
When I try to switch on or off entities I am getting _No network route (205)_ message but there is no problem if I include those failing entities in a group and send actions to those groups.

## What did you expect to happen
A normal operation in both cases or _No network route (205)_ message in both cases.

## How to reproduce it (minimal and precise)
Failing bulbs are random. Normally if I disconect / connect those bulbs they start working... but the problem occurs once in a while with other devices. By the way it is always the same thing: if they stop responding I can add it to a group and they start working (acting over this group not over the individual entity).

## Debug Info
zigbee2mqtt version: latest dev version
CC253X firmware version: 20190619 source routing firmware

`
zigbee2mqtt:error 2020-01-06 19:26:13: Publish 'set' 'state' to 'Azotea doble derecha' failed: 'Error: Data request failed with error: 'No network route' (205)'
zigbee2mqtt:info  2020-01-06 19:26:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Azotea doble derecha' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"Azotea doble derecha"}}'
zigbee2mqtt:error 2020-01-06 19:26:14: Publish 'set' 'state' to 'Azotea doble izquierda' failed: 'Error: Data request failed with error: 'No network route' (205)'
zigbee2mqtt:info  2020-01-06 19:26:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Azotea doble izquierda' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"Azotea doble izquierda"}}'
zigbee2mqtt:info  2020-01-06 19:26:16: MQTT publish: topic 'zigbee2mqtt/Escalera exterior', payload '{"state":"ON","linkquality":52,"brightness":128,"color_temp":454}'
zigbee2mqtt:info  2020-01-06 19:26:16: MQTT publish: topic 'zigbee2mqtt/exterior_azotea_grupo', payload '{"state":"ON","brightness":128,"color_temp":454}'
zigbee2mqtt:info  2020-01-06 19:26:16: MQTT publish: topic 'zigbee2mqtt/Maquinas exterior', payload '{"state":"ON","brightness":128,"color_temp":454,"linkquality":52}'
zigbee2mqtt:info  2020-01-06 19:26:16: MQTT publish: topic 'zigbee2mqtt/Azotea doble izquierda', payload '{"state":"ON","brightness":128,"color_temp":454,"linkquality":31}'
zigbee2mqtt:info  2020-01-06 19:26:16: MQTT publish: topic 'zigbee2mqtt/Azotea doble derecha', payload '{"state":"ON","brightness":128,"color_temp":454,"linkquality":31}'
zigbee2mqtt:info  2020-01-06 19:26:16: MQTT publish: topic 'zigbee2mqtt/Azotea simple', payload '{"state":"ON","brightness":128,"color_temp":454,"linkquality":52}'
zigbee2mqtt:info  2020-01-06 19:26:17: MQTT publish: topic 'zigbee2mqtt/exterior_azotea_grupo', payload '{"state":"ON","brightness":128,"color_temp":454}'
zigbee2mqtt:info  2020-01-06 19:26:17: MQTT publish: topic 'zigbee2mqtt/Escalera exterior', payload '{"state":"ON","linkquality":52,"brightness":128,"color_temp":454}'
zigbee2mqtt:info  2020-01-06 19:26:17: MQTT publish: topic 'zigbee2mqtt/exterior_azotea_grupo', payload '{"state":"ON","brightness":128,"color_temp":454}'
zigbee2mqtt:info  2020-01-06 19:26:17: MQTT publish: topic 'zigbee2mqtt/Maquinas exterior', payload '{"state":"ON","brightness":128,"color_temp":454,"linkquality":52}'
zigbee2mqtt:info  2020-01-06 19:26:18: MQTT publish: topic 'zigbee2mqtt/exterior_azotea_grupo', payload '{"state":"ON","brightness":128,"color_temp":454}'
zigbee2mqtt:info  2020-01-06 19:26:18: MQTT publish: topic 'zigbee2mqtt/Maquinas exterior', payload '{"state":"ON","brightness":128,"color_temp":454,"linkquality":52}'
zigbee2mqtt:info  2020-01-06 19:26:18: MQTT publish: topic 'zigbee2mqtt/exterior_azotea_grupo', payload '{"state":"ON","brightness":128,"color_temp":454}'
zigbee2mqtt:info  2020-01-06 19:26:18: MQTT publish: topic 'zigbee2mqtt/Azotea simple', payload '{"state":"ON","brightness":128,"color_temp":454,"linkquality":49}'
zigbee2mqtt:info  2020-01-06 19:26:18: MQTT publish: topic 'zigbee2mqtt/exterior_azotea_grupo', payload '{"state":"ON","brightness":128,"color_temp":454}'
zigbee2mqtt:info  2020-01-06 19:26:18: MQTT publish: topic 'zigbee2mqtt/Escalera exterior', payload '{"state":"ON","linkquality":47,"brightness":128,"color_temp":454}'
`

myradon, 2020-01-21:
Go the same thing on a Gledopto driver. Yesterday pulled v1.9.0 docker container.

`{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'light_cove_hall' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"light_cove_hall"}}`

ondoteam, 2020-01-21:
> Go the same thing on a Gledopto driver. Yesterday pulled v1.9.0 docker container.
> 
> `{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'light_cove_hall' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"light_cove_hall"}}`

Hello, works inside a group?

myradon, 2020-01-21:
Nope! Just a single device. 64 of these error messages within 30 minutes in bridge/log this afternoon. Now quit for several hours.

LeoDJ, 2020-08-18:
Hey, I have the same issue, it is not stale and should be reopened

Koenkk, 2020-08-18:
@LeoDJ please share some details about your setup, amount of devices, adapter and firmware version.

MattWestb, 2020-08-19:
One thing is that sending to one group is broadcast and the sender or the network dont tracking the message only forwarding it to all devices that is possible. The sending to one device is unicast and the network is tracking it all the way then its traveling thru the net. So if one link is temporary broken its not possible seeing that with group messages only with "personal" messages in the net. I dot have any Gledopto and leaving that to our specialists ;-)

closed 2020-09-27 by github-actions[bot]