Koenkk/zigbee2mqtt #2689: I made my Coordinator to powerful and now nothing connects to any of the router

fefris, 2020-01-07:
Any help appreciated for this self induced issue.

I added a aerial to my coordinator to boost coverage and seem to have made it so powerful that EVERYTHING connects directly to it even if there is a router next to it.

see my map, its a piece of beauty but this issue means i cannot add more than 15 devices. When i try to add #16 it fails to pair until i delete something and then we are good again. 

[https://imgur.com/a/sN6Boxq](url)

Suggestions?

ArkadiuszNiemiec, 2020-01-07:
I had a similar issue and have not found a way to configure a device to connect to a specific router. What I did was repairing Xiaomi devices very close to a router and then move it to its spot. Xiaomi devices stays connected to the router they were paired to. I know it not solved the problem but it might help someone.

GrumpyMeow, 2020-01-07:
I think i had the same issue. I was unable to get all my devices paired.
I eventually flashed the "Sourcerouting" version of Z2M and managed to get everything paired.
The "sourcerouting" firmware only allows 5 devices to pair. Because of this limit, other devices will pair to other routers.

Pockrandt, 2020-01-10:
maybe try and experiment with
```
experimental:
  transmit_power: 
```
Allowed values are -22 dBm (min) till 19 dBm (max), default is 0 except for the CC2530 + CC2591 = 4 and CC2530 + CC2592 = 19
from https://github.com/Koenkk/zigbee2mqtt/issues/2253

closed 2020-03-17 by stale[bot]