Koenkk/zigbee2mqtt #2690: Z2M stopped working overnight, fails to start

EmmanuelLM, 2020-01-07:
# Bug Report
Hassio Z2M v1.8.0 worked fine for the last few weeks and stopped working overnight. It now will not start.
## What happened
No changes, just left running
## What did you expect to happen

## How to reproduce it (minimal and precise)
Unsure
## Debug Info
zigbee2mqtt version: 1.8.0
CC2531 firmware version: CC2531_DEFAULT_20190608

Error log:
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-01-07T09:50:34: PM2 log: Launching in no daemon mode
2020-01-07T09:50:34: PM2 log: App [npm:0] starting in -fork mode-
2020-01-07T09:50:34: PM2 log: App [npm:0] online
> zigbee2mqtt@1.8.0 start /zigbee2mqtt-1.8.0
> node index.js
zigbee2mqtt:info  2020-01-07 09:50:37: Logging to console and directory: '/share/zigbee2mqtt/log/2020-01-07.09-50-37'
zigbee2mqtt:debug 2020-01-07 09:50:37: Removing old log directory '/share/zigbee2mqtt/log/2020-01-07.09-45-30'
zigbee2mqtt:debug 2020-01-07 09:50:38: Loaded state from file /share/zigbee2mqtt/state.json
zigbee2mqtt:info  2020-01-07 09:50:38: Starting zigbee2mqtt version 1.8.0 (commit #unknown)
zigbee2mqtt:info  2020-01-07 09:50:38: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-01-07 09:50:38: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM1"}}'
zigbee2mqtt:error 2020-01-07 09:50:45: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-01-07 09:50:45: Failed to start zigbee
zigbee2mqtt:error 2020-01-07 09:50:45: Exiting...
zigbee2mqtt:error 2020-01-07 09:50:45: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/zigbee2mqtt-1.8.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.8.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.8.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-01-07T09_50_45_686Z-debug.log
2020-01-07T09:50:47: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-01-07T09:50:47: PM2 log: App [npm:0] starting in -fork mode-
2020-01-07T09:50:47: PM2 log: App [npm:0] online

nldroid, 2020-01-07:
Did you check if the device is still on /dev/ttyACM1?

EmmanuelLM, 2020-01-07:
Yes it is

EmmanuelLM, 2020-01-07:
Reflash the stick, now it all works again... how odd

closed 2020-01-07 by EmmanuelLM

pcfreak1, 2020-12-27:
I have the same issue, the only thing is that after reflashing the stick it works again untill the next morning, then i have to reflash the stick again...

Koenkk, 2020-12-27:
@pcfreak1 check https://github.com/Koenkk/zigbee2mqtt/issues/4584#issuecomment-751257141