Koenkk/zigbee2mqtt #2691: 45853GE missing energy reporting

djchen, 2020-01-07:
I re-paired one of my 45853GE plugs which has energy reporting and it no longer reports power usage.

1. Removed plug from network
2. Reset plug and it re-joined network
3. Able to turn it on and off but does not report power usage
I've repeated this 3 times, same results.

Logs from pairing:
```
info  2019-11-14T05:04:41: Device '<snip>' joined
info  2019-11-14T05:04:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"<snip>"}}'
info  2019-11-14T05:04:41: Starting interview of '<snip>'
info  2019-11-14T05:04:52: MQTT publish: topic 'homeassistant/switch/<snip>/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/<snip>/set","state_topic":"zigbee2mqtt/<snip>","name":"<snip>_switch","unique_id":"<snip>_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_<snip>"],"name":"<snip>","sw_version":"Zigbee2mqtt 1.7.1","model":"Plug-in smart switch (45853GE)","manufacturer":"GE"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2019-11-14T05:04:52: MQTT publish: topic 'homeassistant/sensor/<snip>/power/config', payload '{"unit_of_measurement":"W","icon":"mdi:factory","value_template":"{{ value_json.power }}","state_topic":"zigbee2mqtt/<snip>","json_attributes_topic":"zigbee2mqtt/<snip>","name":"<snip>_power","unique_id":"<snip>_power_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_<snip>"],"name":"<snip>","sw_version":"Zigbee2mqtt 1.7.1","model":"Plug-in smart switch (45853GE)","manufacturer":"GE"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2019-11-14T05:04:52: MQTT publish: topic 'homeassistant/sensor/<snip>/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/<snip>","json_attributes_topic":"zigbee2mqtt/<snip>","name":"<snip>_linkquality","unique_id":"<snip>_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_<snip>"],"name":"<snip>","sw_version":"Zigbee2mqtt 1.7.1","model":"Plug-in smart switch (45853GE)","manufacturer":"GE"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2019-11-14T05:04:52: Successfully interviewed '<snip>', device has successfully been paired
info  2019-11-14T05:04:52: Device '<snip>' is supported, identified as: GE Plug-in smart switch (45853GE)
info  2019-11-14T05:04:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"<snip>","model":"45853GE","vendor":"GE","description":"Plug-in smart switch","supported":true}}'
info  2019-11-14T05:04:52: Configuring '<snip>'
info  2019-11-14T05:04:53: Succesfully configured '<snip>'
```

Problem device in database.db
```
{"id":22,"type":"Router","ieeeAddr":"<snip>","nwkAddr":54555,"manufId":4388,"manufName":"Jasco Products","powerSource":"Mains (single phase)","modelId":"45853","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,2821,1794],"outClusterList":[3,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"45853","manufacturerName":"Jasco Products","powerSource":1,"zclVersion":1,"appVersion":18,"stackVersion":80,"hwVersion":1}},"seMetering":{"attributes":{"multiplier":1,"divisor":10000}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"<snip>","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":259,"inClusterList":[0,3,2821],"outClusterList":[3,6],"clusters":{},"binds":[]}},"appVersion":18,"stackVersion":80,"hwVersion":1,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":2}}
```

Fully working device in database.db (another of the same plug)
```
{"id":16,"type":"Router","ieeeAddr":"<diff_device>","nwkAddr":6601,"manufId":4388,"manufName":"Jasco Products","powerSource":"Mains (single phase)","modelId":"45853","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,1794,2821],"outClusterList":[3,10,25],"clusters":{"genBasic":{"dir":{"value":1},"attributes":{"zclVersion":1,"appVersion":18,"stackVersion":80,"hwVersion":1,"manufacturerName":"Jasco Products","modelId":"45853","powerSource":1}},"genIdentify":{"dir":{"value":3},"attributes":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attributes":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attributes":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attributes":{"onOff":0}},"genTime":{"dir":{"value":2},"attributes":{}},"genOta":{"dir":{"value":2},"attributes":{}},"seMetering":{"dir":{"value":1},"attributes":{"currentSummDelivered":{"0":0,"1":0},"status":0,"unitOfMeasure":0,"multiplier":1,"divisor":10000,"summaFormatting":251,"demandFormatting":11,"historicalConsumpFormatting":120,"meteringDeviceType":0,"instantaneousDemand":0}},"haDiagnostic":{"dir":{"value":1},"attributes":{"averageMacRetryPerApsMessageSent":1,"lastMessageLqi":255,"lastMessageRssi":-17}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"<snip>","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":259,"inClusterList":[0,3,2821],"outClusterList":[3,6],"clusters":{"genBasic":{"dir":{"value":1},"attributes":{"zclVersion":1,"appVersion":18,"stackVersion":80,"hwVersion":1,"manufacturerName":"Jasco Products","modelId":"45853","powerSource":1}},"genIdentify":{"dir":{"value":3},"attributes":{"identifyTime":0}},"genOnOff":{"dir":{"value":2},"attributes":{}},"haDiagnostic":{"dir":{"value":1},"attributes":{"averageMacRetryPerApsMessageSent":1,"lastMessageLqi":255,"lastMessageRssi":-17}}},"binds":[]}},"hwVersion":1,"meta":{"configured":2}}
```

Turning it on and off:
```
info  2019-11-14T05:17:03: MQTT publish: topic 'zigbee2mqtt/<problem_device>', payload '{"state":"ON","linkquality":39,"last_seen":1573708623871}'
info  2019-11-14T05:17:03: MQTT publish: topic 'zigbee2mqtt/<problem_device>', payload '{"state":"ON","linkquality":44,"last_seen":1573708623872}'
info  2019-11-14T05:19:37: MQTT publish: topic 'zigbee2mqtt/<problem_device>', payload '{"state":"OFF","linkquality":44,"last_seen":1573708777448}'
info  2019-11-14T05:19:39: MQTT publish: topic 'zigbee2mqtt/<problem_device>', payload '{"state":"OFF","linkquality":141,"last_seen":1573708779002}'
info  2019-11-14T05:19:41: MQTT publish: topic 'zigbee2mqtt/<problem_device>', payload '{"state":"ON","linkquality":141,"last_seen":1573708781801}'
info  2019-11-14T05:19:41: MQTT publish: topic 'zigbee2mqtt/<problem_device>', payload '{"state":"ON","linkquality":141,"last_seen":1573708781802}'
info  2019-11-14T05:20:02: MQTT publish: topic 'zigbee2mqtt/<problem_device>', payload '{"state":"OFF","linkquality":141,"last_seen":1573708802675}'
info  2019-11-14T05:20:02: MQTT publish: topic 'zigbee2mqtt/<problem_device>', payload '{"state":"OFF","linkquality":42,"last_seen":1573708802676}'
info  2019-11-14T05:20:35: MQTT publish: topic 'zigbee2mqtt/<problem_device>', payload '{"state":"ON","linkquality":42,"last_seen":1573708835753}'
info  2019-11-14T05:20:35: MQTT publish: topic 'zigbee2mqtt/<problem_device>', payload '{"state":"ON","linkquality":139,"last_seen":1573708835755}'
```
(I can confirm it physically turns on and off just fine)

passy05, 2020-01-08:
Thanks @Koenkk ! That fixed the issue with my Hive active plug.

passy05, 2020-01-08:
I have now connected my Hive active plug to my flood light and my automation works fine when power is drained from the flood light. However, I have noticed that messages are frequently published when the flood light is inactive as the power fluctuates as 0 or 1.  Is the power value rounded?

Koenkk, 2020-01-08:
@passy05 in that case use the debounce option: https://www.zigbee2mqtt.io/information/configuration.html#all-devices

djchen, 2020-01-16:
I'm still not seeing power usage for the 45853GE. Running ` 1.8.0-dev (commit #7d26873)`

```
zigbee2mqtt:info  2020-01-15 22:11:58: Logging to console and directory: '/app/data/log/2020-01-15.22-11-58'
zigbee2mqtt:debug 2020-01-15 22:11:59: Can't load state from file /app/data/state.json (doesn't exist)
zigbee2mqtt:info  2020-01-15 22:11:59: Starting zigbee2mqtt version 1.8.0-dev (commit #7d26873)
zigbee2mqtt:info  2020-01-15 22:11:59: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-01-15 22:11:59: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/zigbee2mqtt"}}'
zigbee2mqtt:info  2020-01-15 22:12:00: zigbee-herdsman started
zigbee2mqtt:info  2020-01-15 22:12:00: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'
zigbee2mqtt:debug 2020-01-15 22:12:00: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:info  2020-01-15 22:12:00: Currently 0 devices are joined:
zigbee2mqtt:warn  2020-01-15 22:12:00: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-01-15 22:12:00: Allowing new devices to join.
zigbee2mqtt:warn  2020-01-15 22:12:00: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-01-15 22:12:00: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-01-15 22:12:00: Connecting to MQTT server at mqtt://172.17.0.1:1883
zigbee2mqtt:info  2020-01-15 22:12:00: Connected to MQTT server
zigbee2mqtt:info  2020-01-15 22:12:00: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-01-15 22:12:00: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0-dev","commit":"7d26873","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},"log_level":"debug","permit_join":true}'
zigbee2mqtt:warn  2020-01-15 22:12:17: Device '0x0022a300001af3b6' left the network
zigbee2mqtt:info  2020-01-15 22:12:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x0022a300001af3b6"}}'
zigbee2mqtt:warn  2020-01-15 22:12:17: Device '0x0022a300001af3b6' left the network
zigbee2mqtt:info  2020-01-15 22:12:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x0022a300001af3b6"}}'
zigbee2mqtt:warn  2020-01-15 22:12:18: Device '0x0022a300001af3b6' left the network
zigbee2mqtt:info  2020-01-15 22:12:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x0022a300001af3b6"}}'
zigbee2mqtt:info  2020-01-15 22:12:19: Device '0x0022a300001af3b6' joined
zigbee2mqtt:info  2020-01-15 22:12:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0022a300001af3b6"}}'
zigbee2mqtt:info  2020-01-15 22:12:19: Starting interview of '0x0022a300001af3b6'
zigbee2mqtt:info  2020-01-15 22:12:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0022a300001af3b6"}}'
zigbee2mqtt:debug 2020-01-15 22:12:19: Device '0x0022a300001af3b6' announced itself
zigbee2mqtt:debug 2020-01-15 22:12:19: Received Zigbee message from '0x0022a300001af3b6', type 'readResponse', cluster 'genBasic', data '{"modelId":"45853","manufacturerName":"Jasco Products","powerSource":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-01-15 22:12:19: No converter available for '45853GE' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"45853","manufacturerName":"Jasco Products","powerSource":1}'
zigbee2mqtt:info  2020-01-15 22:12:19: MQTT publish: topic 'homeassistant/switch/0x0022a300001af3b6/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x0022a300001af3b6/set","state_topic":"zigbee2mqtt/0x0022a300001af3b6","json_attributes_topic":"zigbee2mqtt/0x0022a300001af3b6","name":"0x0022a300001af3b6_switch","unique_id":"0x0022a300001af3b6_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0022a300001af3b6"],"name":"0x0022a300001af3b6","sw_version":"Zigbee2mqtt 1.8.0-dev","model":"Plug-in smart switch (45853GE)","manufacturer":"GE"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-01-15 22:12:19: MQTT publish: topic 'homeassistant/sensor/0x0022a300001af3b6/power/config', payload '{"unit_of_measurement":"W","icon":"mdi:factory","value_template":"{{ value_json.power }}","state_topic":"zigbee2mqtt/0x0022a300001af3b6","json_attributes_topic":"zigbee2mqtt/0x0022a300001af3b6","name":"0x0022a300001af3b6_power","unique_id":"0x0022a300001af3b6_power_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0022a300001af3b6"],"name":"0x0022a300001af3b6","sw_version":"Zigbee2mqtt 1.8.0-dev","model":"Plug-in smart switch (45853GE)","manufacturer":"GE"},"availability_topic":"zigbee2mqtt/bridge/state"}'
[... cut ...] g":{"attributes":{"multiplier":1,"divisor":10000}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f410","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e1f410","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":259,"inClusterList":[0,3,2821],"outClusterList":[3,6],"clusters":{},"binds":[]}},"appVersion":18,"stackVersion":80,"hwVersion":1,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":3},"lastSeen":1579155140608}
```

djchen, 2020-01-16:
I confirmed that with version 1.6.0 it is working:

```
  zigbee2mqtt:info 1/15/2020, 10:22:31 PM Logging to directory: '/app/data/log/2020-01-15.22-22-31'
  zigbee2mqtt:debug 1/15/2020, 10:22:31 PM Can't load state from file /app/data/state.json (doesn't exist)
  zigbee2mqtt:info 1/15/2020, 10:22:31 PM Starting zigbee2mqtt version 1.6.0 (commit #e26ad2a)
  zigbee2mqtt:info 1/15/2020, 10:22:31 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 1/15/2020, 10:22:31 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/app/data/database.db","coordBackupPath":"/app/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:info 1/15/2020, 10:22:33 PM zigbee-shepherd started
  zigbee2mqtt:info 1/15/2020, 10:22:33 PM Coordinator firmware version: '20190619'
  zigbee2mqtt:debug 1/15/2020, 10:22:33 PM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"11","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b0018e1f410","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190619},"startTime":1579155753,"joinTimeLeft":0}
  zigbee2mqtt:info 1/15/2020, 10:22:33 PM Currently 0 devices are joined:
  zigbee2mqtt:warn 1/15/2020, 10:22:33 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 1/15/2020, 10:22:33 PM Allowing new devices to join.
  zigbee2mqtt:warn 1/15/2020, 10:22:33 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 1/15/2020, 10:22:33 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 1/15/2020, 10:22:33 PM Connecting to MQTT server at mqtt://172.17.0.1:1883
  zigbee2mqtt:info 1/15/2020, 10:22:33 PM zigbee-shepherd ready
  zigbee2mqtt:info 1/15/2020, 10:22:33 PM Connected to MQTT server
  zigbee2mqtt:info 1/15/2020, 10:22:33 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 1/15/2020, 10:22:33 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"e26ad2a","coordinator":20190619,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 1/15/2020, 10:22:33 PM Mounted the cieApp (epId 11)
  zigbee2mqtt:debug 1/15/2020, 10:22:59 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"Jasco Products","modelId":"45853","powerSource":1}}' of device '45853' (0x0022a300001af3b6) of endpoint 1
  zigbee2mqtt:info 1/15/2020, 10:22:59 PM New device '45853' with address 0x0022a300001af3b6 connected!
  zigbee2mqtt:info 1/15/2020, 10:22:59 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x0022a300001af3b6","meta":{"modelID":"45853"}}'
  zigbee2mqtt:debug 1/15/2020, 10:22:59 PM Configuring 0x0022a300001af3b6 (0x0022a300001af3b6) ...
  zigbee2mqtt:info 1/15/2020, 10:22:59 PM MQTT publish: topic 'homeassistant/switch/0x0022a300001af3b6/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x0022a300001af3b6/set","state_topic":"zigbee2mqtt/0x0022a300001af3b6","name":"0x0022a300001af3b6_switch","unique_id":"0x0022a300001af3b6_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0022a300001af3b6"],"name":"0x0022a300001af3b6","sw_version":"Zigbee2mqtt 1.6.0","model":"Plug-in smart switch (45853GE)","manufacturer":"GE"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 1/15/2020, 10:22:59 PM MQTT publish: topic 'homeassistant/sensor/0x0022a300001af3b6/power/config', payload '{"unit_of_measurement":"W","icon":"mdi:factory","value_template":"{{ value_json.power }}","state_topic":"zigbee2mqtt/0x0022a300001af3b6","json_attributes_topic":"zigbee2mqtt/0x0022a300001af3b6","name":"0x0022a300001af3b6_power","unique_id":"0x0022a300001af3b6_power_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0022a300001af3b6"],"name":"0x0022a300001af3b6","sw_version":"Zigbee2mqtt 1.6.0","model":"Plug-in smart switch (45853GE)","manufacturer":"GE"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 1/15/2020, 10:22:59 PM MQTT publish: topic 'homeassistant/sensor/0x0022a300001af3b6/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x0022a300001af3b6","json_attributes_topic":"zigbee2mqtt/0x0022a300001af3b6","name":"0x0022a300001af3b6_linkquality","unique_id":"0x0022a300001af3b6_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0022a300001af3b6"],"name":"0x0022a300001af3b6","sw_version":"Zigbee2mqtt 1.6.0","model":"Plug-in smart switch (45853GE)","manufacturer":"GE"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:debug 1/15/2020, 10:22:59 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":18,"stackVersion":80,"hwVersion":1}}' of device '45853' (0x0022a300001af3b6) of endpoint 1
  zigbee2mqtt:debug 1/15/2020, 10:22:59 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":18,"stackVersion":80,"hwVersion":1,"manufacturerName":"Jasco Products"}}' of device '45853' (0x0022a300001af3b6) of endpoint 1
[... cut ...] 259,"inClusterList":[0,3,2821],"outClusterList":[3,6],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":18,"stackVersion":80,"hwVersion":1,"manufacturerName":"Jasco Products","modelId":"45853","powerSource":1}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genOnOff":{"dir":{"value":2},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{"averageMacRetryPerApsMessageSent":1,"lastMessageLqi":255,"lastMessageRssi":-52}}}}},"_id":"2RrM1KVgho9i974n"}
```

Koenkk, 2020-01-16:
@djchen could you sniff the traffic when pairing the device on both zigbee2mqtt 1.6.0 and 1.9.0? I want to compare them: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

djchen, 2020-01-17:
> @djchen could you sniff the traffic when pairing the device on both zigbee2mqtt 1.6.0 and 1.9.0? I want to compare them: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

I don't have an extra CC sniffer at the moment but I just ordered one.

closed 2020-01-17 by djchen

Koenkk, 2020-01-17:
Great!