Koenkk/zigbee2mqtt #2693: Philips Hue Motion sensors disconnect from the network

ccorderod, 2020-01-08:
# Bug Report
## What happened
I have 10+ philips hue motion sensors paired to my network. I move them all from a philips hue bridge (along with some light bulbs) where them have been working for a while without any problem.
Some of the motion sensors disconnect from time to time from the network now; sometimes they reconnect without doing anything, sometimes not.
This happen even with a hue motion sensor very close to the coordinator.
If they don't, I have to force remove them from the network and pair them again.
The network support more than 100 devices and those are the only devices with this behaviour.
## What did you expect to happen
Once a hue motion sensor connect, it should keep connected.
## How to reproduce it (minimal and precise)
Difficult to reproduce, as the disconnections are random.
## Debug Info
philips hue motion sensors with latest firmware version from Philips
zigbee2mqtt version: 1.8.0 latest dev
CC253X firmware version: P-2 coordinator, latest firmware
2 CC2530+CC2591 on the network running latest routing firmware

Koenkk, 2020-01-08:
I'm having the same problem (https://github.com/Koenkk/zigbee2mqtt/issues/2518#issuecomment-569132387), currently waiting for the problem to manifest itself again, then I can start looking into it.

WouterJN, 2020-01-10:
Same issue here with Hue Switches and Motion Sensor.

papanirual, 2020-01-10:
Same issue here with Hue outdoor motion sensors. At the moment I am still unable to repair two of them. I removed them from the database of zigbee2mqtt and tried to repair them, but the only thing happening is that there is a line in the log that the device announced itself, without the usual pairing procedure.

Looking forward to any fix - many thanks in advance!

Koenkk, 2020-01-11:
Update from my side, right before it stops working it seems that the devices leaves the network:

```
debug 2020-01-11 10:30:40: Device 'veranda_occupancy_sensor' announced itself
warn  2020-01-11 10:30:40: Device '0x001788010644f56d' left the network
```

Then it joins again:
```
info  2020-01-11 10:30:40: Device 'veranda_occupancy_sensor' joined
info  2020-01-11 10:30:40: Starting interview of 'veranda_occupancy_sensor'
```

But this fails:
```
error 2020-01-11 10:32:01: Failed to interview 'veranda_occupancy_sensor', device has not successfully been paired
```

Going to dive further into it.

Question to all: what coordinators and coordinator firmware do you use?

ccorderod, 2020-01-11:
Coordinator: CC1352P_2 firmware: 20191106
Most of my hue motion sensors are able to join again, but leave randomly again.
@Koenkk : is there is any way to log with DEBUG=* npm start but only logging one sensor? My network is too big to run zb2m with Debug enabled as it create enormous log files :-(.

Koenkk, 2020-01-12:
Thats not possible, but setting it to `DEBUG=zigbee-herdsman:controller*` should give the required info.

hi1000, 2020-01-13:
CC253X firmware version: 20190608
Philips Hue (9290012607) firmware version: 6.1.0.18912 and 6.1.1.27575
Bugs: After reboot the coordinator, cannot receive Philips motion sensor's message

Log File:
2020-01-13.15-39-50/log.txt:1/13/2020, 3:41:50 PM - warn: Failed to configure r_MotionD1 (0x0017880104b78b01) ('Error: Timed out after 10000 ms') (attempt #1)
2020-01-13.15-39-50/log.txt:1/13/2020, 3:41:50 PM - warn: This can be ignored if the device is working properly

papanirual, 2020-01-13:
CC26X2R1_20191106, Coordinator firmware version: 
'{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'

zigbee2mqtt version 1.8.0-dev (commit #f04e01f)

sti0, 2020-01-17:
Did you have some Osram plugs in your network? In https://github.com/Koenkk/zigbee2mqtt/issues/1474 it's described that Osram plugs prevent a re-join of some devices (maily Hue motion sensors). By myself I got two Osram plugs in the network and Hue motion and Hue dimmer are randomly disconnected. After I removed the plugs from the network it seems to be stable.

GJT, 2020-01-18:
CC1352P-2 
Coordinator firmware version: '{"type":"zStack3x0","meta":
{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'

same problem, no Osram plugs or devices.

emilklindt, 2020-01-18:
I was experiencing this same problem with the Hue Dimmer Switches, as described [here](https://github.com/Koenkk/zigbee2mqtt/issues/2438#issuecomment-570949109).

A week ago I tried changing the channel of my coordinator to use Zigbee channel 25. After searching for interfering WiFi access points near my apartment, I found that the default channel was quite populated. 

So it seems to be an issue with interference, from my best guess. Can anyone verify this?

Koenkk, 2020-01-18:
Mine dropped off again today, pushed a possible fix in the latest dev branch, fingers crossed 🤞

peterforeman, 2020-01-25:
Running on latest (not -dev), saw this:

```
[...]
zigbee2mqtt:info  2020-01-24 18:18:40: MQTT publish: topic 'zigbee2mqtt-beneden/motion_tuin', payload '{"battery":96,"linkquality":33,"occupancy":false,"temperature":3.34,"illuminance":3242,"device":{"friendlyName":"motion_tuin","model":"9290019758","ieeeAddr":"0x001788010644892e","networkAddress":43045,"type":"EndDevice","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Battery","applicationVersion":2,"stackVersion":1,"zclVersion":1,"hardwareVersion":1,"dateCode":"20180828","softwareBuildID":"6.1.0.25261"}}'
zigbee2mqtt:info  2020-01-24 18:19:04: Device 'motion_tuin' joined
zigbee2mqtt:info  2020-01-24 18:19:04: MQTT publish: topic 'zigbee2mqtt-beneden/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"motion_tuin"}}'
zigbee2mqtt:info  2020-01-24 18:19:04: Starting interview of 'motion_tuin'
zigbee2mqtt:info  2020-01-24 18:19:04: MQTT publish: topic 'zigbee2mqtt-beneden/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"motion_tuin"}}'
zigbee2mqtt:error 2020-01-24 18:20:24: Failed to interview 'motion_tuin', device has not successfully been paired
zigbee2mqtt:info  2020-01-24 18:20:24: MQTT publish: topic 'zigbee2mqtt-beneden/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"motion_tuin"}}'
```

Updated to latest-dev now and will pair again. Let's see how long it will last this time ;-)

hi1000, 2020-01-29:
I setup three zigbee coordinators in the same big room. They are use the different channels and different keys but the same pan_id and ext_pan_id.

WouterJN, 2020-02-01:
> Mine dropped off again today, pushed a possible fix in the latest dev branch, fingers crossed 🤞

When will this be released in the main branch?

Koenkk, 2020-02-01:
@WouterJN in Zigbee2mqtt 1.10 scheduled for Wednesday/Thursday.

Koenkk, 2020-02-01:
Mine hasn't dropped of since 2 weeks so things seem to have improved.

pepp86, 2020-02-06:
I have reconnected my two philips hue motion sensor yesterday with latest dev and they are lost again during the evening.

ccorderod, 2020-02-21:
@Koenkk Hue motion sensors started to drop off the network again yesterday after upgrading to 1.10.0-dev yesterday. Unfortunatly I can't tell wich version I was running previously (it was dev, but I do not know the build #). Will restart z2m with DEBUG=zigbee-herdsman:controller* trying to catch a drop.

ccorderod, 2020-02-21:
@Koenkk Not sure if this will help you in any way, but here you go an extract:
`2020-02-21T08:20:30.487Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[x,x,x,x,x,x,x,x,x,x,x,x,x,x,x,x,],"panID":7100,"extendedPanID":[y,y,y,y,y,y,y,y,y,y],"channelList":[20]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}'
2020-02-21T08:20:31.884Z zigbee-herdsman:controller:log Disable joining
2020-02-21T08:21:57.627Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0017880104b4e792/2 1030(10, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true})
2020-02-21T08:21:57.671Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0017880104b4e792/2 1024(10, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true})
2020-02-21T10:29:48.443Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0017880104b4e792/2 1024(10, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true})
2020-02-21T10:29:51.724Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' joined
2020-02-21T10:29:51.725Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' accepted by handler
2020-02-21T10:29:51.725Z zigbee-herdsman:controller:log Not interviewing '0x0017880104b4e792', completed 'true', in progress 'false'
2020-02-21T10:29:51.736Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T10:29:58.155Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0017880104b4e792/2 1030(10, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true})
2020-02-21T10:34:05.249Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0017880104b4e792/2 1024(10, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true})
2020-02-21T10:34:08.874Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T10:39:04.785Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0017880104b4e792/2 1024(10, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true})
`
Please note this is only an extract. there are hundreds of records with DefaultResponse, as well as many device announcement records. The hue motion sensor also joined and was accepted a few more times than what I have included on the post.
Let me know if you need anything else on my side.
By the way, I have moved the hue motion sensors to a hue bridge but the one showed on the log in the meantime as my setup is mostly useless without them. Thank you.

ccorderod, 2020-02-21:
@Koenkk Apologies for so much post, but I just got a drop :-). Hue motion sensor keeps announcing itself, join, is accepted, not interviewed, join again ...
`2020-02-21T11:01:59.354Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0017880104b4e792/2 1030(10, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true})
2020-02-21T11:01:59.443Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0017880104b4e792/2 1024(10, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true})
2020-02-21T11:02:04.384Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0017880104b4e792/2 1024(10, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true})
2020-02-21T11:02:07.651Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:02:09.341Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0017880104b4e792/2 1024(10, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true})
2020-02-21T11:02:14.282Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0017880104b4e792/2 1024(10, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true})
2020-02-21T11:02:19.252Z zigbee-herdsman:controller:endpoint DefaultResponse 0x0017880104b4e792/2 1024(10, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true})
2020-02-21T11:02:22.505Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:03:22.900Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:04:22.306Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:05:56.998Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:05:56.999Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' joined
2020-02-21T11:05:57.000Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' accepted by handler
2020-02-21T11:05:57.000Z zigbee-herdsman:controller:log Not interviewing '0x0017880104b4e792', completed 'true', in progress 'false'
2020-02-21T11:07:01.082Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' joined
2020-02-21T11:07:01.082Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' accepted by handler
2020-02-21T11:07:01.082Z zigbee-herdsman:controller:log Not interviewing '0x0017880104b4e792', completed 'true', in progress 'false'
2020-02-21T11:07:01.089Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:17:34.210Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:22:00.076Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:27:21.481Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' joined
2020-02-21T11:27:21.481Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' accepted by handler
2020-02-21T11:27:21.482Z zigbee-herdsman:controller:log Not interviewing '0x0017880104b4e792', completed 'true', in progress 'false'
2020-02-21T11:27:21.483Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' joined
2020-02-21T11:27:21.483Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' accepted by handler
2020-02-21T11:27:21.483Z zigbee-herdsman:controller:log Not interviewing '0x0017880104b4e792', completed 'true', in progress 'false'
2020-02-21T11:27:21.484Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:35:47.820Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:37:30.182Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:50:35.146Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:52:46.654Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' joined
2020-02-21T11:52:46.655Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' accepted by handler
2020-02-21T11:52:46.655Z zigbee-herdsman:controller:log Not interviewing '0x0017880104b4e792', completed 'true', in progress 'false'
2020-02-21T11:52:47.392Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:53:09.056Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T11:58:37.810Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' joined
2020-02-21T11:58:37.810Z zigbee-herdsman:controller:log Device '0x0017880104b4e792' accepted by handler
2020-02-21T11:58:37.811Z zigbee-herdsman:controller:log Not interviewing '0x0017880104b4e792', completed 'true', in progress 'false'
2020-02-21T11:58:39.007Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
2020-02-21T12:03:54.281Z zigbee-herdsman:controller:log Device announce '0x0017880104b4e792'
`

GJT, 2020-02-21:
same for me after upgrade, when i try to reconfigure I get this:

> zigbee2mqtt:error 2020-02-21 19:12:32: Failed to configure 'zb_sensor_kueche', attempt 1 (Error: Bind 0x001788010213f153/2 genPowerCfg from '0x00124b001ca61201/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

Koenkk, 2020-02-23:
I've tweaked the route discovery a bit therefore it could be a one-time thing (and afterwards it should improve). 

Could you try repairing it once for now and monitor if it drops off again?

ccorderod, 2020-02-23:
Hi @Koenkk I upgraded to latest dev a few minutes ago. The motion sensor has been working just a few minutes and dropped again. The funny thing is come to live again from time to time, and drop again :-(.

Koenkk, 2020-02-24:
@ccorderod can you update to the latest dev branch and sniff the traffic when this happens? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Koenkk, 2020-02-26:
For CC26xR1 and CC1352-P2 users, I believe this is an issue with the firmware itself, I've asked TI for support: https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/883629

ccorderod, 2020-02-26:
Thank you @Koenkk. I use a cc1352-P2 coordinator. Let's see TI answer to your ticket. Thank you for the update.

Koenkk, 2020-02-28:
TI has provided a possible fix, which I'm going to test now. Firmware: https://drive.google.com/open?id=1Ls8H3R8n2nUxyh9fq-l7hdwdPWg-UD5h

GJT, 2020-03-02:
CC1352P_2 with CC1352P_2_20200229.zip firmware and zigbee2mqtt-1.11.0

Sadly still experiencing hue sensor drop outs.
Had 2 out 6 sensors go dark, one never came back and one after about 30-60 minutes two times.

I don't have a stick to sniff traffic, but got one ordered.

mf-social, 2020-03-09:
I upgraded to the new firmware 6 days ago as per my other issue (linked immediately above this post).

Everything has been perfect until a few hours ago.

My kitchen sensor went offline at 2008 hours, came back of its own accord 62 minutes later at 2110. The last reported link quality was 102 when it dropped off, which is slightly higher than average for that sensor. 

Then my bathroom sensor went offline at 2107. Link quality was 96 (which is bang on average for that sensor) when it dropped off. It is now 2346 hours here and the sensor is still offline. 

I'm noting the times specifically because you'll note that they're almost 'in order', kitchen goes off, then kitchen comes back and bathroom goes off within minutes of each other (could have actually been the exact same time and it's just the next update times). 

Hope this helps diagnose the issue. 

I'm off to bed now, if the bathroom one isn't back online in the morning I'll let you know whether a quick press of the button works or whether a full re-pair is required.

No obvious issues in the logs for any of the relevant time points, but I'm not currently logging debug or herdsman so :man_shrugging:

Koenkk, 2020-03-10:
Mine also dropped off/shows red led on motion detection, I've found 2 other issues:
- Updated firmware (small change): https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin 
- Zigbee2mqtt was sending the default responses to the wrong network address after the motion sensors changed it's network address. This causes the sensor to think it's not connected to the network anymore ending in a reconnect loop (and also causing the red lights). This was actually quite a big Zigbee2mqtt bug which was not only affecting this device (!!).

Changes are available in the latest zigbee2mqtt dev branch.

mf-social, 2020-03-10:
Just a quick update to say that I've had intermittent issues all day and out of sheer curiosity I unplugged the CC26X2R1 from the USB port, and then plugged it back in.

Everything seems to be working fine again, but I'm going to see how it goes through the evening. I've been taking screenshots of the sensor history to show how it's been behaving, which I'll upload when I'm happy (or not!) that they're behaving properly. 

Thinking out loud - Could this be some sort of memory leak in the on board memory of the device?

Koenkk, 2020-03-10:
@mf-social I don't think the issue is on the CC26x2R1 side, when replugging Zigbee2mqtt is also restarted which actually is a workaround for point 2 of https://github.com/Koenkk/zigbee2mqtt/issues/2693#issuecomment-597271715 (which I believe is the (biggest) cause of this issue).

mf-social, 2020-03-10:
I got distracted half way through typing that, and during my distraction you posted the above. So I guess that replugging the USB is just rediscovering the routes, which fixes the problem? 

I'm using the homeassistant addon so I'll have to wait for the fix to filter through. I'll update the firmware tomorrow though :slightly_smiling_face:

ccorderod, 2020-03-10:
Hi @Koenkk My hue motion sensors keep dropping. I am running z2m 1.11.0 (commit 31e5678). Using cc1352-p2 coordinator with firmware 20200229.
And I found a weird thing; let me try to explain:
- I have 2 ZB networks, both supported by a cc1352-p2 coordinator, running z2m and firmware as above. One network support 100 devices (with load of routers) and the other around 60 devices (with load of routers too).
- both have permit_join to false, one on channel 11 the other on channel 20, each with a different pan_id BUT the same network key.
- some of the motion when dropping connect to the other zb network (they have always been on the same network and never joined the other)
- Those are extracts of the devices.yaml:
`'0x0017880104b63528':
  friendly_name: HueMotionPP1
'0x0017880104b6d8d7':
  friendly_name: HueMotionBano
'0x0017880104b52717':
  friendly_name: HueMotionDormitorio
'0x0017880104b5a7c7':
  friendly_name: HueMotionEscalera2
`
`'0x0017880104b63528':
  friendly_name: '0x0017880104b63528'
'0x0017880104b6d8d7':
  friendly_name: '0x0017880104b6d8d7'
'0x0017880104b52717':
  friendly_name: '0x0017880104b52717'
'0x0017880104b5a7c7':
  friendly_name: '0x0017880104b5a7c7'
`
This is an extract of both database.db:
`{"id":96,"type":"EndDevice","ieeeAddr":"0x0017880104b52717","nwkAddr":43921,"manufId":4107,"manufName":"Philips","powerSource":"Battery","modelId":"SML001","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2128,"inClusterList":[0],"outClusterList":[0,3,4,6,8,768,5],"clusters":{"genBasic":{"attributes":{"modelId":"SML001","manufacturerName":"Philips","powerSource":3,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190219","swBuildId":"6.1.1.27575"}}},"binds":[]},"2":{"profId":260,"epId":2,"devId":263,"inClusterList":[0,1,3,1030,1024,1026],"outClusterList":[25],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}},"msOccupancySensing":{"attributes":{"occupancy":0}},"msTemperatureMeasurement":{"attributes":{"measuredValue":1992}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":7236}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca6049b","endpointID":1},{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca6049b","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca6049b","endpointID":1},{"cluster":1030,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca6049b","endpointID":1}]}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190219","swBuildId":"6.1.1.27575","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1583868099875}
{"id":99,"type":"EndDevice","ieeeAddr":"0x0017880104b5a7c7","nwkAddr":39971,"manufId":4107,"manufName":"Philips","powerSource":"Battery","modelId":"SML001","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2128,"inClusterList":[0],"outClusterList":[0,3,4,6,8,768,5],"clusters":{"genBasic":{"attributes":{"modelId":"SML001","manufacturerName":"Philips","powerSource":3,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190219","swBuildId":"6.1.1.27575"}}},"binds":[]},"2":{"profId":260,"epId":2,"devId":263,"inClusterList":[0,1,3,1030,1024,1026],"outClusterList":[25],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}},"msOccupancySensing":{"attributes":{"occupancy":0}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2363}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":0}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca6049b","endpointID":1},{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca6049b","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca6049b","endpointID":1},{"cluster":1030,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca6049b","endpointID":1}]}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190219","swBuildId":"6.1.1.27575","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1583867919456}`

Is this expected because of both networks having the same key? 
I understand the interview fail because join is disabled.
I do not understand why the devices try to join a network which is far away from their original one.

Anyway, there is still something wrong with firmware and/or z2m as devices should not drop from the network (2 of the motion sensors are very close to the coordinator).

Let mw knoe if you require more details. Thanks.
`{"id":52,"type":"EndDevice","ieeeAddr":"0x0017880104b52717","nwkAddr":47680,"manufId":4107,"manufName":"Philips","powerSource":"Battery","modelId":"SML001","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":2128,"inClusterList":[0],"outClusterList":[0,3,4,6,8,768,5],"clusters":{"genBasic":{"attributes":{"modelId":"SML001","manufacturerName":"Philips","powerSource":3,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190219"}}},"binds":[]},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190219","zclVersion":1,"interviewCompleted":false,"meta":{},"lastSeen":1583754171373}{"id":53,"ieeeAddr":"0x0017880104b5a7c7","nwkAddr":23854,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1583781361268}
`

ccorderod, 2020-03-10:
> Mine also dropped off/shows red led on motion detection, I've found 2 other issues:
>
> * Updated firmware (small change): [...]

@Koenkk Apologies. I miss previous posts. Will update firmware and will upgrade to dev.

Koenkk, 2020-03-10:
I've added another improvement in the latest dev branch so please also test with that.

pepp86, 2020-03-11:
I have updated the controller and z2m to latest dev release and powered my hue motion sensors again. Will test and update this issue if they drop again or my zigbee networks goes crazy again

ccorderod, 2020-03-11:
@Koenkk I installed latest dev, but it does not work for me (see #3110). Upgraded to firmware 20200306. Switched to master (1.11.0 commit e140cf5). My network is terribly unstable now. 
`error 2020-03-11 23:51:41: Failed to read state of 'LuzTorre1' after reconnect
error 2020-03-11 23:51:41: Failed to read state of 'LuzTorre2' after reconnect
error 2020-03-11 23:51:42: Publish 'set' 'state' to 'LuzTechoDespacho3' failed: 'Error: Command 0x00178801041557bf/11 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":22280,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":136,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,190,0]}}' failed with status '17' (expected '0'))'
error 2020-03-11 23:51:43: Failed to read state of 'LuzTechoPasilloPlanta1' after reconnect
error 2020-03-11 23:51:54: Publish 'set' 'state' to 'LuzMesaDespacho2' failed: 'Error: Command 0x0017880102c5ae2a/11 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'
error 2020-03-11 23:52:02: Publish 'set' 'state' to 'LuzTechoDespacho3' failed: 'Error: Command 0x00178801041557bf/11 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
error 2020-03-11 23:52:15: Publish 'set' 'state' to 'LuzTechoDespacho1' failed: 'Error: Command 0x0017880104c0fdcd/11 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 16020 - 11 - 196 - 6 - 11 after 6000ms)'
error 2020-03-11 23:52:16: Publish 'set' 'state' to 'LuzTorre2' failed: 'Error: Command 0x001788010230af80/11 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
error 2020-03-11 23:52:32: Publish 'set' 'state' to 'LuzTorre2' failed: 'Error: Command 0x001788010230af80/11 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
error 2020-03-11 23:53:01: Failed to ping 'LuzMesaDespacho2'
error 2020-03-11 23:53:02: Failed to ping 'LuzLavaboBanoJuan3'
error 2020-03-11 23:53:22: Failed to ping 'LuzLavaboBanoJuan1'
error 2020-03-11 23:53:26: Publish 'set' 'state' to 'LuzTorre2' failed: 'Error: Command 0x001788010230af80/11 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
error 2020-03-11 23:53:31: Failed to ping 'LuzRoperoDJ1'
error 2020-03-11 23:53:37: Failed to ping 'LuzTorre2'
error 2020-03-11 23:53:47: Failed to ping 'LuzTechoDespacho3'
error 2020-03-11 23:53:48: Failed to ping 'LuzTorre1'
error 2020-03-11 23:54:14: Publish 'set' 'state' to 'LuzTorre1' failed: 'Error: Command 0x001788010372c624/11 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
error 2020-03-11 23:54:17: Publish 'set' 'state' to 'LuzTechoBanoDespacho' failed: 'Error: Command 0x90fd9ffffe292303/1 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 54444 - 1 - 23 - 6 - 11 after 6000ms)'
error 2020-03-11 23:54:28: Publish 'set' 'state' to 'LuzTorre1' failed: 'Error: Command 0x001788010372c624/11 genOnOff.off({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
error 2020-03-11 23:54:36: Failed to ping 'LuzTechoBanoDespacho'
error 2020-03-11 23:54:53: Failed to ping 'LuzMesaDespacho1'
error 2020-03-11 23:54:56: Failed to ping 'LuzMesaMamaDP'
error 2020-03-11 23:55:29: Publish 'set' 'state' to 'LuzRoperoDespacho' failed: 'Error: Command 0x90fd9ffffe298e81/1 genOnOff.on({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'
error 2020-03-11 23:55:56: Failed to ping 'LuzTorre1'
error 2020-03-11 23:56:01: Publish 'set' 'state' to 'LuzTechoDespacho1' failed: 'Error: Command 0x0017880104c0fdcd/11 genOnOff.on({}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 16020 - 11 - 74 - 6 - 11 after 6000ms)'
error 2020-03-11 23:56:04: Failed to read state of 'LuzTechoDespacho3' after reconnect
error 2020-03-11 23:56:20: Failed to read state of 'LuzTorre2' after reconnect
error 2020-03-11 23:56:29: Failed to ping 'LuzDerLavaboBanoDespacho'
error 2020-03-11 23:57:24: Failed to ping 'LuzMesaDespacho2'
error 2020-03-11 23:57:27: Failed to read state of 'HuePlay2' after reconnect
error 2020-03-11 23:58:16: Failed to ping 'LuzRoperoDJ1'
error 2020-03-11 23:58:25: Failed to ping 'LuzTechoSaladeEstar'
error 2020-03-11 23:58:49: Publish 'set' 'state' to 'g-lucesvestidor' failed: 'Error: Command 2 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":0}) failed (Error: Data request failed with error: 'MAC channel access failure' (225))'
error 2020-03-11 23:59:36: Failed to ping 'HuePlay2'
error 2020-03-11 23:59:55: Failed to read state of 'LuzTorre1' after reconnect
[... cut ...]
(node:1049) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 91)
zigbee2mqtt:error 2020-03-12 00:06:55: Failed to ping 'LuzRoperoDespacho'
(node:1049) PromiseRejectionHandledWarning: Promise rejection was handled asynchronously (rejection id: 91)
`

trekker25, 2020-03-12:
This update will be in the 1.12.0 release? As my network is stable on the 1.7.0 version except the Hue motion giving me the red finger...

Koenkk, 2020-03-12:
@ccorderod when reflashing did you make sure that you executed step 5 of: https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html?

@trekker25 it will be part of 1.12.0

ccorderod, 2020-03-12:
Yes @Koenkk . I always follow the guide when flashing. All steps with no warnings. Do you think I should reflash?

Koenkk, 2020-03-12:
@ccorderod yes, lets try that, looks like some issue at the adapter side.

trekker25, 2020-03-12:
I assume it’s a routing problem. I have 3 hue motion. And 2 disconnect and are quit far away of usb module. And one never disconnects and is really short to usb module and has a direct connection.

ccorderod, 2020-03-12:
@Koenkk Just to be on the safe side, I downloaded the firmware again, then reflashed. Upgraded to 1.11.0-dev commit 97a4b6b. Network is stable again. Let's see if the hue motion sensors do not drop now. May I ask you how did you figure out it was an adapter issue? Txs.

Koenkk, 2020-03-12:
@ccorderod the errors shown were timeouts coming from the adapter (e.g. `Error: Timeout - 14048 - 1 - 114 - 0 - 1 after 6000ms`)

ccorderod, 2020-03-12:
Understood. Thank you @Koenkk

pepp86, 2020-03-16:
One of my hue motion sensor dropped again (and created a lot of trouble again in my network - did not had a chance to sniff the traffic as my lights where not working properly and my wife was not amused...) Looks as it is sending lots of data when it drops... Red light was flashing at motion.
The other one dropped, but came back.

So I am ending the hue motion sensor adventure now as it took my woman acceptance factor nearly to zero....

Koenkk, 2020-03-18:
> I've added another improvement in the latest dev branch so please also test with that.

Since this I've only got once temporary disconnect, it restored automatically after that and seems to works stable now.

Note that with Zigbee2mqtt 1.12.0-dev you can also OTA update this device (https://www.zigbee2mqtt.io/information/ota_updates.html), before the update mine was on `"dateCode":"20180828","swBuildId":"6.1.0.25261"` now on `{"softwareBuildID":"6.1.1.27575","dateCode":"20190219"}`.

pepp86, 2020-03-19:
I tested again with latest z2m dev release and latest CC2652 dev firmware.
Firmware Update of hue motion sensor went well (took 30-40 minutes) and after that the motion sensor was working for 10 minutes. After that, it was completely silence again (flashing red light when motion was detected, but only announcing itself - no data in z2m logs.

`debug 2020-03-19 07:38:57: Received Zigbee message from '0x0017880106f731c9', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":16244}' from endpoint 2 with groupID 0
info  2020-03-19 07:38:57: MQTT publish: topic 'zigbee2mqtt_DE/0x0017880106f731c9', payload '{"battery":43.5,"linkquality":9,"last_seen":"2020-03-19T07:38:57+01:00","occupancy":false,"temperature":22.24,"illuminance":16244,"illuminance_lux":41,"update_available":false,"elapsed":297957,"device":{"friendlyName":"0x0017880106f731c9","model":"9290012607","ieeeAddr":"0x0017880106f731c9","networkAddress":47498,"type":"EndDevice","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Battery","applicationVersion":2,"stackVersion":1,"zclVersion":1,"hardwareVersion":1,"dateCode":"20190219","softwareBuildID":"6.1.1.27575"}}'
debug 2020-03-19 07:39:25: Device '0x0017880106f731c9' announced itself
info  2020-03-19 07:39:25: MQTT publish: topic 'zigbee2mqtt_DE/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x0017880106f731c9"}}'
debug 2020-03-19 07:41:44: Device '0x0017880106f731c9' announced itself
info  2020-03-19 07:41:44: MQTT publish: topic 'zigbee2mqtt_DE/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x0017880106f731c9"}}'
debug 2020-03-19 07:42:53: Device '0x0017880106f731c9' announced itself
info  2020-03-19 07:42:53: MQTT publish: topic 'zigbee2mqtt_DE/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x0017880106f731c9"}}'
debug 2020-03-19 07:38:57: Received Zigbee message from '0x0017880106f731c9', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":16244}' from endpoint 2 with groupID 0
info  2020-03-19 07:38:57: MQTT publish: topic 'zigbee2mqtt_DE/0x0017880106f731c9', payload '{"battery":43.5,"linkquality":9,"last_seen":"2020-03-19T07:38:57+01:00","occupancy":false,"temperature":22.24,"illuminance":16244,"illuminance_lux":41,"update_available":false,"elapsed":297957,"device":{"friendlyName":"0x0017880106f731c9","model":"9290012607","ieeeAddr":"0x0017880106f731c9","networkAddress":47498,"type":"EndDevice","manufacturerID":4107,"manufacturerName":"Philips","powerSource":"Battery","applicationVersion":2,"stackVersion":1,"zclVersion":1,"hardwareVersion":1,"dateCode":"20190219","softwareBuildID":"6.1.1.27575"}}'
debug 2020-03-19 07:39:25: Device '0x0017880106f731c9' announced itself
info  2020-03-19 07:39:25: MQTT publish: topic 'zigbee2mqtt_DE/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x0017880106f731c9"}}'
debug 2020-03-19 07:41:44: Device '0x0017880106f731c9' announced itself
info  2020-03-19 07:41:44: MQTT publish: topic 'zigbee2mqtt_DE/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x0017880106f731c9"}}'
debug 2020-03-19 07:42:53: Device '0x0017880106f731c9' announced itself
info  2020-03-19 07:42:53: MQTT publish: topic 'zigbee2mqtt_DE/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x0017880106f731c9"}}'
`

Koenkk, 2020-03-19:
@pepp86 please sniff the traffic when this happens (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

pepp86, 2020-03-19:
Hi @Koenkk. I setup everything for sniffing now and I can see data from my network. At the moment I power up the hue motion sensor I can see a lot of more traffic happening (my network has more than 30 devices, so normal level is even a lot of communication). What I don't see is, what you exactly need to investigate further. I don't have a clue how to filter in wireshark for example.

Koenkk, 2020-03-19:
Don't worry about the filtering, if you have a sniff send it to me on telegram (@koenkk).

ccorderod, 2020-03-29:
@Koenkk Update: I upgraded both ZB networks to z2m 1.12.0-dev commit a7e38b6 as well as cc2652-p2 coordinators to firmware dev 20200312
Hue motion keep dropping :(. Sometimes they get back in again, sometimes they don't.
I also found a weird thing: my ZB networks have different channels and different panids, BUT the same network key. Sometimes, a sensor drop from ZB network 1 and move to ZB network 2 (permit_join is false always). So, I went to edit configuration.yaml files to ban ALL the addresses of the other ZB network, to avoid this weird effect.
Believe it or not, it does not work!. Sensors still move from one network to the other from time to time.
So, I need some advise now: I have in total 3 ZB networks, 2 z2m using cc2652-p2 as coordinators, and the third with a hue bridge. You can guess it is a pain to manage all this stuff even under HA.
In total, I should have 250+ devices (most of them are hue and ikea lights hence routers). Should I try to move ALL of them to only one ZB network? (please note this is going to be a painful task). Do you think z2m with cc2552-p2 will be able to handle such a massive network? Thank you

Koenkk, 2020-03-29:
250 devices is a lot, I have no clue tbh it has never been tested.

I would recommend to use a different network_key for every instance to avoid them hopping.

upwindanderl, 2020-03-29:
Same here! 
But I'm on a coordinator CC1352P-2
Stack3x0"evision":20191106

Is it enough to try the HomeAssistant edge version or do i need a newer one?

ccorderod, 2020-04-01:
@Koenkk I sent to your discord account a couple of pcapng files. Hopefuly they will help with the hue motion sensor drop issue.

ccorderod, 2020-04-01:
> 250 devices is a lot, I have no clue tbh it has never been tested.
> 
> I would recommend to use a different network_key for every instance to avoid them hopping.

But I do not understand why they hopp to the other network; the ban directive should prohibit that, right?

Koenkk, 2020-04-01:
@ccorderod zigbee has no ban mechanism, what we basically do is remove the device from the network when it connects, but this doesn't prevent the hop.

ccorderod, 2020-04-01:
shame on me ... you are absolutely right!.

[187 more replies]

closed 2023-02-16 by github-actions[bot]