Koenkk/zigbee2mqtt #2695: zigbee2mqtt stops communicating with my adapter

Ramblurr, 2020-01-08:
# Bug Report

## What happened

For about a month I had zigbee2mqtt (installed via HASS) working with socat. A centrally located raspi zero is connected to my CC2531 and it forwards the serial connection to zigbee2mqtt's container. Its an out of the box HASS setup + socat. It was working great.

All of sudden it stopped working around the new year

## What did you expect to happen

herdsman communicates with the device but then fails with a "SRSP - SYS - version after 6000ms" error. See the log file for more details with debugging:

https://pastebin.com/nChiUaaE

## How to reproduce it (minimal and precise)

I haven't deleted/reset my zigbee2mqtt config yet, because repairing is such a PITA. But I will if necessary. 

## Debug Info
zigbee2mqtt hass addon: 1.8.0 (**not** using edge)
CC253X firmware version: cc2531znp-prod.hex 20190608 master z-stack 1.2

shirou93, 2020-01-09:
i have the same issue :(

Koenkk, 2020-01-11:
The cc2531 probably crashed, I would recommend using the https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing as it improves stability.

shirou93, 2020-01-11:
Reflash firmware solve problem

Ramblurr, 2020-01-12:
@Koenkk  Sure I will give it a try. What does this firmware build do different?

shirou93, 2020-01-12:
In my opinion memory on adapter have crash and need reflash to work again.

PedroLamas, 2020-01-13:
@Koenkk will flashing a cc2531 from [default](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default) to [source_routing](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing) require repairing the devices?

I'm seeing this very same issue right now, I've had it in the past and only reflashing (always used "default") fixed the problem.

Off-topic, but I've ordered a CC26X2R1 (from Mouser, should arrive in 3 to 4 days), hopefully, this will be the end of these forced reflashes every ~6months or so...

Koenkk, 2020-01-13:
Nope! switching firmware doesn't require this (https://www.zigbee2mqtt.io/information/FAQ.html#what-does-and-does-not-require-repairing-of-all-devices)

closed 2020-03-14 by Koenkk

Ramblurr, 2020-04-05:
@Koenkk I just found the  time to flash my router.. I see now there is a [version 3](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator) of the firmware.. should I use it instead?

Koenkk, 2020-04-05:
No