Koenkk/zigbee2mqtt #2696: Device paired while "permit_join" is false (interference with Hue bridge?)

TilmanK, 2020-01-08:
# Bug Report
I bought a Philips Hue Bridge to do software updates on my Hue devices. I removed a Motion sensor from Zigbee2Mqtt and paired it with the bridge to get the update. When pairing with the bridge the sensor got paired with Z2M too. The sensor works fine with the Hue bridge. It just sent some messages and then stopped.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4036639/log.txt)

In detail:

**Startup yesterday evening, from the log (permit join is true):**
info  2020-01-07 20:28:28: Motion OG (0x0017880104b7cd06): 9290012607 - Philips Hue motion sensor (EndDevice)
warn  2020-01-07 20:28:28: `permit_join` set to  `true` in configuration.yaml.

**20 Minutes later Z2M Assistant switches off join permit (afterwards no change anymore):**
debug 2020-01-07 20:48:28: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'false'

**I removed the device here:**
debug 2020-01-08 18:40:03: Received MQTT message on 'zigbee2mqtt/bridge/config/remove' with data 'Motion OG'
info  2020-01-08 18:40:03: Removing 'Motion OG'
info  2020-01-08 18:40:08: Successfully removed Motion OG
info  2020-01-08 18:40:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"Motion OG"}'

**And after setting the device to pairing it joins here:**
Device '0x0017880104b7cd06' joined
info  2020-01-08 18:41:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0017880104b7cd06"}}'
info  2020-01-08 18:41:08: Starting interview of '0x0017880104b7cd06'
info  2020-01-08 18:41:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0017880104b7cd06"}}'
debug 2020-01-08 18:41:08: Received Zigbee message from '0x0017880104b7cd06', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":19470}' from endpoint 2 with groupID 0
warn  2020-01-08 18:41:08: Received message from unsupported device with Zigbee model 'undefined'
warn  2020-01-08 18:41:08: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
debug 2020-01-08 18:41:10: Received Zigbee message from '0x0017880104b7cd06', type 'readResponse', cluster 'genBasic', data '{"modelId":"SML001","manufacturerName":"Philips","powerSource":3}' from endpoint 1 with groupID 0
debug 2020-01-08 18:41:11: Received Zigbee message from '0x0017880104b7cd06', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"appVersion":2,"stackVersion":1}' from endpoint 1 with groupID 0
debug 2020-01-08 18:41:11: Received Zigbee message from '0x0017880104b7cd06', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20160630","swBuildId":"6.1.0.18912"}' from endpoint 1 with groupID 0
info  2020-01-08 18:41:11: Successfully interviewed '0x0017880104b7cd06', device has successfully been paired
info  2020-01-08 18:41:11: Device '0x0017880104b7cd06' is supported, identified as: Philips Hue motion sensor (9290012607)

**Z2M configures the sensor:**
info  2020-01-08 18:41:11: Configuring '0x0017880104b7cd06'
info  2020-01-08 18:41:15: Succesfully configured '0x0017880104b7cd06'

## Debug Info
zigbee2mqtt version: version 1.8.0-dev (commit #8a72916)
CC253X firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}'

Koenkk, 2020-01-09:
I think it rejoins because the motion sensor didn't forgot the network encryption key, probably not solvable from the zigbee2mqtt side, read more about it here: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

TilmanK, 2020-01-09:
Yes, I'm aware of the information you linked. I'm pretty sure, that I removed the device, then factory resetted it, so that it joins the Hue bridge. But as of now I'm not sure, if the reconnection didn't happen inbetween. I'll check this out and will reopen the ticket if I can safely reproduce this. Thanks!

closed 2020-01-09 by TilmanK