Koenkk/zigbee2mqtt #2698: zigbee2mqtt:error 2020-01-08 20:17:23: Failed to interview '0x00124b001d63e692', device has not successfully been paired

khaidakin, 2020-01-08:
# Bug Report
## What happened
Trying to pair DIYRuZ relay switch power meter (relay, switch, adc) https://www.zigbee2mqtt.io/devices/DYRuZ_rspm.html
## What did you expect to happen

## How to reproduce it (minimal and precise)
try to add DIYRuZ relay switch to zigbee2mqtt
## Debug Info
zigbee2mqtt version: 1.8.0-dev (commit #cabb1f3)
CC2538 firmware version: 20190523
https://pastebin.com/iL4Fyb78
Prior to this, the relay was sometimes connected to Z2M 1.7.1 with the coordinator CC2530.
Now 15 different devices are connected to my system and they connected without problems.

Koenkk, 2020-01-11:
Can you try pairing it a few more times? I would like to see if it always fails at the same point.

khaidakin, 2020-01-11:
Oh sure. The log of several attempts with small omissions
https://pastebin.com/cxgYk9Bj

Koenkk, 2020-01-13:
@khaidakin somehow the `controller` debug logging is missing, (did you run with `DEBUG=zigbee-herdsman:adapter:*`??), Can you try with `DEBUG=zigbee-herdsman:*`?

khaidakin, 2020-01-14:
Here are two logs: from the test machine and from the working one. On a test machine, there is only one device in the zigbee network and, with a long pause, the DIYRuZ relay switch is connected (https://pastebin.com/5gYxmsxT). On the working machine, 16 zigbee devices and the DIYRuZ relay switch does not connect (https://pastebin.com/MhtZPwdz).

Koenkk, 2020-01-16:
Is the "non-production" zigbee network completely down? (all devices and coordinator powered off), otherwise it might reconnect as it was previously paired to it.

khaidakin, 2020-01-16:
Yes. The "non-production" zigbee network is completely down.

Koenkk, 2020-01-16:
Can you try replacing `node_modules/zigbee-herdsman/dist/controller/model/device.js` with https://gist.github.com/Koenkk/a45bdd9da8f8df31b10490663dae78ab and pair again? 

I'm still missing the `controller` debug logging btw, make sure to read https://github.com/Koenkk/zigbee2mqtt/issues/2698#issuecomment-573846764

khaidakin, 2020-01-17:
I changed device.js and I started with debugging (https://pastebin.com/CVUNmzWR). The result is the same: Failed to interview '0x00124b001d63e692', device has not successfully been paired

khaidakin, 2020-01-17:
And also the interview still fails HEIMAN HS1RC-M. Here is the debugging result for it: https://pastebin.com/zyWZNsZ4

Koenkk, 2020-01-17:
Thanks, can you try with this in `devices.js` and try again? https://gist.github.com/Koenkk/e7e97dd9e9a8ff5e69fb82d2197a2b0a

khaidakin, 2020-01-18:
On the advice of @kirovilya, I switched to  version 1.9.0-dev (commit #ae5330d) and made your corrections to the file devices.js. After updating the version z2m my zigbee network fell apart but then the interview began on the DIYRuZ relay switch and it successfully connected to the network (https://pastebin.com/3hFiU3Wb). I tested this on a network with one device and with all my zigbee devices. The result is positive. I want to test my network for a couple of days and try connecting the DIYRuZ relay switch again and then I will finally give an answer.

khaidakin, 2020-01-19:
Everything works stably. With the DIYRuZ relay switch, the issue is closed.

closed 2020-01-19 by khaidakin