Koenkk/zigbee2mqtt #2699: Failed to call 'BridgeConfig' 'onMQTTMessage' on premit_join: false toggle

jffz, 2020-01-08:
# Bug Report
## What happened
```
zigbee2mqtt:error 2020-01-08 21:52:25: Failed to call 'BridgeConfig' 'onMQTTMessage' (Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10))
```
This happen everytime `permit_join` is toggled back to false.

## What did you expect to happen
n/a

## How to reproduce it (minimal and precise)
Toggle `permit_join` to `true` then toggle it back to false

## Debug Info
zigbee2mqtt version: 1.8.0-dev
CC253X firmware version: 20190608

jffz, 2020-01-08:
Issue seems to happen either on trigger or when I try to pair a device. Not sure exactly.
Tried on latest raspbian with a pi3 and hassio.

Koenkk, 2020-01-11:
Can you try with the following firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

closed 2020-03-11 by Koenkk