Koenkk/zigbee2mqtt #2700: Not able to pair Xiaomi aqara devices

jffz, 2020-01-08:
# Bug Report
## What happened
Tried to pair multiple aqara devices and literaly nothing happend.
- RTCGQ11LM
- MCCGQ11LM

I do have a Philips Hue bridge next to me CC2531. Tried to power it off but still no success

## What did you expect to happen
Get messages of pairing in zigbee2mqtt topic.

## How to reproduce it (minimal and precise)
Try pairing xiaomi aqara devices following official doc

## Debug Info
zigbee2mqtt version: 1.8.0-dev
CC253X firmware version: 20190608

emilss, 2020-01-09:
I currently have 3 Aqara devices - MCCGQ11LM, WXKG03LM and ZNLDP12LM. All 3 of them work fine and I had no problems with pairing. I can't vouch for RTCGQ11LM since I don't own it but I assume it should work as well. Perhaps you have disabled pairing of new devices?

jffz, 2020-01-09:
Pairing is enabled, I see the messages to toggle it on/off sent to the mqtt topic but I can achieve to pair any device. My issue might be related with #2699 not sure.

stuartlt, 2020-01-10:
To pair Xiaomi devices I have found you need to hold down the pair button, then within the first 10 seconds while its pairing keep clicking the pair button at say 1 second intervals for a short pause (just a fraction of a second). This worked reliably for me.

closed 2020-03-18 by stale[bot]