Koenkk/zigbee2mqtt #2701: Pairing of Osram Lightify not possible

janwilke, 2020-01-08:
# Bug Report
## What happened
I changed my hardware from raspberry to ubuntu system and updated my CC2531 to the latest version (CC2531_SOURCE_ROUTING_20190619) and started to connect all devices again.
Philips Hue and Xiaomi Temp/Door worked fine, but the OSRAM Lightfy Lamps will not work.

I reset them by switching them on/off for 5 seconds and they confirmed reset by blinking.
Then I set "permit_join" to true and get the feedback, that they were connected - but I can't control them.

mosquitto_pub -m "OFF" -t 'zigbee2mqtt/0x841826000000df6e/set/state' 
This kind of command is working fine for all HUE devices, but OSRAM does not work.

After several restarts of zigbee service I get now the following error in the logfile
Logfile: 

Jan 08 23:00:12 openhab2 npm[5735]: zigbee2mqtt:error 2020-01-08 23:00:12: Entity '0x841826000000db71' is unknown
Jan 08 23:00:15 openhab2 npm[5735]: zigbee2mqtt:warn  2020-01-08 23:00:15: Device with modelID 'undefined' is not supported.
Jan 08 23:00:15 openhab2 npm[5735]: zigbee2mqtt:warn  2020-01-08 23:00:15: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Can you help me?

## What did you expect to happen

Osram can be again connected to Zigbee adapter and controlled by MQTT.

## How to reproduce it (minimal and precise)

see ## What happened

## Debug Info
zigbee2mqtt version: zigbee2mqtt@1.8.0
CC253X firmware version: CC2531_SOURCE_ROUTING_20190619
node version: v12.14.0

Koenkk, 2020-01-11:
Please try with the latest dev branch, this should already contain a fix for this problem, after this first remove the device and repair it. 
 - https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove
- https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

janwilke, 2020-01-11:
Ok, will try it. Thanks!

closed 2020-03-11 by Koenkk