Koenkk/zigbee2mqtt #2702: Unsupported device: MiJia Light intensity sensor (GZCGQ01LM)

louismulder, 2020-01-08:
Hi Did anyone tried to insert the Xiaomi device MiJia Light intensity sensor (GZCGQ01LM)
see [](https://nl.aliexpress.com/item/4000476379615.html?spm=a2g0s.9042311.0.0.776f4c4drRn2bj)

Thx Louis

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Koenkk, 2020-01-09:
Nice that it is finally available, bought it and will add support for it once I receive it.

rezmus, 2020-01-09:
@koenkk i have it and zboss sniffer. what do you need?

Koenkk, 2020-01-09:
@rezmus first try following https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

rezmus, 2020-01-09:
i've added light sensor to devices.js, but i don't see any illuminance reports (saw a lot when sniffing between sensor and hub). do i miss something here?

info  2020-01-09 21:43:50: Starting zigbee2mqtt version 1.8.0 (commit #unknown)
info  2020-01-09 21:43:50: Starting zigbee-herdsman...
info  2020-01-09 21:43:51: zigbee-herdsman started
info  2020-01-09 21:43:51: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190223}}'
info  2020-01-09 21:43:51: Currently 1 devices are joined:
info  2020-01-09 21:43:51: 0x04cf8cdf3c77xxxx (0x04cf8cdf3c77xxxx): GZCGQ01LM - Xiaomi Mi Light Detection Sensor (EndDevice)
warn  2020-01-09 21:43:51: `permit_join` set to  `true` in configuration.yaml.
warn  2020-01-09 21:43:51: Allowing new devices to join.
warn  2020-01-09 21:43:51: Set `permit_join` to `false` once you joined all devices.
info  2020-01-09 21:43:51: Zigbee: allowing new devices to join.
info  2020-01-09 21:43:51: Connecting to MQTT server at mqtt://localhost
info  2020-01-09 21:43:51: Connected to MQTT server
info  2020-01-09 21:43:51: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-01-09 21:43:51: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190223}},"log_level":"info","permit_join":true}'
warn  2020-01-09 21:45:22: No converter available for 'GZCGQ01LM' with cluster 'genBasic' and type 'attributeReport' and data '{"modelId":"lumi.sen_ill.mgl01"}'
info  2020-01-09 21:45:49: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2020-01-09 21:45:49: Disconnecting from MQTT server
info  2020-01-09 21:45:49: zigbee-herdsman stopped

Koenkk, 2020-01-09:
Can you share the wireshark file?

rezmus, 2020-01-09:
sure https://ufile.io/9q7opqqx

rezmus, 2020-01-11:
@Koenkk any idea why i don't get illuminance reports? btw: i have both aqara/mi eu plug. made a log for poweroff_memory (to keep power state). they use same attributes for both plugs. how can i test it from cli (i don't have HA, etc). if i add it to toZigbee?

Koenkk, 2020-01-13:
Device will be supported in a few hours in the latest dev branch, thanks to @kirovilya

closed 2020-01-13 by Koenkk

thehijacker, 2020-01-15:
Hello. I ordered same device and it is on its way. Just starting with Zigbee network and only yesterday I had set it up by flashing your firmware to stock CC2530 (Coordinator v1.2). Currently I paired two devices (Xiaom water sensor and Xiaomi Door/Window sensor). Both are working perfectly in Home Assistant. Amazing!

Now I discovered that this light sensor (GZCGQ01LM) uses Zigbee 3.0 protocol and it was just now supported by Zigbee2mqtt.

Must I flash my CC2530 with v3.0 coordinator firmware in order for the new sensor to join?

Zigbee2mqtt software is installed using the Virtual Environment guide (https://www.zigbee2mqtt.io/information/virtual_environment.html). If I understand correctly I need dev build in order to have the new devices detected and full functional? What changes in the instruction to get the dev build instead of production?

I only have two devices now but more are on the way I want to make sure I will not need to repair them all when the new light sensor arrives.

On the way is also the CC2530 + CC2591 module that I plan to flash with router firmware (CC2530_CC2591_router_2019_02.zip) to extend the coverage. Is this Zigbee 3.0 compatible and light sensor will be able to join it?

Thank you for answering. Realy looking forward to having Zigbee devices at home.

kirovilya, 2020-01-15:
@thehijacker you can stay with v1.2
https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

thehijacker, 2020-01-15:
Good to know that the newly added light sensor will also work with v1.2 firmware 👍.

I will keep v1.2 on my coordinator and also flash additional CC2530_CC2591 with available router v1.2 firmware.

I must have missed the DEV branch guide. Thank you for the link. Should I use it already or wait a bit that it will get merged with master. My only aim so far is that this new Xiaomi light sensor will be correctly detected.

Koenkk, 2020-01-15:
@thehijacker zigbee2mqtt 1.9 containing support for this device will be released in a few days (so you probably want to wait for that).

thehijacker, 2020-01-22:
Received it today. Upgrade my zigbee2mqtt to version 1.9.0 and it was found perfectly.

jan 22 15:57:29 ubuntuold bash[8650]: zigbee2mqtt:info  2020-01-22 15:57:29: MQTT publish: topic 'zigbee2mqtt/Light Sensor Aquarium', payload '{"battery":"100.00","voltage":3200,"linkquality":31,"illuminance":13618}'

Illuminance value is high. Anyone has some margin values? Like from 1 to 10000 is very low light etc.?

Erelen-Laiquendi, 2020-01-22:
Same high values too.

I think, it must be divided by 1000. But I don't have compatible Xiaomi Gateway to check it.

louismulder, 2020-01-23:
Ho All

See here the values:

[https://en.wikipedia.org/wiki/Lux](url)

Illuminance (lux) | Surfaces illuminated by
-- | --
0.0001 | Moonless, overcast night sky (starlight)[4]
0.002 | Moonless clear night sky with airglow[4]
0.05–0.3 | Full moon on a clear night[5]
3.4 | Dark limit of civil twilight under a clear sky[6]
20–50 | Public areas with dark surroundings[7]
50 | Family living room lights (Australia, 1998)[8]
80 | Office building hallway/toilet lighting[9][10]
100 | Very dark overcast day[4]
150 | Train station platforms[11]
320–500 | Office lighting[8][12][13][14]
400 | Sunrise or sunset on a clear day.
1000 | Overcast day;[4] typical TV studio lighting
10,000–25,000 | Full daylight (not direct sun)[4]
32,000–100,000 | Direct sunlight

kirovilya, 2020-01-23:
Ok, let's divide by 1000

louismulder, 2020-01-23:
> Ok, let's divide by 1000

No, Don't do it the given values are correct

Erelen-Laiquendi, 2020-01-23:
Lowest non-zero value in my DB - 3011. So, darkest state in my bedroom (with one table lamp at minimum brightness) is lighter than "Overcast day / typical TV studio lighting".

Today, with overcast weather and snowfall, in center of my room (3 meters from windows) value 17709. "Full daylight (not direct sun)"

That's really correct?

thehijacker, 2020-01-23:
Everytime I restart zigbee2mqtt it takes time for this sensor to initialize. Here is the log:

> jan 23 20:30:04 ubuntuold bash[25580]: zigbee2mqtt:info  2020-01-23 20:30:04: MQTT publish: topic 'zigbee2mqtt/Light Sensor Aquarium', payload '{"battery":"100.00","voltage":3200,"linkquality":49,"illuminance":3011}'
> jan 23 20:30:09 ubuntuold bash[25580]: zigbee2mqtt:info  2020-01-23 20:30:09: MQTT publish: topic 'zigbee2mqtt/Light Sensor Aquarium', payload '{"battery":"100.00","voltage":3200,"linkquality":49,"illuminance":4772}'
> jan 23 20:30:11 ubuntuold bash[25580]: zigbee2mqtt:error 2020-01-23 20:30:11: Failed to configure 'Light Sensor Aquarium', attempt 1 (Error: AREQ - ZDO -
> bindRsp after 10000ms
> jan 23 20:30:11 ubuntuold bash[25580]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
> jan 23 20:30:11 ubuntuold bash[25580]:     at ontimeout (timers.js:436:11)
> jan 23 20:30:11 ubuntuold bash[25580]:     at tryOnTimeout (timers.js:300:5)
> jan 23 20:30:11 ubuntuold bash[25580]:     at listOnTimeout (timers.js:263:5)
> jan 23 20:30:11 ubuntuold bash[25580]:     at Timer.processTimers (timers.js:223:10))
> jan 23 20:30:24 ubuntuold bash[25580]: zigbee2mqtt:info  2020-01-23 20:30:24: MQTT publish: topic 'zigbee2mqtt/Light Sensor Aquarium', payload '{"battery":"100.00","voltage":3200,"linkquality":52,"illuminance":3011}'
> jan 23 20:30:24 ubuntuold bash[25580]: zigbee2mqtt:info  2020-01-23 20:30:24: Configuring 'Light Sensor Aquarium'
> jan 23 20:30:34 ubuntuold bash[25580]: zigbee2mqtt:info  2020-01-23 20:30:34: MQTT publish: topic 'zigbee2mqtt/Light Sensor Aquarium', payload '{"battery":"100.00","voltage":3200,"linkquality":52,"illuminance":4772}'
> jan 23 20:30:34 ubuntuold bash[25580]: zigbee2mqtt:error 2020-01-23 20:30:34: Failed to configure 'Light Sensor Aquarium', attempt 2 (Error: AREQ - ZDO -
> bindRsp after 10000ms
> jan 23 20:30:34 ubuntuold bash[25580]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
> jan 23 20:30:34 ubuntuold bash[25580]:     at ontimeout (timers.js:436:11)
> jan 23 20:30:34 ubuntuold bash[25580]:     at tryOnTimeout (timers.js:300:5)
> jan 23 20:30:34 ubuntuold bash[25580]:     at listOnTimeout (timers.js:263:5)
> jan 23 20:30:34 ubuntuold bash[25580]:     at Timer.processTimers (timers.js:223:10))

It works OK after few minutes. Anyone with simillar issue?

As for lux value. I am used to the one that I get from Xiaomi Motion Sensor that can also report illuminance. If value is below 25 and motion is detected I will turn on the light.

Not sure what values to catch with this sensor.

louismulder, 2020-01-23:
@thehijacker See #2786 Timeout error during pairing catching with a nice error message?
[https://github.com/Koenkk/zigbee2mqtt/issues/2786](url)
Tomorrow (24-1-2010) I will try to create a zigbee-herdsman debug log

thehijacker, 2020-01-23:
Just made the asked log files.

They can be fetched from here:

https://pisarna.t-2.com/s/NFQ9MjkRTJyewN4

1. log.txt - first start using DEBUG=zigbee-herdsman* npm start, I waited a while and then added second light sensor that I have never paired.

2. log2.txt - second start using DEBUG=zigbee-herdsman* npm start, I waited a while before killing process

3. SSH log. I think more data can be see there.

Maybe I made a mistake and also need to enable debug in configuration.yaml?

Any issues posting my log here?

louismulder, 2020-01-23:
@thehijacker 
This issue #2702 is already closed. I don't know if Koen is still looking in this issue,so my advice is to put your comments/logs in issue  #2786

louismulder, 2020-01-23:
> Lowest non-zero value in my DB - 3011. So, darkest state in my bedroom (with one table lamp at minimum brightness) is lighter than "Overcast day / typical TV studio lighting".
> 
> Today, with overcast weather and snowfall, in center of my room (3 meters from windows) value is 17709. "Full daylight (not direct sun)"
> 
> @louismulder, that's really correct?

I think dividing by 100 is more realistic

Erelen-Laiquendi, 2020-01-23:
Guessing - it's bad way.
True way - compare with value in Xiaomi Mi Home app.
But we need somebody with new Xiaomi Gateway (Zigbee 3.0) :)

rezmus, 2020-01-23:
this is from wireshark with sniffer

Measured Value: 3011 (=1,000322 [lx])
Measured Value: 11761 (=14,000302 [lx])
Measured Value: 13980 (=24,003454 [lx])
Measured Value: 15564 (=35,008083 [lx])
Measured Value: 21493 (=140,026264 [lx])
Measured Value: 17994 (=62,008625 [lx])
Measured Value: 15911 (=38,003178 [lx])
Measured Value: 16628 (=45,004467 [lx])
Measured Value: 17404 (=54,004725 [lx])
Measured Value: 9031 (=7,000184 [lx])
Measured Value: 13011 (=19,003224 [lx])

Erelen-Laiquendi, 2020-01-24:
Look like lx ≈ e^(value * 0.00023985 - 0.18775) :)
But real way - look how Wireshark do it.

Erelen-Laiquendi, 2020-01-27:
Exactly right formula (from Wireshark sources): lx = 10^(value/10000)-1.

Gertjuh, 2020-02-05:
Can someone give me the right value template to use?
This doesn't work: value_template: "{{ (10* e**((value_json.illuminance)/1000)-1)}}"

Erelen-Laiquendi, 2020-02-05:
Right formula and template for template sensor:
`{{ (10**(states('sensor.kitchen_illuminance_raw') | int / 10000) - 1) | round }}`
for mqtt sensor something like this:
`{{ (10**(value_json.illuminance | int / 10000) - 1) | round }}`

Gertjuh, 2020-02-05:
Thanks!

This is my code in 'sensors.yaml'
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/lichtsensor_1"
    availability_topic: "zigbee2mqtt/bridge/state"
    name: Lichtsensor lux
    unit_of_measurement: "lx"
    device_class: "illuminance"
#    value_template: "{{ (value_json.illuminance) }}"  - This is working but gives a wrong result
   value_template: {{ (10**(value_json.illuminance | int / 10000) - 1) | round }}

Now is have this error:
Error loading /config/configuration.yaml: invalid key: "OrderedDict([('(10**(value_json.illuminance | int / 10000) - 1) | round', None)])"   in "/config/sensor.yaml", line 498, column 0

Gertjuh, 2020-02-05:
Thanks for you help!
This is the code I use with Zigbee2mqtt:

# Mijia lichtsensor (GZCGQ01LM)
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/lichtsensor_1"
    availability_topic: "zigbee2mqtt/bridge/state"
    name: Lichtsensor lux
    unit_of_measurement: "lx"
    device_class: "illuminance"
    value_template: "{{ (10**(value_json.illuminance | int / 10000) - 1) | round }}"

Is still gives a value of ~15000 lx. Stange.

Erelen-Laiquendi, 2020-02-05:
Kitchen Illuminance Raw - value from current version of z2m

Kitchen Illuminance:
```
  - platform: template
    sensors:
      kitchen_illuminance:
        friendly_name: Kitchen Illuminance
        unit_of_measurement: lx
        value_template: "{{ (10**(states('sensor.kitchen_illuminance_raw') | int / 10000) - 1) | round }}"
        device_class: illuminance
```

Lichtsensor lux (your code with my state_topic):
```
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/Kitchen Illuminance"
    availability_topic: "zigbee2mqtt/bridge/state"
    name: Lichtsensor lux
    unit_of_measurement: "lx"
    device_class: "illuminance"
    value_template: "{{ (10**(value_json.illuminance | int / 10000) - 1) | round }}"
```

Result:

[image]

(yeah, it's shiny today :)

Gertjuh, 2020-02-05:
I really appreciate your help!
I still do not understand why I have different results. If the raw value of the sensor is 32714 then the calculated value is 1884. What am I doing wrong?

Erelen-Laiquendi, 2020-02-05:
For 32714 "raw zigbee" value, real illuminance value is 1867 lx.

lx = 10^(value/10000)-1

32'714 / 10'000 = 3.2714
10 ^ 3.2714 ≈ 1868.09948
1868.09948 - 1 = 1867.09948
round(1867.09948) = 1867

Gertjuh, 2020-02-05:
My bad. I extracted 1 first. So it is working correctly now. Thanks!!

nphil, 2020-02-06:
Anyone know what the reporting interval/threshold on these are like?

Gertjuh, 2020-02-06:
I have it working for a couple of days now and at daylight it reports every 10-15 seconds or so. When the value stays 0 at night it reports every 5 minutes.

BrianDurward, 2020-02-26:
I received a mijia GZCGQ01LM yesterday and attempted to pair it.  The journal output :"0x04cf8cdf3c77239d","type":"EndDevice","networkAddress":7964,"model":"GZCGQ01LM","friendly_name":"lightsensor","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sen_ill.mgl01","lastSeen":1582733992092}]}'
Suggests it is paired but i'm not not receiving data (in node-red).  I am using a Raspberry Pi with a CC2531 sniffer. This is new territory for me!

Gertjuh, 2020-02-26:
Hi @BrianDurward,

I have the same setup. It took me quite some time to pair the light sensor. After several days of failures I deleted all entries to this device in the database and in HA. Then I paired it again near to the raspberry pi and it worked.

Look in your database.db in /share/zigbee2mqtt
You have to see an entry like this:

{"id":46,"type":"EndDevice","ieeeAddr":"0x04cf8cdf3c77614a","nwkAddr":22617,"manufId":4718,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sen_ill.mgl01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[0,1024,3,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.sen_ill.mgl01","manufacturerName":"LUMI","powerSource":3,"zclVersion":3,"appVersion":21,"stackVersion":2,"hwVersion":1,"dateCode":"20191118","swBuildId":"2019\u0000www."}},"genPowerCfg":{"attributes":{"batteryVoltage":30}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":33097}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e3205e","endpointID":1},{"cluster":1024,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e3205e","endpointID":1}]}},"appVersion":21,"stackVersion":2,"hwVersion":1,"dateCode":"20191118","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1580835714278}

If it has other details in it, then delete the entry and all references to the device you can find. Then try again.
Good luck!

BrianDurward, 2020-02-28:
@Gertjuh many thanks.  Now working.  Not sure what happened but I had to redo the firmware on the CC2531 and that sorted things.

iontichy123, 2020-02-28:
Had no luck to pair this device (version used 1.11 commit 31e5678)

Always got this:

error 2020-02-28 15:33:35: Failed to configure '0xd0cf5efffed9817d', attempt 1 (Error: Bind 0xd0cf5efffed9817d/1 genLevelCtrl from '0x00124b0019366fa4/1' failed (Error: AREQ - ZDO - bindRsp                                                                                                                                 after 10000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

Any suggestion?

iontichy123, 2020-03-02:
The error message from my last post is caused by another device and is not related with the light sensor. There seems to be no entry in the logs, when I try to pair the light sensor. 
If I press the button for 6 seconds, the sensor blinks 3 times, then I release the button. Now it blinks 1 time short every few seconds and 1 time long at the end.

permit_join is set to true.

BrianDurward, 2020-03-02:
I had to reflash my CC2531 and reinstall zigbee2mqtt.  Not sure what went
wrong but my CC2531 stopped working and became corrupted during the pairing
process.
After the reinstall all sensors including the light sensor work really
well.  After each individual sensor was paired (standing near the CC2531
and with all other sensors moved to a separate room) I had to reboot each
time. I hope you manage to get it working.

iontichy123, 2020-03-06:
@BrianDurward 
Thank you! I've used a newly flashed CC2531, now it works. Doesn't have to reinstall zigbee2mqtt.

foobar26, 2020-04-16:
Hi all!
I have an issue with pairing this device to a CC2531. I already successfully paired other Mija devices, like  Xiaomi MiJia temperature & humidity sensor or  Xiaomi Aqara human body movement and illuminance sensor.
But after pressing the pairing button for 6 seconds the device starts to flash the LED but nothing appears in the log of zigbee2mqtt.
permit_join is set to true.

What could I do to analyze the issue?

foobar26, 2020-04-16:
Never mind, i solved it with these hints:
https://github.com/Koenkk/zigbee2mqtt/issues/699#issuecomment-451106523

juliannideltorri, 2021-01-16:
Hi,
I can't get any lux value. This sensor is available with electrolama zzh (firmware CC26X2R1_*.zip)?