Koenkk/zigbee2mqtt #2704: Cannot pair IKEA remote control (E1524/E1810) due to error "Interview failed because can not get active endpoints"

dmak, 2020-01-09:
When trying to pair [IKEA remote control (E1524/E1810)](https://www.zigbee2mqtt.io/devices/E1524_E1810.html) the following error is produced:

## Log
```
2020-01-09T00:40:43.878Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":38336,"extaddr":"0xccccccfffea15dc7","parentaddr":0}
2020-01-09T00:40:43.878Z zigbee-herdsman:controller:log Device '0xccccccfffea15dc7' joined
2020-01-09T00:40:43.878Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-09T00:40:43.878Z zigbee-herdsman:controller:log Device '0xccccccfffea15dc7' accepted by handler
2020-01-09T00:40:43.878Z zigbee-herdsman:controller:log Interview '0xccccccfffea15dc7' start
zigbee2mqtt:info  2020-01-09 01:40:43: Starting interview of 'remote_control'
winston:file written true false
zigbee2mqtt:info  2020-01-09 01:40:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"remote_control"}}'
winston:file written true false
2020-01-09T00:40:43.880Z zigbee-herdsman:controller:device Interview - start device '0xccccccfffea15dc7'
2020-01-09T00:40:43.880Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":38336,"nwkaddrofinterest":38336}
2020-01-09T00:40:43.880Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,192,149,192,149,35]
winston:file logged 1553166 info  2020-01-09 01:40:43: Starting interview of 'remote_control'
winston:file logged 1553348 info  2020-01-09 01:40:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"remote_control"}}'
2020-01-09T00:40:43.888Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-01-09T00:40:43.888Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-01-09T00:40:43.888Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-01-09T00:40:43.888Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-01-09T00:40:43.888Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-09T00:40:53.882Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0xccccccfffea15dc7', attempt 1
2020-01-09T00:40:53.882Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":38336,"nwkaddrofinterest":38336}
2020-01-09T00:40:53.882Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,192,149,192,149,35]
2020-01-09T00:40:53.891Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-01-09T00:40:53.891Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-01-09T00:40:53.891Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-01-09T00:40:53.891Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-01-09T00:40:53.891Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-09T00:41:03.884Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0xccccccfffea15dc7', attempt 2
2020-01-09T00:41:03.885Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":38336,"nwkaddrofinterest":38336}
2020-01-09T00:41:03.885Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,192,149,192,149,35]
2020-01-09T00:41:03.893Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-01-09T00:41:03.894Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-01-09T00:41:03.894Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-01-09T00:41:03.894Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-01-09T00:41:03.894Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-09T00:41:13.888Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0xccccccfffea15dc7', attempt 3
2020-01-09T00:41:13.888Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":38336,"nwkaddrofinterest":38336}
2020-01-09T00:41:13.888Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,192,149,192,149,35]
2020-01-09T00:41:13.896Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-01-09T00:41:13.897Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-01-09T00:41:13.897Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-01-09T00:41:13.897Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-01-09T00:41:13.897Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-09T00:41:20.613Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,2,4,252,197,1,1,0,55,0,195,152,214,0,0,8,24,10,10,0,0,41,172,9,252,197,29,228]
2020-01-09T00:41:20.613Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,2,4,252,197,1,1,0,55,0,195,152,214,0,0,8,24,10,10,0,0,41,172,9,252,197,29,228]
2020-01-09T00:41:20.613Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,2,4,252,197,1,1,0,55,0,195,152,214,0,0,8,24,10,10,0,0,41,172,9,252,197,29] - 228
2020-01-09T00:41:20.613Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1026,"srcaddr":50684,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":55,"securityuse":0,"timestamp":14063811,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,10,10,0,0,41,172,9]}}
2020-01-09T00:41:20.614Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":10,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":41,"attrData":2476}],"Cluster":{"ID":1026,"attributes":{"measuredValue":{"ID":0,"type":41,"name":"measuredValue"},"minMeasuredValue":{"ID":1,"type":41,"name":"minMeasuredValue"},"maxMeasuredValue":{"ID":2,"type":41,"name":"maxMeasuredValue"},"tolerance":{"ID":3,"type":33,"name":"tolerance"},"minPercentChange":{"ID":16,"type":255,"name":"minPercentChange"},"minAbsoluteChange":{"ID":17,"type":255,"name":"minAbsoluteChange"}},"name":"msTemperatureMeasurement","commands":{},"commandsResponse":{}}},"address":50684,"endpoint":1,"linkquality":55,"groupID":0}'
2020-01-09T00:41:23.890Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0xccccccfffea15dc7', attempt 4
2020-01-09T00:41:23.891Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":38336,"nwkaddrofinterest":38336}
2020-01-09T00:41:23.891Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,192,149,192,149,35]
2020-01-09T00:41:23.900Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-01-09T00:41:23.900Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-01-09T00:41:23.900Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-01-09T00:41:23.900Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-01-09T00:41:23.900Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-09T00:41:33.894Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0xccccccfffea15dc7', attempt 5
2020-01-09T00:41:33.894Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":38336,"nwkaddrofinterest":38336}
2020-01-09T00:41:33.894Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,192,149,192,149,35]
2020-01-09T00:41:33.903Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-01-09T00:41:33.903Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-01-09T00:41:33.903Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-01-09T00:41:33.903Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-01-09T00:41:33.903Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-09T00:41:43.896Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0xccccccfffea15dc7', attempt 6
2020-01-09T00:41:43.896Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":38336,"nwkaddrofinterest":38336}
[... 5 lines cut ...]
[... cut ...]
winston:file logged 1554656 error 2020-01-09 01:42:03: Failed to interview 'remote_control', device has not successfully been paired
winston:file logged 1554837 info  2020-01-09 01:42:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"remote_control"}}'
```
## Info
zigbee2mqtt version: 1.8.0 (da4d26a)
CC253X firmware version: [CC2531_SOURCE_ROUTING_20190619](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20190619.zip)

TilmanK, 2020-01-10:
I'm having exactly the same problem on the dev branch with a CC26X2R1 coordinator. Nevertheless I have the problem with the LED driver: https://www.zigbee2mqtt.io/devices/ICPSHC24-30EU-IL-1.html

Attached are the logs and a pcap trace of the zigbee network.

[tradfri.zip](https://github.com/Koenkk/zigbee2mqtt/files/4047399/tradfri.zip)

TilmanK, 2020-01-10:
And it's the same for the smaller version: 
https://www.zigbee2mqtt.io/devices/ICPSHC24-10EU-IL-1.html

dmak, 2020-01-10:
I have re-flashed my stick with [CC2531_DEFAULT_20190608.zip](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608.zip) and pairing worked from the first attempt:
```
zigbee2mqtt:debug 2020-01-10 10:00:41: Received Zigbee message from 'remote_control', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20190211","swBuildId":"1.2.223"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-01-10 10:00:41: No converter available for 'E1524/E1810' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20190211","swBuildId":"1.2.223"}'                                            
2020-01-10T09:00:41.246Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []                  
2020-01-10T09:00:41.246Z zigbee-herdsman:controller:device Interview - got 'hwVersion,dateCode,swBuildId' for device '0xccccccfffea15dc7'                                                                                              
2020-01-10T09:00:41.247Z zigbee-herdsman:controller:device Interview - completed for device '0xccccccfffea15dc7'                                                                                                                                  
2020-01-10T09:00:41.248Z zigbee-herdsman:controller:log Succesfully interviewed '0xccccccfffea15dc7'
zigbee2mqtt:info  2020-01-10 10:00:41: Successfully interviewed 'remote_control', device has successfully been paired                                                                                                   
zigbee2mqtt:info  2020-01-10 10:00:41: Device 'remote_control' is supported, identified as: IKEA TRADFRI remote control (E1524/E1810)
```
Probably there was a stale data in stick memory related to unsuccessful pairing attempt.

However I suffer from the next issue: after remote control is paired, it does not create local bindings. In other words, I expect that when I press control pairing button when placing the control next to the bulb, the bulb is paired with controller. That issue is likely to be discussed in 782.

TilmanK, 2020-01-15:
I didn't do anything different from the last time. I just powered the Tradfri LED drivers, and they paired without problems. Strange. Needless to say that I'm happy.

closed 2020-03-22 by stale[bot]