Koenkk/zigbee2mqtt #2706: Not able to restart zigbee2mqtt

jffz, 2020-01-09:
# Bug Report
## What happened
Zigbee2MQTT only works after host reboot. Noway to only restart npm process.
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.13.4
3 info using node@v12.14.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.8.0~prestart: zigbee2mqtt@1.8.0
6 info lifecycle zigbee2mqtt@1.8.0~start: zigbee2mqtt@1.8.0
7 verbose lifecycle zigbee2mqtt@1.8.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.8.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.8.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.8.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.8.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.8.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.8.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:223:5)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:223:5)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
14 verbose pkgid zigbee2mqtt@1.8.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.75-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v12.14.1
19 verbose npm  v6.13.4
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.8.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.8.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
 ```

The only way to make it work again is by rebooting host

## What did you expect to happen
Being able to restart node process without reboot

## How to reproduce it (minimal and precise)
Restart zigbee2mqtt process

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: 20190608

redge76, 2020-01-09:
Same here. Using latest-dev and reflashing did not help.

Koenkk, 2020-01-11:
Can you post the zigbee2mqtt log itself? You can find it in the `data/log` folder.

sehraf, 2020-01-11:
Same here.
After failing to build z2m (off topic, see below; maybe it does ring a bell?) I've switched to  the docker image. 

> Can you post the zigbee2mqtt log itself? You can find it in the data/log folder.

There is none! The folder created is empty.

Docker log:
```
cat /tmp/2020-01-11T19_59_22_215Z-debug.log   
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@6.11.3
3 info using node@v10.17.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.8.0~prestart: zigbee2mqtt@1.8.0
6 info lifecycle zigbee2mqtt@1.8.0~start: zigbee2mqtt@1.8.0
7 verbose lifecycle zigbee2mqtt@1.8.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.8.0~start: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/app/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.8.0~start: CWD: /app
10 silly lifecycle zigbee2mqtt@1.8.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.8.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.8.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.8.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:198:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:198:13)
13 verbose stack     at maybeClose (internal/child_process.js:982:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.8.0
15 verbose cwd /app
16 verbose Linux 5.4.1-1-ARCH
17 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
18 verbose node v10.17.0
19 verbose npm  v6.11.3
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.8.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.8.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

npm install log:
```
npm install      

> @serialport/bindings@8.0.4 install /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.18.0 runtime=node arch=arm64 libc= platform=linux)
gyp ERR! UNCAUGHT EXCEPTION 
gyp ERR! stack TypeError: Class constructor Range cannot be invoked without 'new'
gyp ERR! stack     at PythonFinder.<anonymous> (/usr/lib/node_modules/node-gyp/lib/find-python.js:229:28)
gyp ERR! stack     at PythonFinder.execFileCallback (/usr/lib/node_modules/node-gyp/lib/find-python.js:274:7)
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:285:7)
gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
gyp ERR! stack     at maybeClose (internal/child_process.js:982:16)
gyp ERR! stack     at Socket.stream.socket.on (internal/child_process.js:389:11)
gyp ERR! stack     at Socket.emit (events.js:198:13)
gyp ERR! stack     at Pipe._handle.close (net.js:607:12)
gyp ERR! System Linux 5.4.1-1-ARCH
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
gyp ERR! node -v v10.18.0
gyp ERR! node-gyp -v v6.0.1
gyp ERR! This is a bug in `node-gyp`.
gyp ERR! Try to update node-gyp and file an Issue if it does not help:
gyp ERR!     <https://github.com/nodejs/node-gyp/issues>
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman-converters/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.11 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.11: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 7
npm ERR! @serialport/bindings@8.0.4 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 7
npm ERR! 
npm ERR! Failed at the @serialport/bindings@8.0.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/user/.npm/_logs/2020-01-11T19_46_35_305Z-debug.log
```

Koenkk, 2020-01-13:
This is the npm debug log, what do you see in the stdout?

sehraf, 2020-01-13:
```
npm start

> zigbee2mqtt@1.8.0-dev start /opt/zigbee2mqtt
> node index.js

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.8.0-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.8.0-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/user/.npm/_logs/2020-01-13T20_10_55_032Z-debug.log
```

Koenkk, 2020-01-14:
What if you execute: `cd /opt/zigbee2mqtt && npm start`?

sehraf, 2020-01-14:
That's what I've executed in two separate commands. (first `cd` and then `npm start`) 
Shall I try both with the `&&`?

jffz, 2020-01-14:
I am not sure of what is the issue here but it only happens on a raspberry(3/4) where Home Assistant is also running. If I move z2m to another rpi, it just works fine. Then I tried to invert both SD Cards, and the issue is back, also fixed by swapping z2m host again.

sehraf, 2020-01-14:
So i've tried @jffz suggestion and tried a spare ARM board, clean install but same result :(

sehraf, 2020-01-15:
FIXED!! 
got it working after commenting some unimportant lines in my config!
you want to guess what caused this non-saying error on my side?

<details>
  <summary>ready to feel stupid? (click me)</summary>  
  log_level: warning<br>
(it's 'warn', not 'warning')
</details>

@Koenkk maybe add some debug output to that part of the code :fearful:  

~~I've reproduces the "problem" with the following setups:
armv8 (64bit) Archlinux, node v13, latest dev, v1.8.0, v1.7.0
armv8 (64bit) Archlinux, node v10, latest dev, v1.8.0
armv8 (64bit) Archlinux, docker (latest)
armv7 (32bit) Archlinux, node v13, latest dev
armv7 (32bit) Archlinux, docker (latest)
armv7 (32bit) Debian 9, node v12, latest dev (this was a clean SD card using vendor image)~~

Koenkk, 2020-01-16:
Good that you found the issue, in the latest dev I improved the logging:
[image]

Thanks!

closed 2020-01-16 by Koenkk

chpins, 2021-03-06:
Same problem. 
No solution ?