Koenkk/zigbee2mqtt #2707: Ikea remote control and bulb not working together ( Group problem )

spider7611, 2020-01-09:
Hi,

I have 1 ikea dimmable bulb (LED1836G9 - IKEA TRADFRI LED bulb E26/E27 806 lumen) and 1 ikea remote control (E1743 - IKEA TRADFRI ON/OFF switch). Wan't to make a group to use together, but I have problem with it. See the log:
In domoticz have the 2 items and they are working, but not together.

`info  2020-01-09 15:33:57: Adding 'IKEA switch' to group 'ikeacucc'
info  2020-01-09 15:33:57: Connected to MQTT server
info  2020-01-09 15:33:57: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-01-09 15:33:57: MQTT publish: topic 'zigbee2mqtt/Xiaomi door contact', payload '{"battery":100,"voltage":3065,"contact":true,"linkquality":102}'
info  2020-01-09 15:33:57: MQTT publish: topic 'zigbee2mqtt/IKEA switch', payload '{"linkquality":36,"battery":60}'
info  2020-01-09 15:33:57: MQTT publish: topic 'zigbee2mqtt/IKEA bulb', payload '{"state":"OFF","brightness":255}'
info  2020-01-09 15:33:57: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0","commit":"da4d26a","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'
error 2020-01-09 15:34:04: Failed to call 'Groups' 'onZigbeeStarted' (Error: Data request failed with error: 'MAC transaction expired' (240)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:578:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:5:58))`

zigbee2mqtt version 1.8.0
coordinator firmware version 20190608

sjorge, 2020-01-09:
Did you add both the bulb and the switch to the group? (I think you did)
If so, that is not how it works.

You only add the bulb(s) to the group, you then bind the group to the switch.

spider7611, 2020-01-10:
Hi,

Yes you right.
Now changed it and after bind show this error and not working:

`warn  2020-01-10 08:54:04: No converter available for 'LED1836G9' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
warn  2020-01-10 08:54:06: No converter available for 'LED1836G9' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
warn  2020-01-10 08:54:08: No converter available for 'LED1836G9' with cluster 'genOnOff' and type 'read' and data '["onOff"]'
error 2020-01-10 09:00:10: Failed to call 'DeviceBind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Source undefined or not a device
    at DeviceBind.onMQTTMessage (/app/lib/extension/deviceBind.js:30:9)
    at Controller.callExtensionMethod (/app/lib/controller.js:322:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))`

sjorge, 2020-01-10:
Was the remote awake when the bind command happened?

spider7611, 2020-01-10:
Yes, pushed the button. 
Sending this command, is it good? `zigbee2mqtt/bridge/bind/IKEAswitch group_1
`
tryed it without friendly name only device id but the error is same

nldroid, 2020-01-10:
The command is good. 

Open the log and keep pressing the button while binding until you see in the log that binding was succeeded.

sjorge, 2020-01-10:
Correct, as @nldroid said you should see afew binding setup for genOnOff, genLevelCtrl,...

spider7611, 2020-01-10:
`warn 2020-01-10 08:54:04: No converter available for 'LED1836G9' with cluster 'genOnOff' and type 'read' and data '["onOff"]' warn 2020-01-10 08:54:06: No converter available for 'LED1836G9' with cluster 'genOnOff' and type 'read' and data '["onOff"]' warn 2020-01-10 08:54:08: No converter available for 'LED1836G9' with cluster 'genOnOff' and type 'read' and data '["onOff"]' `

This message show in the log one time after paired the IKEA bulb. After want to bound with the switch but only show that error message and no more genOnOff etc message. Tryed a lot of time.

Read this: "Note 1: that < zigbee2mqtt 1.4 automatically bound this device to group 99. If you want to bind it to a different group you first have to unbind it from group 99." 
So I must unbind it? Or only on 1.4 version?

nldroid, 2020-01-10:
Unbind it (just to be sure, there's no harm in doing/trying it) because it only can be bound to one group.

spider7611, 2020-01-10:
When I want to unbind get the same error... :(

spider7611, 2020-01-10:
This is show when I clicked the switch and send this messages together: `zigbee2mqtt/bridge/bind/IKEAswitch group_1`

`info  2020-01-10 12:28:54: MQTT publish: topic 'zigbee2mqtt/IKEAswitch', payload '{"linkquality":42,"click":"brightness_stop","device":{"friendlyName":"IKEAswitch","model":"E1743","ieeeAddr":"0x14b457fffe6c5b19","networkAddress":5185,"type":"EndDevice","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","applicationVersion":34,"stackVersion":98,"zclVersion":3,"hardwareVersion":1,"dateCode":"20190410","softwareBuildID":"2.2.008"}}'
info  2020-01-10 12:28:55: MQTT publish: topic 'zigbee2mqtt/IKEAswitch', payload '{"linkquality":44,"click":"on","device":{"friendlyName":"IKEAswitch","model":"E1743","ieeeAddr":"0x14b457fffe6c5b19","networkAddress":5185,"type":"EndDevice","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","applicationVersion":34,"stackVersion":98,"zclVersion":3,"hardwareVersion":1,"dateCode":"20190410","softwareBuildID":"2.2.008"}}'
error 2020-01-10 12:28:55: Failed to call 'DeviceBind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Source undefined or not a device
    at DeviceBind.onMQTTMessage (/app/lib/extension/deviceBind.js:30:9)
    at Controller.callExtensionMethod (/app/lib/controller.js:324:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))
error 2020-01-10 12:28:55: Failed to call 'DeviceBind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Source undefined or not a device
    at DeviceBind.onMQTTMessage (/app/lib/extension/deviceBind.js:30:9)
    at Controller.callExtensionMethod (/app/lib/controller.js:324:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))
error 2020-01-10 12:28:55: Failed to call 'DeviceBind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Source undefined or not a device
    at DeviceBind.onMQTTMessage (/app/lib/extension/deviceBind.js:30:9)
    at Controller.callExtensionMethod (/app/lib/controller.js:324:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))
info  2020-01-10 12:28:55: MQTT publish: topic 'zigbee2mqtt/IKEAswitch', payload '{"linkquality":39,"click":"on","device":{"friendlyName":"IKEAswitch","model":"E1743","ieeeAddr":"0x14b457fffe6c5b19","networkAddress":5185,"type":"EndDevice","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","applicationVersion":34,"stackVersion":98,"zclVersion":3,"hardwareVersion":1,"dateCode":"20190410","softwareBuildID":"2.2.008"}}'
error 2020-01-10 12:28:55: Failed to call 'DeviceBind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Source undefined or not a device
    at DeviceBind.onMQTTMessage (/app/lib/extension/deviceBind.js:30:9)
    at Controller.callExtensionMethod (/app/lib/controller.js:324:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))
error 2020-01-10 12:28:56: Failed to call 'DeviceBind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Source undefined or not a device
    at DeviceBind.onMQTTMessage (/app/lib/extension/deviceBind.js:30:9)
    at Controller.callExtensionMethod (/app/lib/controller.js:324:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))
info  2020-01-10 12:28:56: MQTT publish: topic 'zigbee2mqtt/IKEAswitch', payload '{"linkquality":39,"click":"on","device":{"friendlyName":"IKEAswitch","model":"E1743","ieeeAddr":"0x14b457fffe6c5b19","networkAddress":5185,"type":"EndDevice","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","applicationVersion":34,"stackVersion":98,"zclVersion":3,"hardwareVersion":1,"dateCode":"20190410","softwareBuildID":"2.2.008"}}'
info  2020-01-10 12:28:56: MQTT publish: topic 'zigbee2mqtt/IKEAswitch', payload '{"linkquality":39,"click":"on","device":{"friendlyName":"IKEAswitch","model":"E1743","ieeeAddr":"0x14b457fffe6c5b19","networkAddress":5185,"type":"EndDevice","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","applicationVersion":34,"stackVersion":98,"zclVersion":3,"hardwareVersion":1,"dateCode":"20190410","softwareBuildID":"2.2.008"}}'
error 2020-01-10 12:28:56: Failed to call 'DeviceBind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Source undefined or not a device
    at DeviceBind.onMQTTMessage (/app/lib/extension/deviceBind.js:30:9)
    at Controller.callExtensionMethod (/app/lib/controller.js:324:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))
info  2020-01-10 12:28:57: MQTT publish: topic 'zigbee2mqtt/IKEAswitch', payload '{"linkquality":39,"click":"on","device":{"friendlyName":"IKEAswitch","model":"E1743","ieeeAddr":"0x14b457fffe6c5b19","networkAddress":5185,"type":"EndDevice","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","applicationVersion":34,"stackVersion":98,"zclVersion":3,"hardwareVersion":1,"dateCode":"20190410","softwareBuildID":"2.2.008"}}'
error 2020-01-10 12:28:57: Failed to call 'DeviceBind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Source undefined or not a device
    at DeviceBind.onMQTTMessage (/app/lib/extension/deviceBind.js:30:9)
    at Controller.callExtensionMethod (/app/lib/controller.js:324:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))
info  2020-01-10 12:28:57: MQTT publish: topic 'zigbee2mqtt/IKEAswitch', payload '{"linkquality":39,"click":"on","device":{"friendlyName":"IKEAswitch","model":"E1743","ieeeAddr":"0x14b457fffe6c5b19","networkAddress":5185,"type":"EndDe [... cut ...]
    at process._tickCallback (internal/process/next_tick.js:68:7))
info  2020-01-10 12:28:57: MQTT publish: topic 'zigbee2mqtt/IKEAswitch', payload '{"linkquality":42,"click":"on","device":{"friendlyName":"IKEAswitch","model":"E1743","ieeeAddr":"0x14b457fffe6c5b19","networkAddress":5185,"type":"EndDevice","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","applicationVersion":34,"stackVersion":98,"zclVersion":3,"hardwareVersion":1,"dateCode":"20190410","softwareBuildID":"2.2.008"}}'`

spider7611, 2020-01-10:
This is a new setup. In th domoticz working good the 2 item. Can switch on the IKEAbulb. 
Made the group_1 and put in the IKEAbulb.

`info  2020-01-10 16:56:27: Logging to console and directory: '/app/data/log/2020-01-10.16-56-27'
info  2020-01-10 16:56:28: Starting zigbee2mqtt version 1.8.0-dev (commit #e766335)
info  2020-01-10 16:56:28: Starting zigbee-herdsman...
info  2020-01-10 16:56:29: zigbee-herdsman started
info  2020-01-10 16:56:29: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
info  2020-01-10 16:56:29: Currently 2 devices are joined:
info  2020-01-10 16:56:29: IKEAswitch (0x14b457fffe6c5b19): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
info  2020-01-10 16:56:29: IKEAbulb (0x14b457fffe733d3a): LED1836G9 - IKEA TRADFRI LED bulb E26/E27 806 lumen, dimmable, warm white (Router)
warn  2020-01-10 16:56:29: `permit_join` set to  `true` in configuration.yaml.
warn  2020-01-10 16:56:29: Allowing new devices to join.
warn  2020-01-10 16:56:29: Set `permit_join` to `false` once you joined all devices.
info  2020-01-10 16:56:29: Zigbee: allowing new devices to join.
info  2020-01-10 16:56:29: Connecting to MQTT server at mqtt://192.168.1.242:1883
info  2020-01-10 16:56:29: Connected to MQTT server
info  2020-01-10 16:56:29: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-01-10 16:56:29: MQTT publish: topic 'zigbee2mqtt/IKEAswitch', payload '{"linkquality":47}'
info  2020-01-10 16:56:29: MQTT publish: topic 'zigbee2mqtt/group_1', payload '{"state":"OFF","brightness":255}'
info  2020-01-10 16:56:29: MQTT publish: topic 'zigbee2mqtt/IKEAbulb', payload '{"state":"OFF","brightness":255}'
info  2020-01-10 16:56:29: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0-dev","commit":"e766335","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}'
info  2020-01-10 16:56:29: Zigbee: disabling joining new devices.
info  2020-01-10 16:56:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b00193e530e","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1578671789699},{"ieeeAddr":"0x14b457fffe6c5b19","type":"EndDevice","networkAddress":28798,"model":"E1743","friendly_name":"IKEAswitch","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI on/off switch","hardwareVersion":1,"softwareBuildID":"2.2.008","dateCode":"20190410","lastSeen":1578671594822},{"ieeeAddr":"0x14b457fffe733d3a","type":"Router","networkAddress":1070,"model":"LED1836G9","friendly_name":"IKEAbulb","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E27 WW 806lm","hardwareVersion":2,"softwareBuildID":"2.1.022","dateCode":"20181203","lastSeen":1578671237624}]}'
info  2020-01-10 16:56:29: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.8.0-dev","commit":"e766335","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}'
`

When I publish this: `zigbee2mqtt/bridge/bind/IKEAswitch group_1`

get this error:
`error 2020-01-10 17:03:57: Failed to call 'DeviceBind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Source undefined or not a device
    at DeviceBind.onMQTTMessage (/app/lib/extension/deviceBind.js:30:9)
    at Controller.callExtensionMethod (/app/lib/controller.js:324:44)
    at process._tickCallback (internal/process/next_tick.js:68:7))
`
where is the problem? Please help.

spider7611, 2020-01-10:
This is the configuration.yaml

maybe something wrong here?

`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.1.242:1883'
serial:
  port: /dev/ttyACM0
devices:
  '0x14b457fffe6c5b19':
    friendly_name: IKEAswitch
  '0x14b457fffe733d3a':
    friendly_name: IKEAbulb
    
groups:
  # ID, each group should have a different numerical ID
  '1':
    # Required: Name which will be used to control the group
    friendly_name: group_1
    # Required: Retain messages (true/false) (default: false)
    retain: false
    # Optional: Default transition to be used when e.g. changing brightness (in seconds) (default: 0)
    transition: 2
    # Optional: Change group state when one of the devices in it changes state, see 'State changes' below (default: true)
    optimistic: true
    # Optional: Devices of this group,
    # Note: this has to be the ieeeAddr of the device, not the friendly_name! (default: empty)
    devices:
      - '0x14b457fffe733d3a'
`

spider7611, 2020-01-11:
Another error: When I publish this message: zigbee2mqtt/IKEAbulb/set/state ON
Nothing happend and the log show this error: `zigbee2mqtt:error 2020-01-11 16:32:09: No converter available for 'state ON' ()`

spider7611, 2020-01-15:
Turned on the debug log, and everywhere see groupID 0 then I changed in the .yaml file groupid 1 to 0 and after restart the zigbeemqtt and try the bind again get the same errors but now it is working like a charm. I don't know how to get 0 group id the bulb and the switch too?!

closed 2020-03-22 by stale[bot]