Koenkk/zigbee2mqtt #2709: Changed order in mqtt message after repairing

krysly, 2020-01-10:
Im not sure if it's an issue, but i have no idea for that.

I am using 4 xiaomi MCCGQ01LM sensors. (https://www.zigbee2mqtt.io/devices/MCCGQ01LM.html)
All works fine.

Because of few connection problems with one of the sensors, i unpaired and paired it again.
It works fine so far.
But now i have a different order in the mqtt message.

before rebinding:
         {"contact":true,"linkquality":47,"battery":100,"voltage":3035}

after rebinding:
          {"battery":100,"voltage":3005,"linkquality":118,"contact":false}

I am using some split function with node red in a coding part and now it doesn't work.
I could change the function, but i would like to understand it.

Can I influence the content or why did this happen after the rebinding. 
Is there an explanation for this?

Many thanks for your help.
Christian

carldebilly, 2020-01-10:
According to the JSON specification (<https://www.json.org/json-en.html>), name/value pairs on an _object_ are **unordered**. So `{"a": 1, "b": 2}` and `{"b":2, "a":1}` should be considered as representing the same content.

TilmanK, 2020-01-10:
It seems to me as if you're getting a string from your MQTT node in NodeRed. Set the "Output" in your mqtt in node settings to "a parsed JSON object". You can then easily access properties in a JavaScript node with msg.payload.voltage (as an example). Does this help?

krysly, 2020-01-11:
> According to the JSON specification (https://www.json.org/json-en.html), name/value pairs on an _object_ are **unordered**. So `{"a": 1, "b": 2}` and `{"b":2, "a":1}` should be considered as representing the same content.

Thanks for the json website. It was not clear to me so far :-) Now I have a better understanding of it.

krysly, 2020-01-11:
> It seems to me as if you're getting a string from your MQTT node in NodeRed. Set the "Output" in your mqtt in node settings to "a parsed JSON object". You can then easily access properties in a JavaScript node with msg.payload.voltage (as an example). Does this help?

Yes you are absolutly right.
Fortunately, it always worked in the past.
I will now switch to json so that the order no longer matters in my programming.

Thank you for the quick reply and tips. That definitely helped.

closed 2020-03-11 by Koenkk