Koenkk/zigbee2mqtt #2710: Stelpro 218 not responding to set "occupied_heating_setpoint"

goldriver2, 2020-01-10:
I have 2 Stelpro baseboard reporting as STE218 to the zigbee2mqtt plugin. One of them respond and send a complete status payload when I set the occupied_heating_setpoint with an MQTT request. Immedialty after setting the temperature, the device respond and send the following payload wich indicate the new occupied_heating_setpoint:

{"local_temperature":21,"linkquality":5,"occupied_heating_setpoint":21,"pi_heating_demand":5}

The second baseboard also reporting as a STE218 to zigbee will change the temperature on the baseboard screen if I set he occupied_heating_setpoint with an MQTT request BUT won't respond with any paylod. Eventually when the actual temperature of the sensor will change, only then it will send the payload BUT the occupied_heating_setpoint IS NOT EQUAL to what I set up earlier via the mqtt command

{"local_temperature":23,"linkquality":15,"pi_heating_demand":0,"occupied_heating_setpoint":22,"system_mode":"heat"}

I can see a slight differnence between both response, the deffective one have system_mode":"heat" than the working one does not have

I remove the device from the network, re-installed it, I also did a power shutoff on the device and a complete reset and the problem is still there.

zigbee2mqtt version: zigbee2mqtt version 1.8.0-dev (commit #89fe701)

CC2531 firmware version: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20180815}}

carldebilly, 2020-01-10:
Is it a regression or it never worked?
(because I played with Stelpro's converters recently)

goldriver2, 2020-01-10:
In fact I have 2 of those devices wich are exactly the same and both have the same zigbee version installed. 
One of them reports correctly, I can see all changes make on the thermostat in the zigbee2mqtt log  BUT the second one never shows in the log even if I change the temperature on the device.

goldriver2, 2020-01-10:
Finally, I got it working, I had to 👍 
1- stop zigbe2mqtt
2- remove the devicve from configuration.yaml, database.db
3- stop the zigbee on the device
4- reset completely the device
5- bring the zigbee2mqtt coordinator near the device
6- rerstart zigbe2mqtt
6- re-pair
7- modify the friendly name in the configuration.yaml
8- force a temperature change a few times on the device

Still wondering why it stopped working suddenly after 6 monthe working perfectly. Even more concern that this is the only device that stopped working !

goldriver2, 2020-01-11:
Looks like everything is not perfect, just look at the systemlog and foud the following error

Failed to configure 'Stelpro Bas Escalier', attempt 3 (Error: Timeout - 33183 - 25 - 7 - 513 - 7 after 10000ms
jan 11 09:44:51 raspberrypi npm[2565]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
jan 11 09:44:51 raspberrypi npm[2565]:     at ontimeout (timers.js:436:11)
jan 11 09:44:51 raspberrypi npm[2565]:     at tryOnTimeout (timers.js:300:5)
jan 11 09:44:51 raspberrypi npm[2565]:     at listOnTimeout (timers.js:263:5)
jan 11 09:44:51 raspberrypi npm[2565]:     at Timer.processTimers (timers.js:223:10))

Can someone guide me on fixing the error please

carldebilly, 2020-01-13:
Maybe the Zigbee signal is too weak to downstairs' heater. You may try to put a router nearby, at least for the configuration process.

goldriver2, 2020-01-14:
I just brought the main coordinator right beside the device and still, after unpairing and repairing I have the same problem

closed 2020-03-21 by stale[bot]