Koenkk/zigbee2mqtt #2714: Failed starting herdsman

tbowmo, 2020-01-10:
# Bug Report
## What happened

For some reason zigbee2mqtt stopped working for me, I have a rpi2 running with zigbee2mqtt

During startup I get the following error:
```
 info  2020-01-10 23:23:57: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
info  2020-01-10 23:23:57: Starting zigbee-herdsman...
error 2020-01-10 23:24:10: Error while starting zigbee-herdsman
```
It repeats the above process, just writing the same 3 entries in each logfile.

Looking at dmesg, I see that there is problems with /dev/ttyACM0:

```
[ 2703.668461] cdc_acm 1-1.5:1.0: failed to set dtr/rts
[ 2745.988394] cdc_acm 1-1.5:1.0: failed to set dtr/rts
[ 2760.548386] cdc_acm 1-1.5:1.0: failed to set dtr/rts
[ 2802.948337] cdc_acm 1-1.5:1.0: failed to set dtr/rts
```

The device should be right:
```
➜  log git:(master) ✗ ls -l /dev/serial/by-id                            
totalt 0
lrwxrwxrwx 1 root root 13 jan 10 22:38 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001949FE52-if00 -> ../../ttyACM0
```

It has been running perfectly for the last 6-8 months, but suddenly it just stopped from one day to the other.

## Debug Info
zigbee2mqtt version: latest master (just did an update, to check if that would remedy my problem)
CC253X firmware version:  ?

tbowmo, 2020-01-10:
Moved the dongle to my laptop, and tried starting the latest master from zigbee2mqtt, getting a bit more info in the log:

```
zigbee2mqtt:info  2020-01-11 00:08:52: Logging to console and directory: '/home/thomas/zigbee2mqtt/data/log/2020-01-11.00-08-52'
zigbee2mqtt:info  2020-01-11 00:08:52: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
zigbee2mqtt:info  2020-01-11 00:08:52: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-01-11 00:09:53: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-01-11 00:09:53: Failed to start zigbee
zigbee2mqtt:error 2020-01-11 00:09:53: Exiting...
zigbee2mqtt:error 2020-01-11 00:09:53: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/home/thomas/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
```

voltagex, 2020-01-11:
I'm getting a similar error, too: 

```
zigbee2mqtt:info  2020-01-11 07:58:58: Logging to console and directory: '/srv/zigbee2mqtt/data/log/2020-01-11.07-58-58'
zigbee2mqtt:info  2020-01-11 07:58:59: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
zigbee2mqtt:info  2020-01-11 07:58:59: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-01-11 07:59:06: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-01-11 07:59:06: Failed to start zigbee
zigbee2mqtt:error 2020-01-11 07:59:06: Exiting...
zigbee2mqtt:error 2020-01-11 07:59:06: Error: SRSP - SYS - version after 6000ms
```

Koenkk, 2020-01-11:
> It has been running perfectly for the last 6 months, but suddenly it just stopped from one day to the other a week ago, or so..

This indicates that the stick has crashed, I would recommend switching to https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing which improves stability.

tbowmo, 2020-01-11:
@Koenkk 
Does the stick store anything in an eeprom onboard? If not, then it should work after a powercycle (which mine doesn't) also, it seems that I am using that particular firmware, that you link to. But I'll bring out my programming equipment, and try reprogramming the stick..

tbowmo, 2020-01-11:
I tried reflashing the stick with the same firmware again, (using the _default, instead of source_routing, as I only have 16 devices).

It brought the network back to life.. Closing as it seems to be a hardware glitch..

closed 2020-01-11 by tbowmo

Chompas, 2020-12-28:
> This indicates that the stick has crashed, I would recommend switching to https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing which improves [...]

@Koenkk can you help me pointing me in the right direction on how to reflash one of these sticks? I bought it preflashed and it was working without issues until very recently and I want to try this solution. Thanks.

ardu1, 2021-02-16:
Hello, every time when I start my ioBroker (running on a VM in Proxmox) there is this message after some minutes:

 [    248.471479]  cdc_acm 2-1:1.0: failed to set dtr/rts. 

I use a ConBee II stick with the ZigBee adapter. I checked for updates, all is ok. The only thing I could not check is the firmware of the ConBee II. There is one other problem, on the overview of all my ZigBee sensors, plugs and lamps the signal quality shows very often low, even though the device is directly beside the ConBee II stick.

Could anybody help me to solve the problem?

tbarbette, 2021-03-09:
> Hello, every time when I start my ioBroker (running on a VM in Proxmox) there is this message after some minutes:
>
> [ 248.471479] cdc_acm 2-1:1.0: failed to set dtr/rts.
>
> I use a ConBee II stick with [...]

Same message... If you find anything please ping me :)

ardu1, 2021-03-09:
Hello Tom, it seems that you have the same problem like me. Sometimes the signal is good and sometimes it is very low - but I didn't change any thing. My idea was also to check the firmware of the Conner II. I asked Dresden Elektronic and they wrote me that I could check and update the stick with a Windows computer. I think I will check it the next days.Best regards Am 09.03.21 um 10:03 schrieb Tom Barbette
                
            
            
                
                    Von: "Tom Barbette" <notifications@github.com>Datum: 9. März 2021An: "Koenkk/zigbee2mqtt" <zigbee2mqtt@noreply.github.com>Cc: "Comment" <comment@noreply.github.com>,"ardu1" <ingogl@gmx.de>Betreff: Re: [Koenkk/zigbee2mqtt] Failed starting herdsman (#2714)
                
                

Hello, every time when I start my ioBroker (running on a VM in Proxmox) there is this message after some minutes:
[ 248.471479] cdc_acm 2-1:1.0: failed to set dtr/rts.
I use a ConBee II stick with the ZigBee adapter. I checked for updates, all is ok. The only thing I could not check is the firmware of the ConBee II. There is one other problem, on the overview of all my ZigBee sensors, plugs and lamps the signal quality shows very often low, even though the device is directly beside the ConBee II stick.
Could anybody help me to solve the problem?

Same message... If you find anything please ping me :)

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Koenkk/zigbee2mqtt/issues/2714#issuecomment-793583127",
"url": "https://github.com/Koenkk/zigbee2mqtt/issues/2714#issuecomment-793583127",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]