Koenkk/zigbee2mqtt #2716: SmartThings plug GP-WOU019BBDWG doesn't work (no pairing)

magnayn, 2020-01-11:
# Bug Report

Not sure why this doesn't seem to respond to anything at all.

## What happened
Nothing - no logs even with DEBUG=* npm start

## What did you expect to happen
Plug to pair (and maybe work..)

## How to reproduce it (minimal and precise)
GP-WOU019BBDWG 

## Debug Info
zigbee2mqtt version:
dev

CC253X firmware version:
'{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

eyeson, 2020-01-16:
I am interested in buying this plug, did you manage to make it pair?

magnayn, 2020-01-16:
No - I get nothing out of it at all.

I think it's maybe Zigbee 3.0 -- I'm confused as to whether I need a zStack
3.0.x on my USB stick (when I flashed it with that though, nothing at all
seemed to work, but that may just be me).

It's a shame - I think the plug is really good - it measures current and
overall power consumption (I have a smartthings hub, which I'd love to get
rid of because it doesn't work with the Hue Smart Button, but then I can't
get that to work either).

eyeson, 2020-01-20:
I ordered one - mine paired and I've tried (trying) to add support for it :

https://github.com/Koenkk/zigbee-herdsman-converters/pull/933

magnayn, 2020-01-20:
> Koenkk/zigbee-herdsman-converters#933

Can I ask what zStack you're using on your Zigbee USB ?

Koenkk, 2020-01-20:
@magnayn zstack 3 or 1.2 shouldn't matter

> I have a smartthings hub, which I'd love to get
rid of because it doesn't work with the Hue Smart Button

Make sure to first turn off your complete smartthings network (so the hub itself AND all devices connected to it), otherwise the device might pair to that.

Koenkk, 2020-01-20:
And maybe too obvious, make sure you also set `permit_join: true` in the zigbee2mqtt `configuration.yaml`.

magnayn, 2020-01-22:
I had a spare moment, I got it to pair!

For anyone passing this way, it might have been down to the 'reset' procedure for the plug. 

Simply holding the button on the plug down didn't seem to be sufficient (even though the LED flash sequence looks the same as when it _does_ pair) :- you seem to have to hold the power button down _whilst_ plugging the socket in, then it pairs successfully (see below).

Might be good if the notes/a wiki could support this kind of hint :)

going to close this issue as it now works, will look at @eyeson patch to see if I can get data out of it.

Thx 

--
zigbee2mqtt:debug 2020-01-22 14:50:14: Received Zigbee message from '0xccccccfffebbf81a', type 'readResponse', cluster 'genBasic', data '{"modelId":"ZB-ONOFFPlug-D0005","manufacturerName":"LDS","powerSource":1}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-01-22 14:50:14: Received message from unsupported device with Zigbee model 'ZB-ONOFFPlug-D0005'
zigbee2mqtt:warn  2020-01-22 14:50:14: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-01-22 14:50:14: Received Zigbee message from '0xccccccfffebbf81a', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-01-22 14:50:14: Received message from unsupported device with Zigbee model 'ZB-ONOFFPlug-D0005'
zigbee2mqtt:warn  2020-01-22 14:50:14: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-01-22 14:50:14: Received Zigbee message from '0xccccccfffebbf81a', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3,"appVersion":1,"stackVersion":2}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-01-22 14:50:14: Received message from unsupported device with Zigbee model 'ZB-ONOFFPlug-D0005'
zigbee2mqtt:warn  2020-01-22 14:50:14: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-01-22 14:50:14: Received Zigbee message from '0xccccccfffebbf81a', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20190716-165","swBuildId":"2.3"}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-01-22 14:50:14: Received message from unsupported device with Zigbee model 'ZB-ONOFFPlug-D0005'
zigbee2mqtt:warn  2020-01-22 14:50:14: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:info  2020-01-22 14:50:14: Successfully interviewed '0xccccccfffebbf81a', device has successfully been paired
zigbee2mqtt:warn  2020-01-22 14:50:14: Device '0xccccccfffebbf81a' with Zigbee model 'ZB-ONOFFPlug-D0005' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

closed 2020-01-22 by magnayn