Koenkk/zigbee2mqtt #2717: Missing linkquality with  E1603/E1702 - IKEA TRADFRI control outlet (Router)

turboproc, 2020-01-11:
# Bug Report
## What happened
Having a setup with various  E1603/E1702 - IKEA TRADFRI devices I noticed that 1 device does not publish the link quality in the MQTT publish message while the other 3 devices do so (which I consider the correct behavious).
It's weird as the information received by Z2M from the controller does contain the link quality in all cases.

## What did you expect to happen
Would expect that also that the MQTT publish message contains the linkquality parameter in caase of the 4th device

## How to reproduce it (minimal and precise)
Happens contiously. BTW, on/off switching with this devices causes no issues at all

## Debug Info
zigbee2mqtt version:zigbee2mqtt version 1.8.0 (commit #da4d26a)
CC253X firmware version: 20190608

Traces of "faulty" device:

  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":25374,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":36,"securityuse":0,"timestamp":1417838,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,4,11,0,0]}} +21ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false},"transactionSequenceNumber":4,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":0,"statusCode":0},"Cluster":{"ID":6,"attributes":{"onOff":{"ID":0,"type":16,"name":"onOff"},"globalSceneCtrl":{"ID":16384,"type":16,"name":"globalSceneCtrl"},"onTime":{"ID":16385,"type":33,"name":"onTime"},"offWaitTime":{"ID":16386,"type":33,"name":"offWaitTime"}},"name":"genOnOff","commands":{"off":{"ID":0,"parameters":[],"name":"off"},"on":{"ID":1,"parameters":[],"name":"on"},"toggle":{"ID":2,"parameters":[],"name":"toggle"},"offWithEffect":{"ID":64,"parameters":[{"name":"effectid","type":32},{"name":"effectvariant","type":32}],"name":"offWithEffect"},"onWithRecallGlobalScene":{"ID":65,"parameters":[],"name":"onWithRecallGlobalScene"},"onWithTimedOff":{"ID":66,"parameters":[{"name":"ctrlbits","type":32},{"name":"ontime","type":33},{"name":"offwaittime","type":33}],"name":"onWithTimedOff"}},"commandsResponse":{}}},"address":25374,"endpoint":1,**"linkquality":36**,"groupID":0}' +62ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":49341,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":5,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,2,11,11,0]}} +64ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,189,192,1,1,6,0,5,0,30,5,24,2,11,11,0,85] +64ms
zigbee2mqtt:info  2020-01-11 13:27:40: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffe065ca2', payload '**{"state":"OFF"**}'

Traces of an OK device:

  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":51281,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":39,"securityuse":0,"timestamp":1418204,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,5,11,0,0]}} +8ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false},"transactionSequenceNumber":5,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":0,"statusCode":0},"Cluster":{"ID":6,"attributes":{"onOff":{"ID":0,"type":16,"name":"onOff"},"globalSceneCtrl":{"ID":16384,"type":16,"name":"globalSceneCtrl"},"onTime":{"ID":16385,"type":33,"name":"onTime"},"offWaitTime":{"ID":16386,"type":33,"name":"offWaitTime"}},"name":"genOnOff","commands":{"off":{"ID":0,"parameters":[],"name":"off"},"on":{"ID":1,"parameters":[],"name":"on"},"toggle":{"ID":2,"parameters":[],"name":"toggle"},"offWithEffect":{"ID":64,"parameters":[{"name":"effectid","type":32},{"name":"effectvariant","type":32}],"name":"offWithEffect"},"onWithRecallGlobalScene":{"ID":65,"parameters":[],"name":"onWithRecallGlobalScene"},"onWithTimedOff":{"ID":66,"parameters":[{"name":"ctrlbits","type":32},{"name":"ontime","type":33},{"name":"offwaittime","type":33}],"name":"onWithTimedOff"}},"commandsResponse":{}}},"address":51281,"endpoint":1,"linkquality":39,"groupID":0}' +122ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":51281,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":8,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,5,11,11,0]}} +63ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,81,200,1,1,6,0,8,0,30,5,24,5,11,11,0,187] +63ms
zigbee2mqtt:info  2020-01-11 13:27:40: MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef6e778', payload '{"state":"OFF",**"linkquality":42}**'

aBaDDoNNL, 2020-01-14:
I'm experiencing the same issue after upgrading from 1.7.x to 1.8.0.

I have 3 similar TRADFRI GU10 LED Bulbs both at the same time. Only I have used two of them and those were paired with Zigbee2Mqtt 1.6.x or 1.7.x. These two show the linkquality attribute and have a value. The third bulb I recently started using after running Zigbee2Mqtt 1.8.0. That one is missing the linkquality attribute.

TRADFRI GU10 bulbs registered with Zigbee2Mqtt 1.6.x or 1.7.x:
zigbee2mqtt:info  2020-01-14 19:32:16: MQTT publish: topic 'zigbee2mqtt/0x000b3cfffef0379d', payload '{"state":"OFF","linkquality":13,"brightness":155}'
zigbee2mqtt:info  2020-01-14 19:32:16: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffe1e9aa1', payload '{"state":"OFF","linkquality":99,"brightness":155}'

TRADFRI GU10 bulb registered after the 1.8.0 release:
zigbee2mqtt:info  2020-01-14 19:32:17: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffe2f9f99', payload '{"state":"OFF","brightness":155}'

Same goes for my Xiaomi MiJia motion sensors. The newly paired ones aren't showing illuminance values while the one I paired with Zigbee2Mqtt version 1.7.x does.

Registered sensor with 1.7.x:
zigbee2mqtt:info  2020-01-14 19:38:18: MQTT publish: topic 'zigbee2mqtt/0x00158d00033ad85c', payload '{"illuminance":17,"linkquality":49,"occupancy":false,"battery":100,"voltage":3015,"no_occupancy_since":60}'

Registered sensors after 1.8.0:
zigbee2mqtt:info  2020-01-14 19:38:57: MQTT publish: topic 'zigbee2mqtt/0x00158d0003d29418', payload '{"occupancy":false,"linkquality":42,"battery":100,"voltage":3015}'
zigbee2mqtt:info  2020-01-14 19:38:58: MQTT publish: topic 'zigbee2mqtt/0x00158d0003d0914d', payload '{"occupancy":false,"linkquality":23,"battery":100,"voltage":3045}'

aBaDDoNNL, 2020-01-14:
Actually I also have a third device showing the same.

Added a OSRAM Smart+ lightbulb after 1.8.0 and it is missing the linkquality:
zigbee2mqtt:info  2020-01-14 19:31:38: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa0a045cc4', payload '{"state":"ON","brightness":155}'

But my other two OSRAM bulbs do have linkquality and where registered before 1.8.0:
zigbee2mqtt:info  2020-01-14 20:21:57: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa0a03e503', payload '{"state":"ON","linkquality":28,"brightness":255}'
zigbee2mqtt:info  2020-01-14 20:21:59: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa0a0597b9', payload '{"state":"OFF","linkquality":44,"brightness":255}'

GrammaTonic, 2020-01-17:
I had the same issue, on 1.8.0 and 1.9.0 i was toying around and enabled reporting.

advanced:
  report: true

 Now i have a linkquality available in de mqtt:
`zigbee2mqtt:debug 2020-01-18 00:12:23: Received Zigbee message from '0x14b457fffe44ff29', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":40}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-18 00:12:23: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe44ff29', payload '{"state":"OFF","brightness":40,"linkquality":34}'`

But this is ofcourse from de reporting of the device. 
Maybe a temp workaround for now.

Koenkk, 2020-01-28:
No that shouldn't be needed, I think this was already fixed in 1.9.0 but it requires reconfiguring of the device (once).

https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

bakerboy908, 2020-02-15:
Im having the same issue, no link quality for any of my IKEA GU10 bulbs
after doing a configure for the device the log of zigbee2mqtt outputs
`zigbee2mqtt:warn  2020-02-15 15:57:42: Skipping configure of 'Stair bulb 1', device does not require this.`

I have 14 of these bulbs none report link quality.

socialstijn, 2020-03-14:
I have  the same issue, only the device does not show the 'light' property. Also getting he 'does not require this' when running the configure MQTT command.
[image]

FedeBZ, 2020-04-19:
news on this issue
configure does not work for me:
"Skipping configure of 'Osram_lamp1', device does not require this."

Thanks

silvanverschuur, 2020-04-19:
I've the same issue with multiple Ikea power outlets (E1603/E1702) and a couple of Ikea filament bulbs (LED1842G3). All my other devices do have a link quality. Maybe it's Ikea related?

Koenkk, 2020-04-20:
@silvanverschuur by default bulbs don't send messages to the coordinator therefore we don't know the linkquality.

In case you want to retrieve the linkquality you can force the bulb to send a message by publishing to `zigbee2mqtt/[FRIENDLY_NAME]/get` payload `{"state": ""}`

silvanverschuur, 2020-04-20:
@Koenkk Thanks for your explanation. I didn't know that bulbs don't send their link quality to the coordinator. I've tested your MQTT suggestion and that works fine. I now have the link quality of the Ikea bulbs in Home Assistant.

closed 2020-06-28 by stale[bot]