Koenkk/zigbee2mqtt #2718: No Converters available for IKEA E1524/1810 remote?

Anjerlaan, 2020-01-11:
# Bug Report
## What happened
With the CC2530 Router (Keuken_Router) I paired:
3 IKEA GU10 1737R5 bulbs (in the log paired as 1537R6)
1 IKEA E1524/E1810 remote (firmware 2.3.014)

To control the 3 bulbs I created a group (Keuken_Spots), and added all 3 bulbs
According to the https://www.zigbee2mqtt.io/devices/E1524_E1810.html page
It should be possible to bind (zigbee2mqtt/bridge/bind/Keuken_Spots) the remote to the group, but after countless (different) attempts I still got binding errors.

Then I tried the work around decribed on the same page. After binding the physical remote to the physical bulbs I got a GroupID listed in the Z2M log. In the configuration.yaml I changed the Group number of the bulbs to the GroupID of the remote.....and this works. (however not the left and right buttons by the way??)

Now I don't understand how the Remote in combination with the group of bulbs exactly works...
When I send a set/state (e.g. ON/OFF) command with MQTT.fx to the group the group state changes as expected.

When I send a set/state command to the Remote (I tried ON/OFF/TOGGLE) the log says there are no converters available?
But when I press the ON/OFF/TOGGLE button of the physical remote, the group of bulbs is set to ON/OFF

This behaviour is also in Domoticz, where I use the Zigbee2mqtt Domoticz Plugin.
All 5 devices (left, right, up, down, ON/OFF) of the Remote are listed and I can add them as a device, Also here....When I press buttons of the remote, the domoticz remote devices are responding, but there is no respond when press the the domoticz devices.

When I check the LOG after pressing the physical ON/OFF toggle button of the remote, the following messages are send:
```
debug 2020-01-11 13:43:51: Received Zigbee message from 'Keuken_Router', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2020-01-11 13:43:51: MQTT publish: topic 'zigbee2mqtt/Keuken_Router', payload '{"state":true,"led_state":false,"linkquality":63}'
debug 2020-01-11 13:44:12: Received Zigbee message from 'Remote_E1524-E1810', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 58594
info  2020-01-11 13:44:12: MQTT publish: topic 'zigbee2mqtt/Remote_E1524-E1810', payload '{"linkquality":60,"battery":60,"action":"toggle"}'
```
What I find strange is that the data of the Remote is empty **cluster 'genOnOff', data '{}'**
But message of the Router reports: **'genOnOff', data '{"onOff":0}'**

## What did you expect to happen

I Thought that when a remote is bound to a group, it should be possible to send a state command to the remote?

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.8.0
CC2531 and CC2530 firmware version: latest

Here the full log after restarting zigbee2mqtt: https://pastebin.com/CAvuJneu

Koenkk, 2020-01-13:
> (however not the left and right buttons by the way??)

This is a known issue

> When I send a set/state command to the Remote (I tried ON/OFF/TOGGLE) the log says there are no converters available?

This makes sense as the remote cannot be turned on/off, you should send these commands to the bulb.

Anjerlaan, 2020-01-13:
> This makes sense as the remote cannot be turned on/off, you should send these commands to the bulb.

But if no MQTT commands can be send to the remote, how does the remote know that the group with bulbs is set to OFF with an MQTT command?

Koenkk, 2020-01-14:
The remote doesn't, when pressing the on/off button on the remote it just send a toggle command to the group.

Anjerlaan, 2020-01-14:
Okay, the reason why I ask this is that I use Domoticz and installed the zigbee2mqtt-domoticz plugin.
All paired bulbs and remote are showing up as devices in the list.
Sadly it doesn't create a Group/Scene of the zigbee2mqtt Group i have created :-(  

[image]

Anyway, the 5 buttons of the Remote show up as 5 light/switch devices,
When I activate them like this example (this switch is the TOGGLE button of the remote):

[image]

and I try turn on/off the switch, there is no respond at all.
They only change state when I physically press the remote button
So they are basically unuseless in Domoticz.

Therefor I need to find a way to control the zigbee2mqtt Group in Domoticz.
I have read about this zigbee2mqtt Group setting:
```
# Optional: Change group state when one of the devices in it changes state, see 'State changes' below (default: true)
optimistic: true
```
I tested this setting with a **zigbee2mqtt/one_of_the_bulbs/set** command but this doesn't work, only the state of the bulb changes, but not the group and members.

Am I misreading the explanation for group State Changes or is something wrong with my group configuration?

If this setting works as I read it, then I only have to add one of the Light Switches of the group in domoticz to control the group

Koenkk, 2020-01-16:
- The integration of the E1524 is definitely wrong, you cannot control this device, you can only receive events from it (when e.g. pushing buttons on it). In home assistant this is integrated as a sensor. For that please create an issue here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues

- To control a group of device first make sure you make a zigbee2mqtt group (https://www.zigbee2mqtt.io/information/groups.html) after that control it by e.g. sending to `zigbee2mqtt/[GROUP_FRIENDLY_NAME]/set` payload `{"state": "ON"}`

Anjerlaan, 2020-01-16:
I will create an issue for the remote.

Group control is already working as you described.
Only the binding of the remote (2.3.014) didn't work (lots of times, also with button pressed), so I used the workaround decribed on the remotes page, which worked.

I have only one question left...this is my group config:
```
groups:
  '58594':
    friendly_name: Keuken_Spots
    retain: false
    transition: 2
    optimistic: true
    devices:
      - 0x14b457fffe4594ed/1
      - 0x14b457fffe266637/1
      - 0x14b457fffe437ffb/1
```
When I added the bulbs with "zigbee2mqtt/bridge/group/Keuken_Spots/add" -m "Keuken_Spot1",
/1 was added too.
Is this because I paired them with a router, which is on node 1?

Anjerlaan, 2020-01-16:
Ooops...this should not be here...how can I remove this?

Koenkk, 2020-01-17:
`/1` is the endpoint of the device that was added, so this looks fine.

Dont worry about the link, it is nice to refer to the discussion here.

Anjerlaan, 2020-01-17:
Thanx!

Anyway....This was the response of the issue i created:
https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues/231

Anjerlaan, 2020-01-29:
Koen,

When I turn the Group ON/OFF with the Remote toggle button, the ON/OFF state of the Group is dynamically updated (I use MQTT explorer to check).
But when I turn the "brightness" up or down with the Remote, the "brightness" of the Group is not updated, but the bulbs are visually dimming.

Why is the "brightness" value not updated in the Group? Is that normal behaviour?

closed 2020-04-05 by stale[bot]