Koenkk/zigbee2mqtt #2719: JG Speedfit / Salus thermostat - Doesn't join?

cyclops1982, 2020-01-11:
# Bug Report
## What happened
I'm trying to add the JG Speedfit Aura Wireless Thermostat. These are 'rebranded' Salus devices.
From their website, i understand that they should be zigbee, and i can even see them show up in the logfiles, so they seem to join. However, on the device itself, it doesn't move past the 'join' phase. It never shows that it's paired and once i restart zigbee2mqtt, i also see the device going into a reset mode where it's back to factory settings.

## What did you expect to happen
I'd like to add this device to zigbee2mqtt so that i can read and control the temperature :)

## How to reproduce it (minimal and precise)
1. Start Zigbee2mqtt (new install, in python3 virtualenv). Make sure it's in pair mode.
2. Plug in the thermostat. You'll get a question about what type of usage you have (UHF or Radiator). Skip that.
3. Device is in join mode. Zigbee2mqtt reports that it's joined (but not supported), but the device itself never moves past the join phase.

## Debug Info
zigbee2mqtt version: 1.8.0 (commit #da4d26a)
CC253X firmware version:  Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

Logfile: https://cyclops.nettrends.nl/stuff/zigbee/debuglog1.txt

cyclops1982, 2020-01-11:
After some more looking around and looking at other Salus devices, i've now got the following log output: https://cyclops.nettrends.nl/stuff/zigbee/debuglog2.txt

I'm using this config in devices.js that i'm using.
`{
                zigbeeModel: ['SAL6DT1'],
                model: 'SAL6DT1',
                vendor: 'Computime',
                description: 'JGSTATW2 Wireless Thermostat',
                supports: 'Temperature',
                fromZigbee: [
                        fz.generic_battery_voltage,
                        fz.thermostat_att_report,
                        ],
                toZigbee: [],
                meta: {options: {disableDefaultResponse: true}, configureKey: 1},
                configure:  async (device, coordinatorEndpoint) => {
                        const endpoint = device.getEndpoint(9);
                        await bind(endpoint, coordinatorEndpoint, ['genBasic']);
                },
        },`

Is there any documentation around the meta and configure options on how this works? I know a bit of javascript, but i don't understand the current bits of code.
As an example, i've got no clue what this ConfigureKey number is for.

I have the 'feeling' that this device expects some kind of code back after it has joined the network, so that the device moves forward into some other details. If that's the case, then how would i know what it would be, and also how would i do that?

Koenkk, 2020-01-13:
> I have the 'feeling' that this device expects some kind of code back after it has joined the network, so that the device moves forward into some other details. If that's the case, then how would i know what it would be, and also how would i do that?

As this is not default zigbee behaviour you probably need to have the original gateway to sniff what needs to be send. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

cyclops1982, 2020-01-13:
Hi @Koenkk 
Thanks for responding.

Let's say i do get a official controller. Would we then be able to 'garantee' that i would be able to get this to  work?

The reason i'm asking is that i already have this setup with wired thermostats. I've got 3 in total. I also need to replace the controller unit. So at the moment i'm looking at around 500EUR that i would eventually have to bring up to automate all 3 thermostats. The controller unit (to which the thermostats connect) is around 300EUR.

I figured i'd give the thermostat a quick go as i can easily return it. I'm happy to buy things (And even return them :)) if i can get them to work.

So the question is - if i do have the sniffer up and running, would that (more or less) garantee that i can get this to work?

Koenkk, 2020-01-16:
No guarantees but if you are willing to invest enough time it you should be able to get it working. It highly depends on the time you want to invest.

cyclops1982, 2020-01-17:
Thanks @Koenkk 

I'm going to ponder about this for a bit before committing to this. It's good to know that we can try to get it to work!

closed 2020-01-17 by cyclops1982