Koenkk/zigbee2mqtt #2721: Color values reported in inconsistent manner

stoinov, 2020-01-11:
# Bug Report
## What happened
I do not use HomeAssistant bit instead my own Node-RED implementation. Thus I am using `output: attribute` to flatten all output. I am setting and reading `color-hue` and `color-saturation` in my automation with GL-MC-001 which works fine. This is what my controller (HomeKit emulator) returns already.
But then I tried changing the `color_temp` and this made Z2M reporting `color-x` and `color-y` values. I then send my original hue and saturation values to revert to color, and it continue reporting all 4 values: `color-hue`, `color-saturation`, `color-x` and `color-y` on each change.

This makes it impossible for me to sync multiple control paths as I will be unable to determine which value is being active at any given moment - RGB with x and y, RGB with hue and sat or CCT with color_temp.

## What did you expect to happen
After an update, Z2M should broadcast only the changed values in a consistent manner, regardless of the input values.

Example: setting `color_temp` should only return changed value of `color_temp` instead of both it and `color-x` and `color-y`.

## How to reproduce it (minimal and precise)
Try setting RGB and CCT values by using each of the [available methods in the documentation](https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameset), and see if you can deduct what has been changed based on the following MQTT payload.

## Debug Info
zigbee2mqtt version: `latest-dev`
CC253X firmware version:  `20190608`

Some examples of what I set in the MQTT topic and what I get reported back by Z2M:
`set color_temp: 100`
```
state', payload 'ON'
brightness', payload '10'
last_seen', payload '1578763883568'
color_temp', payload '100'
linkquality', payload '36'
color-x', payload '0.280632719756407'
color-y', payload '0.288286029784579'
color-hue', payload '230'
color-saturation', payload '17'
```
`set r: 46, g: 102, b: 193`
```
state', payload 'ON'
brightness', payload '10'
last_seen', payload '1578763943512'
color_temp', payload '18'
linkquality', payload '36'
color-x', payload '0.1598'
color-y', payload '0.1559'
color-hue', payload '230'
color-saturation', payload '17'
```
`set hue: 30, saturation: 100`
```
state', payload 'ON'
brightness', payload '10'
last_seen', payload '1578763996764'
color_temp', payload '18'
linkquality', payload '36'
color-x', payload '0.1598'
color-y', payload '0.1559'
color-hue', payload '30'
color-saturation', payload '100'
```
`set brightness: 50`
```
state', payload 'ON'
brightness', payload '50'
last_seen', payload '1578764156858'
color_temp', payload '18'
linkquality', payload '36'
color-x', payload '0.1598'
color-y', payload '0.1559'
color-hue', payload '30'
color-saturation', payload '100'
```

TilmanK, 2020-01-11:
I do the same in my home automation and this hasn't been any problem so far. What HomeKit emulator are you using in NodeRed. I can recommend https://www.npmjs.com/package/node-red-contrib-homekit-bridged. I'll post a simple example flow for you tomorrow if this helps.

stoinov, 2020-01-11:
@TilmanK it's what I use in the form of the excellent [Docker image of Raymond](https://github.com/RaymondMouthaan/node-red-homekit-docker).

I'd love to see an example of how you handle the sync if you use multiple ways for controlling a light.

TilmanK, 2020-01-12:
Ok, you're right @stoinov. I, too, have a problem with the color sync but I guess different from your problem.

When setting a color via hue and saturation, setting the brightness will erase that color. This will produce strange results in the home app. Let me see what I can find out about this.

This is truly not satisfying, but I don't know if this has anything to do with Z2M. From the sniffed traffic it looks like Z2M is just passing values trough.

TilmanK, 2020-01-12:
@stoinov: Here's the flow I use for controlling a Hue Smart Bulb with the Apple Home app, a switch and a Dimmer. The switch triggers a Shelly (https://shelly.cloud/shelly1-open-source/) that sends out an MQTT message when triggered (it does not control the bulb anymore, the bulb always has power - the Shelly operates in so called "Detached Mode"). The dimmer is a common Hue Dimmer.

[{"id":"67935987.67aea8","type":"tab","label":"Licht Arbeitszimmer","disabled":false,"info":""},{"id":"c7887ade.befd98","type":"homekit-service","z":"67935987.67aea8","isParent":true,"bridge":"8f52f202.615ad","accessoryCategory":"LIGHTBULB","parentService":"","name":"Licht Arbeitszimmer","serviceName":"Lightbulb","topic":"","filter":false,"manufacturer":"Default Manufacturer","model":"Default Model","serialNo":"Default Serial Number","cameraConfigVideoProcessor":"","cameraConfigSource":"","cameraConfigStillImageSource":"","cameraConfigMaxStreams":"","cameraConfigMaxWidth":"","cameraConfigMaxHeight":"","cameraConfigMaxFPS":"","cameraConfigMaxBitrate":"","cameraConfigVideoCodec":"","cameraConfigAudioCodec":"","cameraConfigAudio":false,"cameraConfigPacketSize":"","cameraConfigVerticalFlip":false,"cameraConfigHorizontalFlip":false,"cameraConfigMapVideo":"","cameraConfigMapAudio":"","cameraConfigVideoFilter":"","cameraConfigAdditionalCommandLine":"","cameraConfigDebug":false,"characteristicProperties":"{\n    \"Brightness\":true,\n    \"ColorTemperature\":true,\n    \"Hue\":true,\n    \"Saturation\":true\n}","x":690,"y":160,"wires":[["aea4e94.81ff018"],[]]},{"id":"84a46157.32c43","type":"function","z":"67935987.67aea8","name":"HomeKit Converter","func":"const r_msg = {payload:{}};\nif (msg.payload.hasOwnProperty('state')){\n    r_msg.payload.On = msg.payload.state == \"ON\"\n    flow.set(\"light\", msg.payload.state == \"ON\")\n}\nif (msg.payload.hasOwnProperty('brightness')){\n    flow.set(\"brightness\", msg.payload.brightness)\n    r_msg.payload.Brightness = Math.round(msg.payload.brightness/2.55);\n}\nif (msg.payload.hasOwnProperty('color_temp')){\n    r_msg.payload.ColorTemperature = msg.payload.color_temp;\n}\nif (msg.payload.hasOwnProperty('color')){\n    if (msg.payload.color.hasOwnProperty('hue')){\n        r_msg.payload.Hue = msg.payload.color.hue;\n    }\n    if (msg.payload.color.hasOwnProperty('saturation')){\n        r_msg.payload.Saturation = msg.payload.color.saturation;\n    }\n}\nreturn r_msg","outputs":1,"noerr":0,"x":470,"y":160,"wires":[["c7887ade.befd98"]]},{"id":"aea4e94.81ff018","type":"function","z":"67935987.67aea8","name":"Homekit MQTT Converter","func":"if (msg.payload.hasOwnProperty('Hue') || msg.payload.hasOwnProperty('Saturation')){\n    if (msg.payload.hasOwnProperty('Hue')) {\n        flow.set(\"hue\", msg.payload.Hue);\n    }\n    if (msg.payload.hasOwnProperty('Saturation')) {\n        flow.set(\"saturation\", msg.payload.Saturation);\n    }\n    if (flow.get(\"hue\") && flow.get(\"saturation\"))\n    {\n        color = {\n            hue: flow.get(\"hue\"),\n            saturation: flow.get(\"saturation\")\n        };\n        flow.set(\"hue\", undefined);\n        flow.set(\"saturation\", undefined);\n        return {payload: {color}};\n    }\n}\nelse {\n    const payload = {};\n    if (msg.payload.hasOwnProperty('On')) {\n        payload.state = msg.payload.On ? \"on\" : \"off\";\n    }\n    if (msg.payload.hasOwnProperty('Brightness')) {\n        payload.brightness = Math.round(msg.payload.Brightness * 2.55);\n    }\n    if (msg.payload.hasOwnProperty('ColorTemperature')) {\n        payload.color_temp = msg.payload.ColorTemperature;\n    }\n    return {payload: payload};\n}\n","outputs":1,"noerr":0,"x":950,"y":160,"wires":[["48579c65.a51464"]]},{"id":"70cb59fd.9996d8","type":"mqtt in","z":"67935987.67aea8","name":"Licht Arbeitszimmer","topic":"zigbee2mqtt/Licht Arbeitszimmer","qos":"2","datatype":"json","broker":"ff1894f1.62d6f8","x":250,"y":160,"wires":[["84a46157.32c43"]]},{"id":"48579c65.a51464","type":"mqtt out","z":"67935987.67aea8","name":"Licht Arbeitszimmer","topic":"zigbee2mqtt/Licht Arbeitszimmer/set","qos":"","retain":"","broker":"ff1894f1.62d6f8","x":1230,"y":220,"wires":[]},{"id":"2265dc36.cfa604","type":"mqtt in","z":"67935987.67aea8","name":"","topic":"shellies/Shelly-Licht-Arbeitszimmer/input/0","qos":"2","datatype":"utf8","broker":"ff1894f1.62d6f8","x":320,"y":220,"wires":[["e5f52d1f.a8bb8"]]},{"id":"e5f52d1f.a8bb8","type":"function","z":"67935987.67aea8","name":"Shelly MQTT Converter","func":"if (flow.last_state === undefined) {\n    flow.last_state = msg.payload;\n}\nelse if (flow.last_state != msg.payload) {\n    if (flow.get(\"light\")) {\n        flow.last_state = msg.payload;\n        return {payload:{state: \"off\"}};\n    }\n    else {\n        flow.last_state = msg.payload;\n        return {payload:{state: \"on\"}};\n    }\n}","outputs":1,"noerr":0,"x":950,"y":220,"wires":[["48579c65.a51464"]]},{"id":"68841841.81a848","type":"mqtt in","z":"67935987.67aea8","name":"Dimmer Arbeitszimmer","topic":"zigbee2mqtt/Dimmer Arbeitszimmer","qos":"2","datatype":"json","broker":"ff1894f1.62d6f8","x":260,"y":280,"wires":[["befdb623.17ec98"]]},{"id":"befdb623.17ec98","type":"function","z":"67935987.67aea8","name":"Dimmer MQTT Converter","func":"if (flow.get(\"brightness\") === undefined){\n    flow.set(\"brightness\", 255)\n}\n\nif (msg.payload.action == \"on-press\") {\n    return {payload: {state: \"on\"}}\n}\nelse if (msg.payload.action == \"off-press\") {\n    return {pay [... cut ...]
        Data Type: 8-Bit Enumeration (0x30)
        Color Mode: Color X and Y (0x01)
    Status Record
        Attribute: Color Loop Active (0x4002)
        Status: Success (0x00)
        Data Type: 8-Bit Unsigned Integer (0x20)
        Active: False

From what I see on the sniffed Zigbee traffic from Z2M we just issue a message "Move to Color Temperature" and then get the ack. I assume, that calculation of x and y is done within Z2M.

TLDNR: We'd need Hue and Saturation reported every time x and y is reported. :)

stoinov, 2020-01-12:
Mine issue is the opposite, but I assume has a common root cause:
I always get all the values (as shown in the debug info). The problem comes from the fact that I would have to keep track of the last state of the light, otherwise I would not be able to consistently control it. 

Here's an example:
I have the light set to some white using the CCT. If then Brightness update is sent using non HomeKit option (hardware dimmer for instance), I would get MQTT update including all the previous values but with the new brightness (see last debug example). I have to keep the last state of all the values in Node-RED and compare any new update to it, then only apply differences to the HomeKit node. Although doable, this is not optimal and also makes MQTT VERY chatty (try moving your finger around the color selection circle and monitor your MQTT).
And **_the main issue_** I have is that if I apply `color_temp`, it also updates the `color-x` and `color-y` values for some reason, but not the other way around.

I would be happy if posting only updates is settable property as I am aware that our cases are not majority and HA is the main usage.

@TilmanK try to see if the values will start reporting every time if you set them at least once. I think mine are only reporting them if they have been set at least once before.

TilmanK, 2020-01-12:
You can control that only values that are sent on Zigbee are forwarded to MQTT with the flag "cache_state" in your configuration. This is what I do. https://www.zigbee2mqtt.io/information/configuration.html

The update of x and y is done explicitly in zigbee-herdsman-converters. In theory this could be switched off. What I would like to know is how you use x and y with Apple HomeKit. I haven't figured out how to convert them to anything useful. Mind sharing one of your flows too? 

What works perfectly for me is setting a color in the Home App. Switching color temperature will produce strange results as I would need to update Hue and Saturation (which I can't atm).

stoinov, 2020-01-12:
Thanks! How did I missed that `cached_state`...

As for your question - I am not sure I understand it. I do not use it explicitly, but it was reported by my light. At some point it can turn out that a different controller uses it so I would just use [Color light converter](https://flows.nodered.org/node/node-red-contrib-color-convert) node.

I will be refactoring my flow to account for the new setting and will post updates later.

stoinov, 2020-01-12:
OK so I've tested the new functionality without `cached_state` and it does most of what I want. The question about the changing x and y when using `color_temp` [has been asked already](https://github.com/Koenkk/zigbee2mqtt/issues/2335#issuecomment-573463410). But I hit a [potential new bug](#2731), for which I had to implement the workaround on line 46 in my script.

```
[{"id":"45e9dcd.6c9cfa4","type":"debug","z":"53e84a0a.4a9cf4","name":"HomeKit","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","x":240,"y":340,"wires":[]},{"id":"6d446569.835b8c","type":"homekit-service","z":"53e84a0a.4a9cf4","isParent":true,"bridge":"8d4739c2.39c0d","accessoryCategory":"LIGHTBULB","parentService":"","name":"LED Strip","serviceName":"Lightbulb","topic":"","filter":false,"manufacturer":"Gledopto","model":"GL-MC-001","serialNo":"00124b001ee96a78","cameraConfigVideoProcessor":"ffmpeg","cameraConfigSource":"","cameraConfigStillImageSource":"","cameraConfigMaxStreams":2,"cameraConfigMaxWidth":1280,"cameraConfigMaxHeight":720,"cameraConfigMaxFPS":10,"cameraConfigMaxBitrate":300,"cameraConfigVideoCodec":"libx264","cameraConfigAudioCodec":"libfdk_aac","cameraConfigAudio":false,"cameraConfigPacketSize":1316,"cameraConfigVerticalFlip":false,"cameraConfigHorizontalFlip":false,"cameraConfigMapVideo":"0:0","cameraConfigMapAudio":"0:1","cameraConfigVideoFilter":"scale=1280:720","cameraConfigAdditionalCommandLine":"-tune zerolatency","cameraConfigDebug":false,"cameraConfigSnapshotOutput":"disabled","characteristicProperties":"{\n    \"Hue\":true,\n    \"Saturation\":true,\n    \"Brightness\":true,\n    \"ColorTemperature\":true\n}","x":80,"y":380,"wires":[["45e9dcd.6c9cfa4","8ad8870c.a461a8"],[]]},{"id":"8ad8870c.a461a8","type":"function","z":"53e84a0a.4a9cf4","name":"HK>Z2M","func":"var Hue = flow.get('Hue')||0;\nvar Saturation = flow.get('Saturation')||0;\nvar Brightness = flow.get('Brightness')||100;\nvar topic = 'topic/light/set'\n\nconst Properties = Object.keys(msg.payload);\n//node.warn(Object.keys(Properties).length);\nif (Properties.length > 1){\n    msg.payload = \"Unknown payload! \" + msg.payload;\n    return msg;\n}\n\nfor (const entry of Properties) {\n    switch(entry) {\n        case \"On\":\n            msg.topic = topic+\"/state\";\n            if(msg.payload.On){\n                msg.payload = \"on\";\n            }\n            else {\n                msg.payload = \"off\";\n            }\n            break;\n        case \"Brightness\":\n            Brightness = msg.payload.Brightness;\n            flow.set('Brightness',Brightness);\n            msg.topic = topic+\"/brightness\";\n            msg.payload = Math.round(Brightness * 1.55)+15;\n            break;\n        case \"ColorTemperature\":\n            ColorTemperature = msg.payload.ColorTemperature;\n            flow.set('ColorTemperature',ColorTemperature);\n            msg.topic = topic+\"/color_temp\";\n            msg.payload = ColorTemperature;\n            break;\n        case \"Hue\":\n            Hue = msg.payload.Hue;\n            flow.set('Hue',Hue);\n            msg.topic = topic;\n            msg.payload = {\"color\" : {\"hue\" : Hue}};\n            break;\n        case \"Saturation\":\n            Saturation = msg.payload.Saturation;\n            flow.set('Saturation',Saturation);\n            msg.topic = topic;\n            msg.payload = {\"color\" : {\"hue\" : Hue, \"saturation\" : Saturation}};\n            break;\n        default:\n    } \n}\nreturn msg;","outputs":1,"noerr":0,"x":240,"y":380,"wires":[["6b950306.2f15bc","437b555b.dc186c"]]},{"id":"6b950306.2f15bc","type":"debug","z":"53e84a0a.4a9cf4","name":"MQTT","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","x":410,"y":340,"wires":[]},{"id":"437b555b.dc186c","type":"mqtt out","z":"53e84a0a.4a9cf4","name":"MQTT","topic":"","qos":"0","retain":"false","broker":"308ecfd8.746ff8","x":410,"y":380,"wires":[]},{"id":"7aef1c6d.f76f0c","type":"mqtt in","z":"53e84a0a.4a9cf4","name":"LED input","topic":"topic/light/+","qos":"2","datatype":"auto","broker":"308ecfd8.746ff8","x":80,"y":340,"wires":[["45e9dcd.6c9cfa4"]]},{"id":"8d4739c2.39c0d","type":"homekit-bridge","z":"","bridgeName":"Home","pinCode":"111-11-111","port":"","allowInsecureRequest":false,"manufacturer":"Home","model":"Beta Home","serialNo":"00001","customMdnsConfig":false,"mdnsMulticast":true,"mdnsInterface":"","mdnsPort":"","mdnsIp":"","mdnsTtl":"","mdnsLoopback":true,"mdnsReuseAddr":true,"allowMessagePassthrough":true},{"id":"308ecfd8.746ff8","type":"mqtt-broker","z":"","name":"Home TCP","broker":"home","port":"1883","clientid":"node-red","usetls":false,"compatmode":false,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]
```

BTW @TilmanK your previous post with the script does not paste properly. Ty enclosing it in triple grave characters like those ```.

sjorge, 2020-03-13:
Perhaps we should fix this yeah,...

I was fixing some enhancedHue vs hue stuff last weekend and the code is a bit ugh...

xy || (hue || saturation || (hue && saturation) || rgb || r + g + b || h + s + (b  || v)) combos basically are all accepted and going to one converter.

That then does roughtly the following:
- xy
  - set x and y as received
  - send the zcl frame to change the bulb
  - read x and y
- hue
  - set hue as received
  - send zcl frame to change bulb
  - read x and y
- hue + saturation
  - set hue as received
  - set saturatio nas received
  - send zcl frame to change bulb
  - read x and y
...

I fixed it to differentiat between setting enhancedHue or normal hue and to also now read hue and saturation, most bulbs don't return the later at all... so only x and y are set/update.

If reporting is in play, those can update the values too.

This has some fun results on bulbs that support xy and hue/sat (e.g. innr) where you can set one but not get both updated back.

I'm not really sure how we can fix this and make this more unified without breaking some existing behaviors though... an idea I had that I abandoned was the following:

- accept xy, hue, sat, hue+sat, ... as input
- have a 'output' format configurable in the configuration.yaml

- set via whatever we do now, store and internal 'last set methode' which can be xy, hue/sat or color_temp
- use that internal meta data to do the read of either:
  - currentX + currentY
  - currentHue + currentSaturation
  - currentTemperature (forgot the zcl name for this one)
  - if set_methode == color_temp
    - return color_temp
    - remove others
  - if set_methode != color_temp
    - return xy if output == xy
    - return hue + saturation if output == hue/sat

This would require us to potentialy do some conversions, xy -> hue/sat and hue/sat -> xy to have the proper output.

But this would make it consitant, for homekit we can set output = hue/sat and just pass it along as is. 

This introduces a new problem though :( you can have a bulb that only does xy (most) but you might want to set it with hue/saturation... do we also handle conversion for those? That will make the setting code super complex.

As we need to keep track of a bulb support xy, hue/sat or enhancedHue/sat.
Based on that we can set the value 'as-is' or we might need to convert, e.g. hue/sat provided but only xy supported... *sighs in frustration*

I don't think there is an easy fix for this tbh. But that doesn't mean we should form a plan of action and try and implement it...

But basically, unstale as this is confusing and not really ideal behavior as-is currently.

closed 2020-05-20 by stale[bot]