Koenkk/zigbee2mqtt #2722: Failed to start:  SREQ '--> SYS - osalNvWrite (...) failed with status '2' (expected '0')

odx, 2020-01-11:
# Bug Report
## What happened
I recently set up a CC1352P-S as new Zigbee network with a new Raspi. Unfortunately the power supply broke down within the first 24 hrs of operation. It looks like only the supply is damaged as I still can boot the pi and all services launch as expected.

Only zigbee2mqtt refuses to start up. I can't tell if the issue described below is a result of the failure or derives from the general behavior of my configuration.

Here is a debug log of the startup process:
https://pastebin.com/9rB1DtN3

I tries refreshing the coordinator which verifies fine.

## What did you expect to happen
zigbee2mqtt should start up.

## How to reproduce it (minimal and precise)
No idea.

## Debug Info
zigbee2mqtt version: 1.8.0 in docker
CC1352P-2 firmware version: CC1352P_2_20191106

Koenkk, 2020-01-12:
Probably your cc1352 got corrupted, reflashing should fix it.

odx, 2020-01-12:
I actually did so already. (Autocorrection changed reflashing to refreshing...)

What is also tried is to remove all persisted data except the configuration.

odx, 2020-01-12:
I have a CC2531 that I could cross check. Does that make sense?

Koenkk, 2020-01-12:
did you also execute step 5 when reflashing (https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html), this is important when you have this error.

odx, 2020-01-12:
Thanks I was not aware about those instructions. I tinkered my own way to flash which basically used the default settings to load the program in uniflash.

```
zigbee2mqtt    | zigbee2mqtt:info  2020-01-12 20:48:57: zigbee-herdsman started
```

closed 2020-01-12 by odx