Koenkk/zigbee2mqtt #2724: Forcing a flush of configuration.yaml

mikeage, 2020-01-12:
# Bug Report
## What happened
I accidentally erased (well, ran `git checkout .`) my configuration.yaml, but zigbee2mqtt is still running. Is there any way to tell it to dump the current configuration to the file?

## What did you expect to happen
n/a

## How to reproduce it (minimal and precise)
n/a

## Debug Info
zigbee2mqtt version: 260d290
CC253X firmware version: n/a

Koenkk, 2020-01-13:
Unfortunately this is not possible

mikeage, 2020-01-13:
ok, thanks! I'll try tonight

mikeage, 2020-01-13:
I wasn't clear on whether I had to first create the group then assign a device, so I tried something else; setting `last_seen` to `disable` did the trick.

`mosquitto_pub -t zigbee2mqtt/bridge/config/last_seen -m 'disable' -h localhost -p 1883`

Thanks!

closed 2020-01-13 by mikeage