Koenkk/zigbee2mqtt #2726: Error while starting zigbee-herdsman

glsf91, 2020-01-12:
# Bug Report
## What happened
I have installed zigbee2mqtt for the first time on Ubuntu 16.0.4.

When starting I get:

`zigbee2mqtt:info  2020-01-12 14:17:39: Logging to console and directory: '/home/john/zigbee2mqtt/data/log/2020-01-12.14-17-39'

zigbee2mqtt:info  2020-01-12 14:17:39: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
zigbee2mqtt:info  2020-01-12 14:17:39: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-01-12 14:18:40: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-01-12 14:18:40: Failed to start zigbee
zigbee2mqtt:error 2020-01-12 14:18:40: Exiting...
zigbee2mqtt:error 2020-01-12 14:18:40: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/home/john/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
`

I tried reflashing and also another CC2531 but still with the same error.

Searched for a few hours without success. Also seen the FAQ.

## Debug Info
zigbee2mqtt version: I use zigbee2mqtt 1.8.0 master.
CC253X firmware version:  I have CC2531 flashed from  CC2531_SOURCE_ROUTING_20190619.zip but also tried CC2531_DEFAULT_20190608.zip. 

Debug log: [https://pastebin.com/mbTPeWXs](url)

glsf91, 2020-01-12:
I think I solved it. 
I saw somewhere on another about pressing the button close to the usb port. Then the green light goes off.  When I do this I don't get the error anymore.

Also don't know what this and the other button does?
Maybe add this to the instruction / FAQ?

Koenkk, 2020-01-13:
This is the reset button, added it to the FAQ. thanks! (assuming this can be closed now).

closed 2020-01-13 by Koenkk

djpollum, 2020-12-16:
I have the same issue, but pushing the restart button doesn't solve my problem.
Any other ideas?

>Zigbee2MQTT:info  2020-12-16 22:08:22: Logging to console and directory: '/share/zigbee2mqtt/log/2020-12-16.22-08-22' filename: log.txt
Zigbee2MQTT:info  2020-12-16 22:08:22: Starting Zigbee2MQTT version 1.16.1 (commit #1.16.1)
Zigbee2MQTT:info  2020-12-16 22:08:22: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-12-16 22:09:24: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-12-16 22:09:24: Failed to start zigbee
Zigbee2MQTT:error 2020-12-16 22:09:24: Exiting...
Zigbee2MQTT:error 2020-12-16 22:09:24: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR!
 zigbee2mqtt@1.16.1 start: node index.js
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-12-16T21_09_24_445Z-debug.log
2020-12-16T22:09:24: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-12-16T22:09:24: PM2 log: App [npm:0] starting in -fork mode-
2020-12-16T22:09:24: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.1 start /zigbee2mqtt-1.16.1
> node index.js`

matthloplop, 2021-01-07:
+1

I have the same issue. 

@glsf91 do you have the correct method of pushing the button?

When I push the button, the light stay green but the error message change...
From:

Zigbee2MQTT:error 2021-01-07 23:09:20: Error: AREQ - ZDO - simpleDescRsp after 10000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1

To:
Zigbee2MQTT:error 2021-01-07 23:14:51: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1

Thanks for help

glsf91, 2021-01-08:
> +1
> 
> I have the same issue.
> 
> @glsf91 do you have the correct method of pushing the button?

It was a long time ago. See https://github.com/Koenkk/zigbee2mqtt.io/commit/277a1d32a11d295135c409dcba1907999af493d5

hvalev, 2021-01-20:
+1. I'm facing the same issue. The difference being I'm using the CC2652 stick

G4rpez, 2021-02-04:
Same issue with CC2652RB. If I turn off all no-battery devices (Ikea lights or zigbee smart plugs) zigbee2mqtt service restarts but all devices must be re paired (even after changed pan_id)

hvalev, 2021-02-04:
I forgot about that. Yes, that resolved it for me as well. I had to bring the whole network down, start the coordinator and re-pair everything. I wasn't aware that some of my switches also act as routers and I believe they were keeping the network alive on the same pan id, which prevented the coordinator from recreating the network.

Gio-dot, 2021-02-16:
> Same issue with CC2652RB. If I turn off all no-battery devices (Ikea lights or zigbee smart plugs) zigbee2mqtt service restarts but all devices must be re paired (even after changed pan_id)

Very strange behaviour! What was the origin of that error?

bojanpotocnik, 2021-03-11:
The same happens for me every time I restart zigbee2mqtt (using zzh/CC2652R), however I don't need to repair all devices. I just need to turn off all routers, wait for zigbee2mqtt to start, and then I can power routers back and everything eventually works. If I don't turn off the routers I consistently get `AREQ - ZDO - stateChangeInd after 60000ms` every few seconds.

Koenkk, 2021-03-11:
@bojanpotocnik that looks exactly the same as https://github.com/Koenkk/zigbee2mqtt/issues/6302 . Try flashing your stick on a different PC (with different python version). For an unknown reason setups incorrectly flash the firmware.

bojanpotocnik, 2021-03-26:
Today it happened again, and in this moment I realized that I'm flashing my CC2652R coordinator following [official guide](https://electrolama.com/radio-docs/#step-3-flash-the-firmware-on-your-stick) using TI's Flash Programmer 2, and I flashed routers following your guide using SmartRF Flash Programmer, so no Python involved. Using full readback verify in all cases.

It also happened just now when I updated coordinator to `CC2652R_coordinator_20210120.hex` - after plugging it back in, zigbee2mqtt fails to start
```
Zigbee2MQTT:error 2021-03-26 23:51:38: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
```
until routers are turned off. I will try to reflash and repair routers, otherwise I will try using CC2652R/P2 as routers instead of CC2530.

Huh-David, 2022-01-25:
For me changing the USB port (from 3.0 to 2.0) did the trick. Maybe that helps someone

gruennie, 2025-03-18:
For me, it helped to connect the Zigbee stick not directly to the Raspberry's USB port, but further away using a USB extension cable. Probably too much interference with the router, etc.