Koenkk/zigbee2mqtt #2727: Philips Hue outdoor Impress not supported

steveddt, 2020-01-12:
Any possibility to add the device Hue  Outdoor Impress 
I want to add it by the actions to follow for unsupported devices but don't know how to retrieve the functionality for using also the colors and brightness of the Philips Hue.

My logging:

zigbee2mqtt:info  2020-01-12 09:05:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0017880104f95f19"}}'
zigbee2mqtt:debug 2020-01-12 09:05:20: Device '0x0017880104f95f19' announced itself
zigbee2mqtt:debug 2020-01-12 09:05:20: Received Zigbee message from '0x0017880104f95f19', type 'readResponse', cluster 'genBasic', data '{"modelId":"1743230P7","manufacturerName":"Philips","powerSource":1}' from endpoint 11 with groupID 0
zigbee2mqtt:warn  2020-01-12 09:05:20: Received message from unsupported device with Zigbee model '1743230P7'
zigbee2mqtt:warn  2020-01-12 09:05:20: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-01-12 09:05:20: Received Zigbee message from '0x0017880104f95f19', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"appVersion":2,"stackVersion":1}' from endpoint 11 with groupID 0
zigbee2mqtt:warn  2020-01-12 09:05:20: Received message from unsupported device with Zigbee model '1743230P7'
zigbee2mqtt:warn  2020-01-12 09:05:20: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-01-12 09:05:20: Received Zigbee message from '0x0017880104f95f19', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0,"dateCode":"20181115","swBuildId":"1.46.13_r26312"}' from endpoint 11 with groupID 0
zigbee2mqtt:warn  2020-01-12 09:05:20: Received message from unsupported device with Zigbee model '1743230P7'
zigbee2mqtt:warn  2020-01-12 09:05:20: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:info  2020-01-12 09:05:20: Successfully interviewed '0x0017880104f95f19', device has successfully been paired
zigbee2mqtt:warn  2020-01-12 09:05:20: Device '0x0017880104f95f19' with Zigbee model '1743230P7' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:info  2020-01-12 09:05:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x0017880104f95f19","supported":false}}'

Hopefully this device is already existing somewhere for Zigbee2mqtt.

sjorge, 2020-01-12:
You can try adding a new entry to devices.js

```
    {
        zigbeeModel: ['1743230P7'],
        model: '1743230P7',
        vendor: 'Philips',
        description: 'Hue Outdoor Impress',
        extend: hue.light_onoff_brightness_colorxy,
    },
```

steveddt, 2020-01-13:
Thank you very much for your respond. It seems that I don't have a devices.js file on the disk. 
I have to created the file? Store on a folder?

sjorge, 2020-01-13:
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html, section 2. should get you further.

Koenkk, 2020-01-13:
Added support for it, will be supported in the latest dev branch in a few hours. Thanks!

closed 2020-01-13 by Koenkk

steveddt, 2020-01-18:
Hello Koen, was not good when trying in the latest dev branch. Maybe because I don't know how to do this. I just saw that the production environment had an update to version 1.9. Just installed it and saw that the lamp is now working! Thank you a lott.