Koenkk/zigbee2mqtt #2728: IKEA TRADFRI  E1743 unsupported

joaocmfcm, 2020-01-12:
Hello there,
I flawlessy added the IKEA E1743 (https://www.zigbee2mqtt.io/devices/E1743.html) to zigbee2mqtt, the device is in the devices.yaml, but in the zigbee2mqtt logs I get the following on startup and when I press a button on the switch: 
```
> zigbee2mqtt@1.8.0 start /zigbee2mqtt-1.8.0
> node index.js
zigbee2mqtt:info  2020-01-12 16:23:53: Logging to console and directory: '/share/zigbee2mqtt/log/2020-01-12.16-23-53'
zigbee2mqtt:info  2020-01-12 16:23:54: Starting zigbee2mqtt version 1.8.0 (commit #unknown)
zigbee2mqtt:info  2020-01-12 16:23:54: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-01-12 16:23:55: zigbee-herdsman started
zigbee2mqtt:info  2020-01-12 16:23:55: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:info  2020-01-12 16:23:55: Currently 7 devices are joined:
zigbee2mqtt:info  2020-01-12 16:23:55: ikea_switch_1 (0xccccccfffe364d13): Not supported (EndDevice)
...
...
zigbee2mqtt:warn  2020-01-12 16:25:01: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-01-12 16:25:01: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

What am I doing incorrectly?
Thanks in advance,
Cheers

Venice-89, 2020-01-13:
hmm I also have that device and don't have any problems with that button device. 
I don't see any error message in my logs.

I have Z2M 1.8.0 as HASS.IO Addon

joaocmfcm, 2020-01-13:
Issue solved!

1. Forced the device removal
2. Stopped zigbee2mqtt on hass.io
3. Removed the coordinator usb pen from the raspberry pi
4. Reattached the coordinator
5. Restarted zigbee2mqtt
6. Re-paired the E1743 again
7. Worked!

Koenkk, 2020-01-13:
Great! (assuming this can be closed now).

closed 2020-01-13 by Koenkk