Koenkk/zigbee2mqtt #2733: zigbee2mqtt Cannot restore backup

nacherasgb, 2020-01-13:
A question?
Read the documentation: https://www.zigbee2mqtt.io/

Bug report:
- First make sure you are on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.
-->

# Bug Report
Hi! First of all, this is the first time Im trying this device so forgive if it is an stupid question :) 

Im using the version 1.8.0 of zigbee2mqtt in Hass.io and the latest revision of the 1.2 version for the CC2531 hardware

## What happened
When trying to start the add-on zigbee2mqtt, it gets aborted due to that there is an error while starting zigbee-herdsman, where it cannot restart the backup, see the following line from the log:

Error: Cannot restore backup, backup is for 'zStack30x', current is 'zStack12'
    at Object.<anonymous> (/zigbee2mqtt-1.8.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/backup.js:158:19)

Find the entire log here: https://hastebin.com/mubidolaji.sql 

Observe that the first firmware that I fasted into the CC2531 was the 3.0 version but then I reflashed it again but this time with the 1.2 version

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: 1.2 coordinator CC2531

many thanks in advance!!

nacherasgb, 2020-01-13:
Configuration used:

{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "http://192.168.0.111:1883",
    "user": "zigbeeuser",
    "password": "password"
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blacklist": []
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1,
    "initialdelay": 1,
    "options": "-d -d",
    "log": false
  }
}

Koenkk, 2020-01-14:
Delete the `coordinator_backup.json` in the zigbee2mqtt data directory fixes it (same directory where `configuration.yaml` is).

nacherasgb, 2020-01-14:
it worked! thanks so much!!

closed 2020-01-14 by nacherasgb