Koenkk/zigbee2mqtt #2734: AV2010/32 does not support MQTT discovery properly

y0ast, 2020-01-13:
# Bug Report
## What happened

AV2010/32 does not have correct MQTT discovery. Only the linkquality is discovered but the HA interface does show:

```
occupied heating setpoint 22
linkquality 94
local temperature 23
```
In the linkquality sensor combined. Also the z2m logs show these values.

From previous PRs on this device I have seen that more values can come up. The documentation also mentions way more values: https://www.zigbee2mqtt.io/devices/AV2010_32.html

## What did you expect to happen

All sensors are automatically discovered

## How to reproduce it (minimal and precise)

Pair AV2010/32 and check the mqtt discovery

## Debug Info
zigbee2mqtt version: 1.8.0
CC253X firmware version: `'{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'`

nldroid, 2020-01-14:
Can you look in this file:
lib/extension/homeassistant.js

And find the line: 
`'AV2010/32': [],`

It's empty in current version. You can try to add things you expect and see if it works. If it works, report it here.

y0ast, 2020-01-14:
Unfortunately the screen of my AV2010/32 broke already, so I won't be able to work on this further. If anyone else is interested they can take this up, otherwise I'll report back when/if I get a new one.

y0ast, 2020-03-01:
closed by #3038

closed 2020-03-01 by y0ast