Koenkk/zigbee2mqtt #2737: zigbee-herdsman ts files not building to js

Arminhosseini, 2020-01-14:
Hi,
I made some changes in zigbee-herdsman src file and push it to my git
and change package.json in zigbee2mqtt that take zigbee-herdsman file from my git but when I'm running npm install zigbee-herdsman typescript files not building to nodejs. and dist file not created.

Koenkk, 2020-01-14:
After changing the ts files you need to compile them with `npm run build`, closing this as it is not a zigbee2mqtt issue.

closed 2020-01-14 by Koenkk

Arminhosseini, 2020-01-14:
I know but why when I run npm install command with your repo its doing it by itself?

Arminhosseini, 2020-01-14:
and another problem is when I clone zigbee2mqtt and running npm install zigbee-herdsman module is created but there is no src file there that I can run npm run build command.

Koenkk, 2020-01-14:
The files are compiled by azure pipelines, the package published to npm.js only contains the `dist` (no `src`). 

To setup local development environment.
- Go to zigbee2mqtt dir
- `cd node_modules`
- `rm -rf zigbee-herdsman`
- `git clone https://github.com/Koenkk/zigbee-herdsman.git` (but then your own repo)
- `cd zigbee-herdsman`
- `npm install`
- Now change the `.ts` files
- `npm run build`

When now starting zigbee2mqtt it will use the new files

Arminhosseini, 2020-01-14:
thank's