Koenkk/zigbee2mqtt #2738: ikea bulb E27 WW 806lm not working

jerome70, 2020-01-14:
Bug report:

bulbs paired easyly in zigbee2mqtt and domoticz but signal at 0 in domotocz and nothing appens when pressing button in domoticz.

- zigbee2mqtt version 1.8.0-dev (commit #9890fb3)
Coordinator firmware version: '{"type":"zStack12","meta":20190619

message in log :

error 2020-01-14 14:01:15: Publish 'set' 'state' to 'chaufferie_1' failed: 'Error: Timeout - 6501 - 1 - 193 - 6 - 11 after 15000ms'
debug 2020-01-14 14:01:15: Error: Timeout - 6501 - 1 - 193 - 6 - 11 after 15000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
info  2020-01-14 14:01:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'chaufferie_1' failed: 'Error: Timeout - 6501 - 1 - 193 - 6 - 11 after 15000ms'","meta":{"friendly_name":"chaufferie_1"}}'

Thanks for your work

jerome70, 2020-01-23:
What may i do else ?

martonmodis13, 2020-01-23:
Buy a Male to female USB extension cable,and use it.This is solved the problem for me.

jerome70, 2020-01-26:
Thanks for your answer,
I allready have an USB extension cable and the last firmware.
Does someone have this buld workind with domoticz ?

closed 2020-04-02 by stale[bot]