Koenkk/zigbee2mqtt #2740: reporting error on BASICZBR3 - Sonoff switch

MaZZly, 2020-01-14:
# Bug Report
## What happened

Recently upgraded to better adapter (CC26X2R1). Enabled reporting (didn't have enabled previously on CC2531)
Seeing this in the logs every time I'm starting zigbee2mqtt:
```
zigbee2mqtt:error 2020-01-14 18:27:52: Failed to setup reporting for '0x00124b001e732769' - Error: Timeout - 4989 - 1 - 2 - 6 - 7 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/zigbee2mqtt-1.8.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
```

## What did you expect to happen

Not seeing this error :)

## How to reproduce it (minimal and precise)

Restart mqtt

## Debug Info

device in database.db:
```javascript
{"id":32,"type":"Router","ieeeAddr":"0x00124b001e732769","nwkAddr":4989,"manufId":0,"manufName":"SONOFF","powerSource":"Mains (single phase)","modelId":"BASICZBR3","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"BASICZBR3","manufacturerName":"SONOFF","powerSource":1,"zclVersion":1,"hwVersion":1,"dateCode":"20190907-V102"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca156a8","endpointID":1}]}},"hwVersion":1,"dateCode":"20190907-V102","zclVersion":1,"interviewCompleted":true,"meta":{}}
```

zigbee2mqtt version: 1.8.0
CC26X2R1 firmware version: 20191106 Z-Stack_3.x.0

Erelen-Laiquendi, 2020-01-15:
Same issue: https://github.com/Koenkk/zigbee-herdsman-converters/issues/902

Koenkk, 2020-01-16:
Devices indeed doesn't seem to support it, fixed in latest dev branch.

closed 2020-01-16 by Koenkk