Koenkk/zigbee2mqtt #2744: Cannot pair Osram Smart+ B22 A60 Lightbulbs

mantis64, 2020-01-14:
# Bug Report
Cannot pair with bulbs.

## What happened
I get following error in logs. 
zigbee2mqtt:error 2020-01-14 20:03:37: Failed to interview '0x7cb03eaa00adb76c', device has not successfully been paired

## What did you expect to happen
Hoping that these bulbs would connect

## How to reproduce it (minimal and precise)
I did the reset procedure for the bulbs and zigbee2mqtt can see them but they will not pair. Have reset the bulbs numerous times as per the instructions for Osram bulbs and I get a blink to confirm they have been reset

## Debug Info
zigbee2mqtt version: 1.8.0 
CC253X firmware version: transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608

I am running this on Raspberry PI 3b.  Running on Hassio

Please can you helpm I am also a newbie!

sjorge, 2020-01-14:
I managed to pair one of my test osram bulbs when holding it 1cm away from the coordinator (the base section just above the metal.

mantis64, 2020-01-14:
I tried this, I taped the coordinator to the shade, it was around 10cm away.  It did not work.  If you are saying I need to have it closer, I will try this now.

mantis64, 2020-01-14:
No still not working. This is a B22 bulb and not an E27 bulb that is in the supported list.   I didn't think it would make a difference.  But perhaps Osram have changed the firmware or something.  Perhaps I need to send these bulbs back, but B22 zigbee bulbs are hard to find.  Hope you can suggest something else?

sjorge, 2020-01-14:
Sadly now, it took me over an hour to get it paired for me. It was practictically touching the coordinator when it finally worked.

I would have returned it if I was actually going to use it. But since it is a bulb to test and compare with I kept it.

I just tried to pair and all 5 attempts failed, it’s definitely not you. It might be the newer zigbee 3.0 firmware that shipped on mine.

~ sjorge

mantis64, 2020-01-15:
Thanks for coming back to me.  Yes I finally got one of the 2 bulbs I have paired!  However I cannot get the other one paired, so will try again over the next few days since the first one took so long.  However once paired the range is awful, I literally have to have the stick below the bulb.  Is this normal?  I've ordered an antenna to solder to the sniffer board, will this make it better, and is this the issue with range and pairing?

mantis64, 2020-01-15:
Also would the newer Zigbee be an issue for the CC2531 or Zigbee2mqtt?

mantis64, 2020-01-15:
Can you recommend any other bulbs which are B22, have good brightness and full colour and that will work better with the hardware?

mantis64, 2020-01-16:
Ok I got some Xiaomi sensors and they work fine and have great range even without an antenna.  So I can only conclude that the issue is the bulbs.  I think you should remove the Osram compatibility since they do not work with zigbee2mqtt.  I bought these bulbs based on the fact they were on the compatible devices list.  Hopefully no one else should buy these bulbs

closed 2020-05-23 by stale[bot]