Koenkk/zigbee2mqtt #2745: Smart Home Pty HGZB-20-DE not supported with version 1.8.0?

nacherasgb, 2020-01-14:
# Bug Report
## What happened
Trying to pair the socket Smart Home Pty HGZB-20-DE (included in the "supported devices list" (https://www.zigbee2mqtt.io/devices/HGZB-20-DE.html). However, even though it says it is paired, it also says that this device is not supported. Do you know how to work with it in that case? or if there is any workaround to fix this?

See the log file here: https://hastebin.com/racalibire.sql

thanks in advance!

## Debug Info
zigbee2mqtt version: 1.8.0
CC2531 firmware version:  latest revision of the 1.2 version

nacherasgb, 2020-01-14:
I tried the manual home assistant configuration, but it didn't work

switch:
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/<FRIENDLY_NAME>"
    availability_topic: "zigbee2mqtt/bridge/state"
    payload_off: "OFF"
    payload_on: "ON"
    value_template: "{{ value_json.state }}"
    command_topic: "zigbee2mqtt/<FRIENDLY_NAME>/set"

sensor:
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/<FRIENDLY_NAME>"
    availability_topic: "zigbee2mqtt/bridge/state"
    unit_of_measurement: "-"
    value_template: "{{ value_json.linkquality }}"

nldroid, 2020-01-15:
Can you send commands to it with a MQTT client? (http://mqtt-explorer.com/)

nacherasgb, 2020-02-13:
no it doesn't work with the latest update of the zigbee2mqtt
thanks a lot!!

closed 2020-02-13 by nacherasgb