Koenkk/zigbee2mqtt #2749: Error: SREQ '--> AF - dataRequest failed with status '17' (expected '0')

TilmanK, 2020-01-15:
# Bug Report
I had this messages over and over again after restarting Z2M:

debug 2020-01-15 17:34:39: Publishing 'set' 'state' to 'Küche Oberschränke 1'
error 2020-01-15 17:34:39: Publish 'set' 'state' to 'Küche Oberschränke 1' failed: 'Error: SREQ '--> AF - dataRequest - {"dstaddr":28433,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":45,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,36,0]}}' failed with status '17' (expected '0')'
debug 2020-01-15 17:34:39: Error: SREQ '--> AF - dataRequest - {"dstaddr":28433,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":45,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,36,0]}}' failed with status '17' (expected '0')
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)

I do not have herdsman DEBUG messages, but some pcap traces (more or less on accident). Can you please check the error? If you need DEBUG traces I'll try to get some.

[logs.zip](https://github.com/Koenkk/zigbee2mqtt/files/4066096/logs.zip)
## Debug Info
zigbee2mqtt version: 1.8.0-dev (last pull on 12.01)
CC253X firmware version: 20191106

Koenkk, 2020-01-17:
This means that your coordinator has memory issues (which i found strange because you seem to have a 1352 or 2652?). I havent seen this yet for these coordinators.

Maybe you can try reflashing and erase whole the memory, especially pay attention to step 5 of https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html

Repairing isnt necessary as the `coordinator_backup.json` will be restored.

TilmanK, 2020-01-17:
A 2652, yes. This was directly after a Z2Mrestart and fixed itself 2 minutes after the start or so. Does this help? I'd rather not try to alter anything if not absolutely necessary since I got everything working like a charm right now ;)

Koenkk, 2020-01-17:
Yes I understand, let's leave this open for now to see if anyone else also experiences it.

airdrummingfool, 2020-01-23:
I'm encountering a possibly similar(?) issue on my network:
```
(node:16) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - extRouteDisc - {"dstAddr":47821,"options":2,"radius":30}' failed with status '17' (expected '0')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
```

Originally I noticed I was having problems with a certain model of Zigbee lightbulb (my Sengled `E11-G13`s), while all other devices on my network seemed to work fine. A couple days into debugging, my CC2531 crashed and wouldn't come back up (tried restarting z2m, unplugging and replugging, etc), so I re-flashed with the 1.2 coordinator source routing firmware (was previously running default). Since then there seem to be a lot more issues with message deliverability, with `extRouteDisc` error messages similar to the one I posted above showing up every so often.

I also updated my core z2m from 1.8 to 1.9-dev around the same time, though I doubt that's the cause.

I'm running 22 devices, including 5-6 that should act as repeaters. Is it possible that my CC2531 is getting overwhelmed?

If this is a separate thing, please let me know and I'll move it to its own issue.

Koenkk, 2020-01-23:
@airdrummingfool are you already running the source routing firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

airdrummingfool, 2020-01-23:
@Koenkk yes, I flashed over to that after the 2531 crashed. However, things seem to be worse after switching to it.

Koenkk, 2020-01-23:
@airdrummingfool do you use the availability or report feature?

airdrummingfool, 2020-01-23:
@Koenkk No, I do not. There is likely a decent amount of traffic generated by the 5 power reporting smart plugs (mostly Iris 3210-L) that I have, though I haven't added or changed these in any way in months.

Koenkk, 2020-01-23:
Does it only happen for the `extRouteDisc ` messages? Can you provide the complete herdsman debug logging?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

TilmanK, 2020-01-23:
FYI: I just checked the logs over the past few days. Since the original issue I reported, this hasn't occurred again. I haven't touched the coordinator, the system is up since 13 days (so it was 5 days up when the issue happened).

airdrummingfool, 2020-01-24:
@Koenkk Anecdotally, every error that I've noticed was an `extRouteDisc` error.

I enabled herdsman debug logging, but it is _super_ noisy. I'm not sure any of this is relevant, but here's a 90-second sample of what the logger outputs: https://pastebin.com/XvLvrZEQ

The only thing that sticks out is a repeated (180 times in 90 seconds) read error, but I'm not sure that it's actually an error (seems to be just a 'no data available' message):
```
2020-01-24T08:34:31.036Z serialport/bindings/unixRead Starting read
2020-01-24T08:34:31.037Z serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  syscall: 'read'
}
2020-01-24T08:34:31.037Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
2020-01-24T08:34:31.038Z serialport/bindings/poller Polling for "readable"
```

I'll work on getting some debug messages from around a promise rejection.

airdrummingfool, 2020-01-25:
I've been collecting data for a bit, and so far I'm still only seeing `extRouteDisc` errors. Here's a couple with some lines of herdsman debug around them for context:

```
2020-01-25T04:17:22.563Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
2020-01-25T04:17:22.563Z serialport/bindings/poller Polling for "readable"
2020-01-25T04:17:22.567Z serialport/bindings/poller received "readable"
2020-01-25T04:17:22.567Z serialport/bindings/unixRead Starting read
2020-01-25T04:17:22.567Z serialport/bindings/unixRead Finished read 6 bytes
2020-01-25T04:17:22.567Z serialport/stream binding.read finished { bytesRead: 6 }
2020-01-25T04:17:22.568Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,17,48]
2020-01-25T04:17:22.568Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,17,48]
2020-01-25T04:17:22.568Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [17] - 48
2020-01-25T04:17:22.568Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":17}
2020-01-25T04:17:22.568Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-25T04:17:22.569Z serialport/stream _read reading { start: 44446, toRead: 21090 }
2020-01-25T04:17:22.569Z serialport/binding-abstract read
2020-01-25T04:17:22.569Z serialport/bindings/unixRead Starting read
(node:16) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - extRouteDisc - {"dstAddr":39811,"options":2,"radius":30}' failed with status '17' (expected '0')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 80)
2020-01-25T04:17:22.572Z serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  syscall: 'read'
}
2020-01-25T04:17:22.573Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
2020-01-25T04:17:22.573Z serialport/bindings/poller Polling for "readable"
2020-01-25T04:17:23.545Z serialport/bindings/poller received "readable"
2020-01-25T04:17:23.545Z serialport/bindings/unixRead Starting read
2020-01-25T04:17:23.545Z serialport/bindings/unixRead Finished read 33 bytes
2020-01-25T04:17:23.546Z serialport/stream binding.read finished { bytesRead: 33 }
2020-01-25T04:17:23.546Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,4,11,131,155,1,1,0,42,0,47,145,204,0,0,8,8,79,10,8,5,33,73,1,39,228,27,111]
2020-01-25T04:17:23.546Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,4,11,131,155,1,1,0,42,0,47,145,204,0,0,8,8,79,10,8,5,33,73,1,39,228,27,111]
2020-01-25T04:17:23.546Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,4,11,131,155,1,1,0,42,0,47,145,204,0,0,8,8,79,10,8,5,33,73,1,39,228,27] - 111
2020-01-25T04:17:23.546Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":39811,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":42,"securityuse":0,"timestamp":13406511,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[8,79,10,8,5,33,73,1]}}
```
---
```
2020-01-25T00:38:28.889Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
2020-01-25T00:38:28.889Z serialport/bindings/poller Polling for "readable"
2020-01-25T00:38:28.893Z serialport/bindings/poller received "readable"
2020-01-25T00:38:28.894Z serialport/bindings/unixRead Starting read
2020-01-25T00:38:28.894Z serialport/bindings/unixRead Finished read 6 bytes
2020-01-25T00:38:28.894Z serialport/stream binding.read finished { bytesRead: 6 }
2020-01-25T00:38:28.894Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,17,48]
2020-01-25T00:38:28.894Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,17,48]
2020-01-25T00:38:28.894Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [17] - 48
2020-01-25T00:38:28.894Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":17}
2020-01-25T00:38:28.894Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-25T00:38:28.894Z serialport/stream _read reading { start: 4388, toRead: 61148 }
2020-01-25T00:38:28.894Z serialport/binding-abstract read
2020-01-25T00:38:28.894Z serialport/bindings/unixRead Starting read
(node:16) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - extRouteDisc - {"dstAddr":28250,"options":2,"radius":30}' failed with status '17' (expected '0')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
(node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, o
r by rejecting a promise which was not handled with .catch(). (rejection id: 69)
2020-01-25T00:38:28.894Z serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  syscall: 'read'
}
[... cut ...]
2020-01-25T00:38:28.906Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,4,11,90,110,1,1,0,31,0,254,65,88,0,0,8,8,38,10,5,5,33,0,0,39,22
8,28] - 200
2020-01-25T00:38:28.906Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":28250,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":31,"securityuse":0,"timestamp":5784062,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[8,38,10,5,5,33,0,0]}}
```

VooZ2, 2020-01-25:
Hi,

I'm newbie and looks like I have similar issue. Yesterday I have flashed zigbee CC2531  USB adapter from ali by [official manual](https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html). I have added all my lamps (7 IKEA and 3 Xiaomi). But sometimes when I turn off any lamp from switch (not HASS) I got error and then all lamps become Unavailable on HASS.

Error:

```
zigbee2mqtt:error 2020-01-25 17:49:28: Failed to setup reporting for '0x000b57fffe9c2b7d' - Error: SREQ '--> AF - dataRequest - {"dstaddr":44140,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":45,"options":0,"radius":30,"len":11,"data":{"type":"Buffer","data":[16,77,6,0,0,0,16,0,0,44,1]}}' failed with status '2' (expected '0')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
zigbee2mqtt:error 2020-01-25 17:40:02: Failed to setup reporting for '0x14b457fffe27a435' - Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
```

after that I'm trying to restart zigbee2mqtt in docker but always getting error:

```
zigbee2mqtt:error 2020-01-25 23:13:08: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-01-25 23:13:08: Failed to start zigbee
zigbee2mqtt:error 2020-01-25 23:13:08: Exiting...
zigbee2mqtt:error 2020-01-25 23:13:08: Error: SRSP - SYS - version after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
```

Koenkk, 2020-01-26:
@airdrummingfool found out that zstack 1.2 doesn't have this command, executing this command on zstack 1.2 leads to unspecified behaviour. This has been added in zigbee2mqtt 1.9.0-dev so it was definitely due to this update. Fixed it in the latest dev branch

@VooZ2 please try the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

VooZ2, 2020-01-26:
I reflashed yesterday already. Now network much more stable. But getting errors at random bulb:

```
info  2020-01-26 20:33:39: MQTT publish: topic 'zigbee2mqtt/sveataines_sviestuvas_22/availability', payload 'offline'
error 2020-01-26 20:33:39: Failed to ping 'sveataines_sviestuvas_22'
info  2020-01-26 20:34:09: MQTT publish: topic 'zigbee2mqtt/sveataines_sviestuvas_22/availability', payload 'online'
info  2020-01-26 20:34:09: MQTT publish: topic 'zigbee2mqtt/sveataines_sviestuvas_22', payload '{"state":"ON","brightness":255,"color_temp":370,"linkquality":55}'
info  2020-01-26 20:34:09: MQTT publish: topic 'zigbee2mqtt/sveataines_sviestuvas_22', payload '{"state":"ON","brightness":255,"color_temp":370,"linkquality":55}'
info  2020-01-26 20:34:09: MQTT publish: topic 'zigbee2mqtt/sveataines_sviestuvas_22', payload '{"state":"ON","brightness":255,"color_temp":370,"linkquality":55}'
info  2020-01-26 20:34:09: MQTT publish: topic 'zigbee2mqtt/sveataines_sviestuvas_22', payload '{"state":"ON","brightness":255,"color_temp":370,"linkquality":55}'
info  2020-01-26 20:34:09: MQTT publish: topic 'zigbee2mqtt/sveataines_sviestuvas_22', payload '{"state":"ON","brightness":255,"color_temp":370,"linkquality":55}'
info  2020-01-26 20:34:10: MQTT publish: topic 'zigbee2mqtt/sveataines_sviestuvas_22', payload '{"state":"ON","brightness":255,"color_temp":370,"linkquality":55}'

```

```
info  2020-01-26 20:36:49: MQTT publish: topic 'zigbee2mqtt/kolidorius_2/availability', payload 'offline'
error 2020-01-26 20:36:49: Failed to ping 'kolidorius_2'
info  2020-01-26 20:37:19: MQTT publish: topic 'zigbee2mqtt/kolidorius_2/availability', payload 'online'
info  2020-01-26 20:37:20: MQTT publish: topic 'zigbee2mqtt/kolidorius_2', payload '{"state":"OFF","linkquality":23,"brightness":254}'
info  2020-01-26 20:37:20: MQTT publish: topic 'zigbee2mqtt/kolidorius_2', payload '{"state":"OFF","linkquality":23,"brightness":254}'
```

my network map:
[image]

on HASS I have grouped bulbs:

Group A:
3x IKEA TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (LED1545G12)
3x Xiaomi Aqara smart LED bulb (ZNLDP12LM)

Group B:
3x IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1837R5)

VooZ2, 2020-01-27:
Found in confif file that channel was 20. Set it to 11 and now working fine. Case closed from my side.

Koenkk, 2020-01-27:
Thanks, as @TilmanK also didn't experience it anymore, and the case for @airdrummingfool is fixed, let's close it for now.

closed 2020-01-27 by Koenkk