Koenkk/zigbee2mqtt #2750: zigbee2mqtt:error 2020-01-15 20:34:23: Failed to interview '0x000d6f00104f248d', device has not successfully been paired

khaidakin, 2020-01-15:
# Bug Report
## What happened
Trying to pair HEIMAN HS1RC-M
https://www.zigbee2mqtt.io/devices/HS1RC-M.html
## What did you expect to happen

## How to reproduce it (minimal and precise)
try to add HEIMAN HS1RC-M to zigbee2mqtt
## Debug Info
zigbee2mqtt version: 1.8.0-dev (commit #cabb1f3)
CC2538 firmware version: 20190523
https://pastebin.com/6dTn7PKu

khaidakin, 2020-01-16:
I edited the device.ts on the advice of Ilya Kirov and the interview began to pass. The device is connected.
https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/device.ts#L398
Line
for (const endpoint of this.endpoints.filter((e): boolean => e.supportsInputCluster('ssIasZone'))) {
I changed to
for (const endpoint of []) {
Here is the log after correction
https://pastebin.com/iGBUbEk1
But the  DIYRuZ relay switch power meter (https://github.com/Koenkk/zigbee2mqtt/issues/2698) still does not connect.

Koenkk, 2020-01-16:
There seem to be many people having issues of the HS1RC-M not working: #1957 #2418. 

In https://github.com/Koenkk/zigbee2mqtt/issues/1957 I've already spend a lot of time investigating this without luck unfortunately.

As nobody can get this device working, I will remove it from the supported devices list (to avoid confusion).

closed 2020-01-16 by Koenkk