Koenkk/zigbee2mqtt #2751: Paul Neuhaus "Error while parsing to ZpiObject"

oepoemoepoe, 2020-01-15:
# Bug Report
## What happened
When trying to pair [this ](https://www.paul-neuhaus.de/shop/de/led-strahler-smart-home-alexa-tauglich-100-082-47.html) light, I get the following error (see [log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4067273/log.txt) ):
`Error while parsing to ZpiObject 'TypeError: Cannot read property 'updateLastSeen' of undefined`
and later on:
`Interview - failed for device '0x00158d0002574367' with error 'Error: Timeout - 32624 - 1 - 2 - 0 - 1 after 10000ms`

## What did you expect to happen
I believed that this light works as it is similar to 100.110.39 . My other Paul Neuhaus also worked out of the box.

## How to reproduce it (minimal and precise)
Resetting the light and waiting ~1min.

## Debug Info
zigbee2mqtt version: dev-branch commit 5c18734e6818fc85ec8f2a42cdbdd71e501499aa
CC253X firmware version: CC2531_DEFAULT_20190608

oepoemoepoe, 2020-01-16:
I discovered that this problem does actually not occur when using 1.8.0 - so I believe something disrupted it from functioning correctly afterwards.

Koenkk, 2020-01-18:
`Cannot read property 'updateLastSeen' of undefined` should be fixed in the latest dev branch, please try again.

oepoemoepoe, 2020-02-07:
Late response. It works now, thanks.

closed 2020-02-07 by oepoemoepoe