Koenkk/zigbee2mqtt #2756: Adding Nue/3A RGB Downlight

wrcvr4, 2020-01-16:
## What happened
I have added a NUE/3A Smart Home RGB downlight HGZB-DLC4-N12B
It showed up as not supported, the model is similar to  #1998

I have managed to get it showing as supported by adding the following to share\zigbee2mqtt\devices.js
```
{
        zigbeeModel: ['NUET56-DL27LX1.2'],
        model: 'DL27LX1.2',
        vendor: '3A Smart Home DE',
        description: 'RGB LED downlight',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
    },
```
```
{"ieeeAddr":"xxx","type":"Router","networkAddress":34126,"model":"DL27LX1.2","friendly_name":"xxx","manufacturerID":0,"manufacturerName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelID":"NUET56-DL27LX1.2","hardwareVersion":1,"softwareBuildID":"1.0.2","dateCode":"20160422","lastSeen":1579159412881}]}
```
```
MQTT publish: topic 'zigbee2mqtt/xxxx', payload '{"state":"ON","linkquality":26,"brightness":224,"color_temp":222,"color":{"x":0.3125,"y":0.3333}}'
```
Hassio does not automatically discover it, if I work out how to add to 'homeassistant.js' then it should

## Debug Info
Hassos: 3.7
Hassio: 0.103.5
zigbee2mqtt version: 1.8.0
CC253X firmware version: 20190608

sjorge, 2020-01-16:
In think you just add a lime like this:
https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js#L526

```
'NUET56-DL27LX1.2': [cfg.light_brightness_colortemp_colorxy],
```

Or maybe just DL27LX1.2, not 100% sure

wrcvr4, 2020-01-16:
Thanks @sjorge 
I moved hassio from hassos to Ubuntu so I could easily get to the docker container 
Have added the line (I think it should be 'DL27LX1.2,' based on the issue I linked, hassio has not discovered/added it yet

Trying a system reboot

wrcvr4, 2020-01-16:
Still not being discovered by hassio post reboot, I added the following line
```'DL27LX1.2': [cfg.light_brightness_colortemp_colorxy],```

Also the following error is in zigbee2mqtt logs, which looks to be related to zigbee2mqttassistant network scans?
```Failed to execute LQI for '0x00124b001d2050ba'```

Here are the logs if I re-pair it
```
igbee2mqtt:info  2020-01-16 23:03:18: Device '0x00124b001d2050ba' joined
zigbee2mqtt:info  2020-01-16 23:03:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001d2050ba"}}'
zigbee2mqtt:info  2020-01-16 23:03:18: Starting interview of '0x00124b001d2050ba'
zigbee2mqtt:info  2020-01-16 23:03:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001d2050ba"}}'
zigbee2mqtt:info  2020-01-16 23:03:29: Successfully interviewed '0x00124b001d2050ba', device has successfully been paired
zigbee2mqtt:info  2020-01-16 23:03:29: Device '0x00124b001d2050ba' is supported, identified as: 3A Smart Home DE RGB LED downlight (DL27LX1.2)
zigbee2mqtt:info  2020-01-16 23:03:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001d2050ba","model":"DL27LX1.2","vendor":"3A Smart Home DE","description":"RGB LED downlight","supported":true}}'
```

Koenkk, 2020-01-16:
@wrcvr4 please make a PR for the `devices.js` changes in https://github.com/Koenkk/zigbee2mqtt/issues/2756#issue-550634749.

I will handle the Home Assistant discovery then.

wrcvr4, 2020-01-16:
Thanks @Koenkk 

It would take me awhile to figure out git clone etc
I'm happy if you could add it ?
Then I can test

wrcvr4, 2020-01-17:
I must be blind or crazy, I cannot find the devices.js in the repository to change ?

wrcvr4, 2020-01-17:
Update:
I have it fully working and discovered in hassio now, I just cannot work out the PR process :(

devices,js
```
{
        zigbeeModel: ['NUET56-DL27LX1.2'],
        model: 'DL27LX1.2',
        vendor: 'NUE / 3A',
        description: 'RGB LED downlight',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
    },
```
homeassistant.js
```'DL27LX1.2': [cfg.light_brightness_colortemp_colorxy],```

Koenkk, 2020-01-17:
There are some guides available on how to create a pr, e.g. https://youtu.be/rgbCcBNZcdQ

If you cannot no problem, then I can add them, let me know.

Probably the device doesnt support a lqi scan (issue of the device itself).

wrcvr4, 2020-01-20:
I have created two PR's
One in this project and one in the  zigbee-herdsman-converters
#2784 

Please let me know if I have done it correctly

wrcvr4, 2020-01-20:
Thanks for the merge
If I install the latest -edge version, will that now have the changes so I can confirm ?

Koenkk, 2020-01-21:
@wrcvr4 yes

wrcvr4, 2020-01-21:
Looks like its working

closed 2020-01-21 by wrcvr4