Koenkk/zigbee2mqtt #2757: Gledopto GL-S-007ZS GU10 Bulb Support Continued...

milobt, 2020-01-16:
# Bug Report
## What happened
I am still having issues with my new Gledopto GL-S-007ZS bulbs and think it may be something to do with the xy values when converted to mqtt.

I have a number of bulbs that have been grouped and tend to work ~OK, but still do not always turn ON when stating a brightness, etc. If just turning ON or OFF, they generally work fine.

I have two bulbs (either side of our bed) that I've left as individuals and control separately, these hardly ever turn on when sent with a brightness, but will work fine when just ON or OFF.

I have noticed that the x and y values sent to these bulbs are generally 15 decimal places long, rather than the general 3 decimal places; some times these values are 4 places, but sometimes work, e.g.:

zigbee2mqtt:info 2020-01-16 08:23:30: MQTT publish: topic 'zigbee2mqtt/Miles Spot', payload '{"state":"OFF","brightness":190,"color_temp":386,"color":{"x":0.469092603126886,"y":0.412452218918178}}' zigbee2mqtt:info 2020-01-16 08:23:30: MQTT publish: topic 'zigbee2mqtt/Nichola Spot', payload '{"state":"OFF","color_temp":401,"color":{"x":0.477890571505866,"y":0.413792775585208},"brightness":190}' zigbee2mqtt:info 2020-01-16 08:23:30: MQTT publish: topic 'zigbee2mqtt/kitchen_spots', payload '{"state":"OFF","brightness":130,"color_temp":374,"color":{"x":0.4534,"y":0.432}}'

The top two devices (Miles Spot and Nichola Spot) are the individual bulbs, the third is a group (kitchen_spots).

Is there a converter missing for these bulbs?
I noticed that there are no converters for the GL-S bulbs or RGB strip controllers GL-C-008S / 9 (there is one for GL-C-008), so may just needs the S variant adding.
Although, I am unsure whether the converter would limit the xy value and whether the current 15 decimal places is normal or not?

I also don't get a Link Quality for my GL-S-007ZS bulbs, so can't tell if their problem is related to a weak signal.

I notice that in the devices.js files for the GL-S-007Z that you have the additional line:
meta: {options: {disableDefaultResponse: true}},

But not for the GL-S-007ZS, is this intentional?

Finally, I tend to get a lot of Mac and timeout errors, are these caused by weak signals/interference or other reason?
zigbee2mqtt:error 2020-01-16 08:51:30: Publish 'set' 'state' to 'Headboard' failed: 'Error: Data request failed with error: 'MAC no ack' (233)' zigbee2mqtt:error 2020-01-16 08:51:15: Publish 'set' 'state' to 'Headboard' failed: 'Error: Timeout - 17817 - 15 - 131 - 6 - 11 after 15000ms'

## Debug Info
zigbee2mqtt version: 1.8.0
CC26X2R1 firmware version: 20191106

Koenkk, 2020-01-16:
> I have noticed that the x and y values sent to these bulbs are generally 15 decimal places long

Can you post the log before the `MQTT publish`?

> I also don't get a Link Quality for my GL-S-007ZS bulbs

Expected, as the bulb doesn't send messages (you only send messages to it)

> I notice that in the devices.js files for the GL-S-007Z

That line prevents the timeout error you mentioned, adding it should solve it.

milobt, 2020-01-22:
Sorry, for the delay, have not had a chance to look at this again properly.

The problem seems to be whenever I try to turn these bulbs on with a brightness.
If I sent the light.turn_on, without a brightness level, they turn on every time.
If I turn on the switch in HA it turns on every time.
But as soon as a send a brightness level it doesn't turn on.

I have found that by sending the turn_on command (only) followed with another one with a brightness, that the bulbs will turn on and adjust to that brightness.
This does cause a problem where I have chosen to use one automation and toggle the light (with brightness), rather than two automations (turn_on with this brightness) and simply turn_off.

E.g.:
`alias: Miles Bedside Button Switch Single Click
  hide_entity: false
  initial_state: 'on'
  trigger:
    platform: state
    entity_id: sensor.miles_bedside_button_switch_click
    to: single
  action:
    service: light.toggle
    data_template:
      entity_id: light.miles_spot
      brightness: "{{ states('sensor.master_bedroom_brightness') }}"`

I have now set up a sensor to state the brightness level, so that it's easier to vary and can be influenced by a light sensor level.
Before I just used IF statements based on the time on day:
`{% if now().hour >= 7 and now().hour < 9 %}
          190
        {% elif now().hour >= 9 and now().hour < 12 %}
          190
        {% elif now().hour >= 12 and now().hour < 16 %}
          230
        {% elif now().hour >= 16 and now().hour < 22 %}
          230
        {% elif now().hour >= 22 and now().hour < 24 %}
          75
        {% elif now().hour >= 0 and now().hour < 7 %}
          50
        {% endif %}`

I thought that this statement could be the issue, when it was using percentage levels (0-100), so changed to brightness scale (0-255), but made no difference.

I just seems to be an issue when turning on with a brightness, different or the same as the one currently set.

Log of turning Kitchen_Spots (group of 6 bulbs) off, then on:
`zigbee2mqtt:info  2020-01-22 10:16:09: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 1', payload '{"state":"OFF","color":{"x":0.4534,"y":0.432},"color_temp":374,"brightness":190}'
zigbee2mqtt:info  2020-01-22 10:16:09: MQTT publish: topic 'zigbee2mqtt/kitchen_spots', payload '{"state":"OFF","brightness":190,"color_temp":374,"color":{"x":0.4534,"y":0.432}}'
zigbee2mqtt:info  2020-01-22 10:16:09: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 2', payload '{"state":"OFF","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-22 10:16:09: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 3', payload '{"state":"OFF","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-22 10:16:09: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 4', payload '{"state":"OFF","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-22 10:16:09: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 5', payload '{"state":"OFF","brightness":190,"color_temp":374,"color":{"x":0.4534,"y":0.432}}'
zigbee2mqtt:info  2020-01-22 10:16:09: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 6', payload '{"state":"OFF","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-22 10:16:09: MQTT publish: topic 'zigbee2mqtt/Glasses Cabinet', payload '{"state":"OFF","linkquality":111}'
zigbee2mqtt:info  2020-01-22 10:16:09: MQTT publish: topic 'zigbee2mqtt/Glasses Cabinet', payload '{"state":"OFF","linkquality":111}'
zigbee2mqtt:info  2020-01-22 10:16:09: MQTT publish: topic 'zigbee2mqtt/Glasses Cabinet', payload '{"state":"OFF","linkquality":111}'
zigbee2mqtt:info  2020-01-22 10:16:09: MQTT publish: topic 'zigbee2mqtt/Glasses Cabinet', payload '{"state":"OFF","linkquality":111}'
zigbee2mqtt:info  2020-01-22 10:16:13: MQTT publish: topic 'zigbee2mqtt/Kitchen Cabinet Light', payload '{"state":"ON","linkquality":123}'
zigbee2mqtt:info  2020-01-22 10:16:13: MQTT publish: topic 'zigbee2mqtt/Kitchen Cabinet Light', payload '{"state":"ON","linkquality":123}'
zigbee2mqtt:info  2020-01-22 10:16:13: MQTT publish: topic 'zigbee2mqtt/Kitchen Cabinet Light', payload '{"state":"ON","linkquality":123}'
zigbee2mqtt:info  2020-01-22 10:16:13: MQTT publish: topic 'zigbee2mqtt/Kitchen Cabinet Light', payload '{"state":"ON","linkquality":123}'
zigbee2mqtt:info  2020-01-22 10:16:13: MQTT publish: topic 'zigbee2mqtt/Kitchen Cabinet Light', payload '{"state":"ON","linkquality":123}'
zigbee2mqtt:info  2020-01-22 10:16:13: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 1', payload '{"state":"ON","color":{"x":0.4534,"y":0.432},"color_temp":374,"brightness":190}'
zigbee2mqtt:info  2020-01-22 10:16:13: MQTT publish: topic 'zigbee2mqtt/kitchen_spots', payload '{"state":"ON","brightness":190,"color_temp":374,"color":{"x":0.4534,"y":0.432}}'
zigbee2mqtt:info  2020-01-22 10:16:13: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 2', payload '{"state":"ON","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-22 10:16:13: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 3', payload '{"state":"ON","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-22 10:16:13: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 4', payload '{"state":"ON","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-22 10:16:13: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 5', payload '{"state":"ON","brightness":190,"color_temp":374,"color":{"x":0.4534,"y":0.432}}'
zigbee2mqtt:info  2020-01-22 10:16:13: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 6', payload '{"state":"ON","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'`

Koenkk, 2020-01-22:
Can you try if the problem has been fixed in the lates dev branch?

milobt, 2020-01-23:
I have just updated from 1.8.0 to 1.9.0, as realised that I was behind on the update - didn't have the repository installed, after a snapshot restore.

Have remove the double light.turn_on command and swapped to the dev branch and same result, the light appears on in HA, but doesn't turn on the actual light.

`zigbee2mqtt:info  2020-01-23 12:29:57: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 1', payload '{"state":"OFF","color":{"x":0.4534,"y":0.432},"color_temp":374,"brightness":190}'
zigbee2mqtt:info  2020-01-23 12:29:57: MQTT publish: topic 'zigbee2mqtt/kitchen_spots', payload '{"state":"OFF","brightness":190,"color_temp":374,"color":{"x":0.4534,"y":0.432}}'
zigbee2mqtt:info  2020-01-23 12:29:57: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 2', payload '{"state":"OFF","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-23 12:29:57: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 3', payload '{"state":"OFF","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-23 12:29:57: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 4', payload '{"state":"OFF","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-23 12:29:57: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 5', payload '{"state":"OFF","brightness":190,"color_temp":374,"color":{"x":0.4534,"y":0.432}}'
zigbee2mqtt:info  2020-01-23 12:29:57: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 6', payload '{"state":"OFF","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-23 12:30:00: Starting network scan (includeRoutes 'false')
zigbee2mqtt:info  2020-01-23 12:30:00: MQTT publish: topic 'zigbee2mqtt/Miles Spot', payload '{"state":"OFF","brightness":190,"color_temp":393,"color":{"x":0.473443147918802,"y":0.413167472190078}}'
zigbee2mqtt:info  2020-01-23 12:30:00: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 1', payload '{"state":"OFF","color":{"x":0.4534,"y":0.432},"color_temp":374,"brightness":190}'
zigbee2mqtt:info  2020-01-23 12:30:00: MQTT publish: topic 'zigbee2mqtt/kitchen_spots', payload '{"state":"OFF","brightness":190,"color_temp":374,"color":{"x":0.4534,"y":0.432}}'
zigbee2mqtt:info  2020-01-23 12:30:00: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 2', payload '{"state":"OFF","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-23 12:30:00: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 3', payload '{"state":"OFF","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-23 12:30:00: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 4', payload '{"state":"OFF","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'
zigbee2mqtt:info  2020-01-23 12:30:00: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 5', payload '{"state":"OFF","brightness":190,"color_temp":374,"color":{"x":0.4534,"y":0.432}}'
zigbee2mqtt:info  2020-01-23 12:30:00: MQTT publish: topic 'zigbee2mqtt/Kitchen Spot Light 6', payload '{"state":"OFF","color_temp":374,"color":{"x":0.4534,"y":0.432},"brightness":190}'`

Koenkk, 2020-01-23:
You need to update to the latest dev branch, not 1.9. https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

milobt, 2020-01-23:
> Have remove the double light.turn_on command and swapped to the dev branch

Koenkk, 2020-01-23:
Ok, made another attempt to fix it, please try again with latest dev.

If it doesn't work please provide the sniffing logs when controlling it with brightness and state at the same time and when controlling it with a separate state and brightness command. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

milobt, 2020-01-23:
I though that had fixed it; it did improve control for groups, but individual lights miles_spot and nichola_spot, now do not turn on and instead, there is an error in the log:

`zigbee2mqtt:error 2020-01-23 21:01:46: Publish 'set' 'brightness' to 'Miles Spot' failed: 'TypeError: value.toLowerCase is not a function'
zigbee2mqtt:info  2020-01-23 21:01:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'brightness' to 'Miles Spot' failed: 'TypeError: value.toLowerCase is not a function'","meta":{"friendly_name":"Miles Spot"}}'
zigbee2mqtt:error 2020-01-23 21:01:46: Publish 'set' 'brightness' to 'Nichola Spot' failed: 'TypeError: value.toLowerCase is not a function'
zigbee2mqtt:info  2020-01-23 21:01:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'brightness' to 'Nichola Spot' failed: 'TypeError: value.toLowerCase is not a function'","meta":{"friendly_name":"Nichola Spot"}}' `

milobt, 2020-01-24:
I will create a sniffer dongle ASAP - likely Sunday evening now - and then paste the logs from wireshark.
I use a CC26X2R1 now, but still have a couple CC2531 in a drawer.

There's clearly a error in your last update that is now creating the error in the log: 
`TypeError: value.toLowerCase is not a function`

Koenkk, 2020-01-24:
Can you provide the log with debug enabled? 

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

milobt, 2020-01-24:
So with automation, for miles_spot and nichola_spot with light.turn_on + light.turn_on (with brightness):

(Z2M 1.9.0):

`zigbee2mqtt:debug 2020-01-24 15:43:47: Received MQTT message on 'zigbee2mqtt/Miles Spot/set' with data '{"state": "ON"}'
zigbee2mqtt:debug 2020-01-24 15:43:47: Publishing 'set' 'state' to 'Miles Spot'
zigbee2mqtt:debug 2020-01-24 15:43:47: Received MQTT message on 'zigbee2mqtt/Miles Spot/set' with data '{"state": "ON", "brightness": 230}'
zigbee2mqtt:debug 2020-01-24 15:43:47: Publishing 'set' 'brightness' to 'Miles Spot'
zigbee2mqtt:debug 2020-01-24 15:43:47: Received MQTT message on 'zigbee2mqtt/Nichola Spot/set' with data '{"state": "ON"}'
zigbee2mqtt:debug 2020-01-24 15:43:47: Publishing 'set' 'state' to 'Nichola Spot'
zigbee2mqtt:debug 2020-01-24 15:43:47: Received MQTT message on 'zigbee2mqtt/Nichola Spot/set' with data '{"state": "ON", "brightness": 230}'
zigbee2mqtt:debug 2020-01-24 15:43:47: Publishing 'set' 'brightness' to 'Nichola Spot'
zigbee2mqtt:debug 2020-01-24 15:43:47: Received MQTT message on 'zigbee2mqtt/Headboard/set' with data '{"state": "ON", "brightness": 230}'
zigbee2mqtt:debug 2020-01-24 15:43:47: Publishing 'set' 'brightness' to 'Headboard'
zigbee2mqtt:info  2020-01-24 15:43:47: MQTT publish: topic 'zigbee2mqtt/Nichola Spot', payload '{"state":"ON","color_temp":384,"color":{"x":0.468234043017033,"y":0.41229905027829},"brightness":190}'
zigbee2mqtt:info  2020-01-24 15:43:47: MQTT publish: topic 'zigbee2mqtt/Nichola Spot', payload '{"state":"ON","color_temp":384,"color":{"x":0.468234043017033,"y":0.41229905027829},"brightness":230}'`

(Z2M Edge):

`zigbee2mqtt:debug 2020-01-24 15:45:50: Received MQTT message on 'zigbee2mqtt/Miles Spot/set' with data '{"state": "ON"}'
zigbee2mqtt:debug 2020-01-24 15:45:50: Publishing 'set' 'state' to 'Miles Spot'
zigbee2mqtt:debug 2020-01-24 15:45:50: Received MQTT message on 'zigbee2mqtt/Miles Spot/set' with data '{"state": "ON", "brightness": 230}'
zigbee2mqtt:debug 2020-01-24 15:45:50: Publishing 'set' 'brightness' to 'Miles Spot'
zigbee2mqtt:debug 2020-01-24 15:45:50: Received MQTT message on 'zigbee2mqtt/Nichola Spot/set' with data '{"state": "ON"}'
zigbee2mqtt:debug 2020-01-24 15:45:50: Publishing 'set' 'state' to 'Nichola Spot'
zigbee2mqtt:error 2020-01-24 15:45:50: Publish 'set' 'brightness' to 'Miles Spot' failed: 'TypeError: value.toLowerCase is not a function'
zigbee2mqtt:debug 2020-01-24 15:45:50: TypeError: value.toLowerCase is not a function
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:101:52)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1126:45)
    at EntityPublish.onMQTTMessage (/app/lib/extension/entityPublish.js:173:52)
    at Controller.callExtensionMethod (/app/lib/controller.js:327:44)
    at Controller.onMQTTMessage (/app/lib/controller.js:239:14)
    at MQTT.emit (events.js:210:5)
    at MQTT.onMessage (/app/lib/mqtt.js:91:14)
    at MqttClient.emit (events.js:210:5)
    at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1162:12)
    at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:351:12)
zigbee2mqtt:info  2020-01-24 15:45:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'brightness' to 'Miles Spot' failed: 'TypeError: value.toLowerCase is not a function'","meta":{"friendly_name":"Miles Spot"}}'
zigbee2mqtt:debug 2020-01-24 15:45:50: Received MQTT message on 'zigbee2mqtt/Nichola Spot/set' with data '{"state": "ON", "brightness": 230}'
zigbee2mqtt:debug 2020-01-24 15:45:50: Publishing 'set' 'brightness' to 'Nichola Spot'
zigbee2mqtt:debug 2020-01-24 15:45:50: Received MQTT message on 'zigbee2mqtt/Headboard/set' with data '{"state": "ON", "brightness": 230}'
zigbee2mqtt:debug 2020-01-24 15:45:50: Publishing 'set' 'brightness' to 'Headboard'
zigbee2mqtt:debug 2020-01-24 15:45:50: Received MQTT message on 'zigbee2mqtt/Master Bedroom Ceiling Light/set' with data 'ON'
zigbee2mqtt:debug 2020-01-24 15:45:50: Publishing 'set' 'state' to 'Master Bedroom Ceiling Light'
zigbee2mqtt:error 2020-01-24 15:45:50: Publish 'set' 'brightness' to 'Nichola Spot' failed: 'TypeError: value.toLowerCase is not a function'
zigbee2mqtt:debug 2020-01-24 15:45:50: TypeError: value.toLowerCase is not a function
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:101:52)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1126:45)
    at EntityPublish.onMQTTMessage (/app/lib/extension/entityPublish.js:173:52)
    at Controller.callExtensionMethod (/app/lib/controller.js:327:44)
    at Controller.onMQTTMessage (/app/lib/controller.js:239:14)
    at MQTT.emit (events.js:210:5)
    at MQTT.onMessage (/app/lib/mqtt.js:91:14)
    at MqttClient.emit (events.js:210:5)
    at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1162:12)
    at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:351:12)
zigbee2mqtt:info  2020-01-24 15:45:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'brightness' to 'Nichola Spot' failed: 'TypeError: value.toLowerCase is not a function'","meta":{"friendly_name":"Nichola Spot"}}'
zigbee2mqtt:error 2020-01-24 15:45:50: Publish 'set' 'state' to 'Nichola Spot' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'
[... cut ...]
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:584:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:5:58)
zigbee2mqtt:info  2020-01-24 15:45:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Nichola Spot' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'","meta":{"friendly_name":"Nichola Spot"}}'`

Koenkk, 2020-01-25:
Should be fixed in the latest dev branch, will be available in a few hours.

milobt, 2020-02-10:
Thanks for the latest fix @Koenkk, however, the fix for the individual bulbs seems to have caused issues for the grouped bulbs - these now only turn on - with brightness setting - in the evening, when the brightness value should be set too 100%.

`entrance_brightness:
        friendly_name: 'Entrance Brightness Level'
        entity_id: sensor.time
        value_template: >
          {% if now().hour >= 6 and now().hour < 9 %}
            50
          {% elif now().hour >= 9 and now().hour < 12 %}
            75
          {% elif now().hour >= 12 and now().hour < 16 %}
            75
          {% elif now().hour >= 16 and now().hour < 22 %}
            100
          {% elif now().hour >= 22 and now().hour < 24 %}
            75
          {% elif now().hour >= 0 and now().hour < 6 %}
            50
          {% endif %}
        icon_template: 'mdi:brightness-percent'`

They can be turned toggled on/off without brightness, by physical switches (e.g xiaomi buttons/wall switches) and can be turned on/off with/without brightness via HA GUI buttons, but automations don't work when setting a brightness.

I have compared the logs for HA switch toggle versus the automation and both look identical:

**HA GUI Switches:**
Landing Spots (grouped bulbs):

Just turning on switch:
`zigbee2mqtt:debug 2020-02-10 09:28:11: Received MQTT message on 'zigbee2mqtt/landing_spots/set' with data '{"state": "ON"}'
zigbee2mqtt:debug 2020-02-10 09:28:11: Publishing 'set' 'state' to 'landing_spots'
zigbee2mqtt:info  2020-02-10 09:28:11: MQTT publish: topic 'zigbee2mqtt/Landing Spot Light 2', payload '{"state":"ON","color_temp":391,"color":{"x":0.3227,"y":0.329},"brightness":191}'
zigbee2mqtt:info  2020-02-10 09:28:11: MQTT publish: topic 'zigbee2mqtt/landing_spots', payload '{"state":"ON","brightness":191,"color_temp":391,"color":{"x":0.3227,"y":0.329}}'
zigbee2mqtt:info  2020-02-10 09:28:11: MQTT publish: topic 'zigbee2mqtt/Landing Spot Light 4', payload '{"state":"ON","color_temp":391,"color":{"x":0.3227,"y":0.329},"brightness":191}'
zigbee2mqtt:info  2020-02-10 09:28:11: MQTT publish: topic 'zigbee2mqtt/Landing Spot Light 3', payload '{"state":"ON","color_temp":391,"color":{"x":0.3227,"y":0.329},"brightness":191}'
zigbee2mqtt:info  2020-02-10 09:28:11: MQTT publish: topic 'zigbee2mqtt/Landing Spot Light 1', payload '{"state":"ON","color_temp":391,"color":{"x":0.3227,"y":0.329},"brightness":191}'`

Setting brightness when off, causing the light to turn on with a brightness:
`zigbee2mqtt:debug 2020-02-10 09:29:24: Received MQTT message on 'zigbee2mqtt/landing_spots/set' with data '{"state": "ON", "brightness": 147}'
zigbee2mqtt:debug 2020-02-10 09:29:24: Publishing 'set' 'brightness' to 'landing_spots'
zigbee2mqtt:info  2020-02-10 09:29:24: MQTT publish: topic 'zigbee2mqtt/Landing Spot Light 2', payload '{"state":"ON","color_temp":391,"color":{"x":0.3227,"y":0.329},"brightness":147}'
zigbee2mqtt:info  2020-02-10 09:29:24: MQTT publish: topic 'zigbee2mqtt/landing_spots', payload '{"state":"ON","brightness":147,"color_temp":391,"color":{"x":0.3227,"y":0.329}}'
zigbee2mqtt:info  2020-02-10 09:29:24: MQTT publish: topic 'zigbee2mqtt/Landing Spot Light 4', payload '{"state":"ON","color_temp":391,"color":{"x":0.3227,"y":0.329},"brightness":147}'
zigbee2mqtt:info  2020-02-10 09:29:24: MQTT publish: topic 'zigbee2mqtt/Landing Spot Light 3', payload '{"state":"ON","color_temp":391,"color":{"x":0.3227,"y":0.329},"brightness":147}'
zigbee2mqtt:info  2020-02-10 09:29:24: MQTT publish: topic 'zigbee2mqtt/Landing Spot Light 1', payload '{"state":"ON","color_temp":391,"color":{"x":0.3227,"y":0.329},"brightness":147}'`

Miles Spot (individual bulb):
Just turning on switch:
`zigbee2mqtt:debug 2020-02-10 09:30:43: Received MQTT message on 'zigbee2mqtt/Miles Spot/set' with data '{"state": "ON"}'
zigbee2mqtt:debug 2020-02-10 09:30:43: Publishing 'set' 'state' to 'Miles Spot'
zigbee2mqtt:info  2020-02-10 09:30:44: MQTT publish: topic 'zigbee2mqtt/Miles Spot', payload '{"state":"ON","brightness":191,"color_temp":348,"color":{"x":0.446471069076735,"y":0.407117718004646}}'

Setting brightness when off, causing the light to turn on with a brightness:
zigbee2mqtt:debug 2020-02-10 09:31:50: Received MQTT message on 'zigbee2mqtt/Miles Spot/set' with data '{"state": "ON", "brightness": 171}'
zigbee2mqtt:debug 2020-02-10 09:31:50: Publishing 'set' 'brightness' to 'Miles Spot'
zigbee2mqtt:info  2020-02-10 09:31:51: MQTT publish: topic 'zigbee2mqtt/Miles Spot', payload '{"state":"ON","brightness":171,"color_temp":348,"color":{"x":0.446471069076735,"y":0.407117718004646}}'`

**Automation:**
Landing Spots (grouped):
`zigbee2mqtt:debug 2020-02-10 09:26:19: Received MQTT message on 'zigbee2mqtt/landing_spots/set' with data '{"state": "ON", "brightness": 191}'
zigbee2mqtt:debug 2020-02-10 09:26:19: Publishing 'set' 'brightness' to 'landing_spots'
zigbee2mqtt:info  2020-02-10 09:26:19: MQTT publish: topic 'zigbee2mqtt/Landing Spot Light 2', payload '{"state":"ON","color_temp":391,"color":{"x":0.3227,"y":0.329},"brightness":191}'
zigbee2mqtt:info  2020-02-10 09:26:19: MQTT publish: topic 'zigbee2mqtt/landing_spots', payload '{"state":"ON","brightness":191,"color_temp":391,"color":{"x":0.3227,"y":0.329}}'
zigbee2mqtt:info  2020-02-10 09:26:19: MQTT publish: topic 'zigbee2mqtt/Landing Spot Light 4', payload '{"state":"ON","color_temp":391,"color":{"x":0.3227,"y":0.329},"brightness":191}'
[... cut ...]
I have also noticed with the bulbs that when you turn them off, they set themselves to full brightness and then fade off, rather than just fading off from current brightness.
This is rather annoying and wonder what the cause is?

There also seems to be the same issue with the GL-C-008 (RGB+CCT) LED strip controller, which is not turning on with a brightness via automation, but will via HA GUI button.

Could this issue be a HA problem or is it just the translation of data from HA to Z2M?

Many thanks.

Koenkk, 2020-02-12:
Unfortunately the same fix ("hack") cannot be used for groups (as we don't want this specific logic to be used for all lights). The problem is that these bulbs behave differently compared to normal bulbs.

closed 2020-04-19 by stale[bot]