Koenkk/zigbee2mqtt #2759: Failed to interview Hue Color E27

lukeIam, 2020-01-16:
## What happened
Can't pair my new Hue White & Color Ambiance E27 (edition with bluetooth) - it don't know it's model number.
I get the message:   
`Failed to interview '0x00178801061f8ad0', device has not successfully been paired`.

Full log:
https://gist.github.com/lukeIam/a9633355814d118c262291100c27f2f2

## What did you expect to happen
Successful pairing

## How to reproduce it (minimal and precise)
- "permit_join":true
- reset the bulb with a hue dimmer.
- wait for the error

## Debug Info
zigbee2mqtt version: 1.8.0 and 1.8.0-dev
CC253X firmware version: 2.6.3 (20190109)

BasVerkooijen, 2020-01-16:
Hi,

I found your issue report and I would like to add that I have observed the same issue with the Hue White E27 with Bluetooth (9290018216). The debug log which you attached describes the same timeout error.

However, the Hue bulb works fine after the 'failed to interview' message. Though, I haven't used Hue Power-on behavior yet.

### My setup:
zigbee2mqtt version: 1.8.0
CC2531 firmware version: 2.6.3 (20190608)

### What I did
I attempted to investigate what is happening and I suspect it is trying to discover a cluster/feature which isn't supported. However, I have no clue how to continue debugging this.

I have attempted multiple pairings with 2 different bulbs and reset them a few times through TouchLink.

If someone knows how to debug this, needs more logs or knows how to continue, let me know. I would like to solve this.

Koenkk, 2020-01-16:
Can you try replacing `node_modules/zigbee-herdsman/dist/controller/model/device.js` with https://gist.github.com/Koenkk/a45bdd9da8f8df31b10490663dae78ab and pair again? (please provide the same log if it doesn't work)

lukeIam, 2020-01-17:
thank you for your fast fix! The bulb is now successfully paired. 😃 

But one last think is not working:
The attribute `brightness` is not published although it's in the json.
Do I need to adjust some kind of device config to enable attribute parsing for `brightness`?

BasVerkooijen, 2020-01-17:
Just tested your change and pairing now succeeds.

### Logs
**Touchlink reset**

> Jan 17 19:22:34 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:22:34: Starting touchlink factory reset...
Jan 17 19:22:36 raspberrypi npm[9100]: zigbee2mqtt:warn  2020-01-17 19:22:36: Device '0x0017880106d277f4' left the network
Jan 17 19:22:36 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:22:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x0017880106d277f4"}}'
Jan 17 19:22:36 raspberrypi npm[9100]: zigbee2mqtt:warn  2020-01-17 19:22:36: Device '0x0017880106d277f4' left the network
Jan 17 19:22:36 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:22:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x0017880106d277f4"}}'
Jan 17 19:22:37 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:22:37: Succesfully factory reset device through Touchlink
Jan 17 19:22:37 raspberrypi npm[9100]: zigbee2mqtt:warn  2020-01-17 19:22:37: Device '0x0017880106d277f4' left the network
Jan 17 19:22:37 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:22:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x0017880106d277f4"}}'

**Pairing**
> Jan 17 19:22:40 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:22:40: Device '0x0017880106d277f4' joined
Jan 17 19:22:40 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:22:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0017880106d277f4"}}'
Jan 17 19:22:40 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:22:40: Starting interview of '0x0017880106d277f4'
Jan 17 19:22:40 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:22:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0017880106d277f4"}}'
Jan 17 19:22:41 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:22:41: Successfully interviewed '0x0017880106d277f4', device has successfully been paired
Jan 17 19:22:41 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:22:41: Device '0x0017880106d277f4' is supported, identified as: Philips Hue white A60 bulb E27 bluetooth (8718699673147)
Jan 17 19:22:41 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:22:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x0017880106d277f4","model":"8718699673147","vendor":"Philips","description":"Hue white A60 bulb E27 bluetooth","supported":true}}'

**Control**
> Jan 17 19:23:31 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:23:31: MQTT publish: topic 'zigbee2mqtt/0x0017880106d277f4', payload '{"state":"OFF","brightness":255}'
Jan 17 19:23:32 raspberrypi npm[9100]: zigbee2mqtt:info  2020-01-17 19:23:32: MQTT publish: topic 'zigbee2mqtt/0x0017880106d277f4', payload '{"state":"ON","brightness":255}'

Thank you! Hue Bulb is working correctly :+1:

Koenkk, 2020-01-17:
Changes are available in latest dev, closing this

@lukeIam can you create a new issue and share your debug log in it?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

closed 2020-01-17 by Koenkk