Koenkk/zigbee2mqtt #2761: Failed to interview 'xxxx' device has not successfully been paired

buteo72, 2020-01-16:
# Bug Report
## What happened
Xiaomi temperature sensor [WSDCGQ11LM] is joined to the network 
Zigbee2MQTT is starting to interview
after 1:20min the message is "Failed to interview 'xxx', device has not successfully been paired"

## What did you expect to happen
That the device is correctly reconiced

## How to reproduce it (minimal and precise)
Reset the sensor by pressing the button for >4s
LED starts flashing, getting into paring mode, 
But directly after the 3 flashes, it flash once more very quick.
Flashes after a while once more.

## Extra info
This sensor was correctly paired. Then it stopt certainly after a lot of days.
A second sensor stoped also in a few hours after the first sensor stoped. (they were paired at almost the same time).

I'm able to pair both sensors to 'KlikAanKlikUit' and that works correctly. That is telling me that the sensor is working properly.

## Debug Info
zigbee2mqtt version: 1.8.0-dev
CC253X firmware version: 20190608

Koenkk, 2020-01-17:
Can you provide the herdsman debug logging while pairing this device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

buteo72, 2020-01-18:
Can I put zigbee_herdsman_debug: true in the configuration.yaml file?
I'm on Windows.

Also my third temperature sensor has stopped working today.

Koenkk, 2020-01-18:
For windows run zigbee2mqtt with: `DEBUG=zigbee-herdsman* npm start`.

buteo72, 2020-01-18:
Sorry, normally I start with PowerShell
Go to the directoy C:\Zigebee2MQTT then `npm start`
With the command `DEBUG=zigbee-herdsman* npm start` I get the following error message:

```
PS C:\Users\domoticz> cd\
PS C:\> cd .\zigbee2mqtt\
PS C:\zigbee2mqtt> DEBUG=zigbee-herdsman* npm start
DEBUG=zigbee-herdsman* : The term 'DEBUG=zigbee-herdsman*' is not recognized as the name of a cmdlet, function, script
file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ DEBUG=zigbee-herdsman* npm start
+ ~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (DEBUG=zigbee-herdsman*:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

```

odx, 2020-01-19:
I have a similar issue. A IKEA Tradfri Bulb that worked like charm with my CC2530 coordinator refuses to pair with my CC1352P-2.

```
info  2020-01-19 21:52:25: Starting interview of '0x000b57fffe976159'
info  2020-01-19 21:52:25: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000b57fffe976159"}}'
debug 2020-01-19 21:52:31: Device '0x000b57fffe976159' announced itself
error 2020-01-19 21:53:45: Failed to interview '0x000b57fffe976159', device has not successfully been paired
info  2020-01-19 21:53:45: MQTT publish: topic 'zigbee/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x000b57fffe976159"}}'
```

So I tried enabling the herdsman debugging but could not get it to work. Here is my docker-compose.yml, I added DEBUG=* to the environment section but I see no additional logs?
```
version: '3'
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    volumes:
      - ./data:/app/data
      - /run/udev:/run/udev:ro
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
    restart: always
    network_mode: host
    privileged: true
    environment:
      - TZ=Europe/Berlin
      - DEBUG=*
```

Koenkk, 2020-01-20:
@odx check https://github.com/Koenkk/zigbee2mqtt/issues/2779#issuecomment-576357656

@buteo72 you need to set the DEBUG environment variable, I think with

```
setx DEBUG "zigbee-herdsman*"
npm start
```

buteo72, 2020-01-21:
Hi Koen, thanks for helping me out here.
DEBUG herdsman works (don;t forget to start dosbox as administrator)

here you can find my log:  https://pastebin.com/B3NnCf9t

I deleted the database.db and state.json before starting Zigbee2MQTT.

2 strange things I found. 
 - Interview starts with "friendly_name":"0x0000000000000000", after that it start with the right device: 0x00158d0003f0471c
- Interviews ends with 
```
zigbee-herdsman:controller:device Interview - active endpoints request failed for '0x00158d0003f0471c', attempt 2 +10s
  zigbee-herdsman:controller:device Interview - failed for device '0x00158d0003f0471c' with error 'Error: Interview failed because can not get active endpoints ('0x00158d0003f0471c')
  zigbee-herdsman:controller:device     at Device.<anonymous> (C:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\model\device.js:273:23)
  zigbee-herdsman:controller:device     at Generator.throw (<anonymous>)
  zigbee-herdsman:controller:device     at rejected (C:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\model\device.js:6:65)' +7ms
```
```

Koenkk, 2020-01-22:
@buteo72 indeed looks really strange. Can you try replacing the battery of the device just to be sure? Please provide the debug log again when pairing it.

buteo72, 2020-01-22:
Hi Koen, already tried to replace the battery with 3 different batteries. (duracell)
I get also the 0x0000000000000000 when I try to pair an OSRAM Smart+ plug.
Already tried older versions from Zigbee2MQTT and also the new DEV versions. (By renaming the directory, copy the files and `npm install`).

It looks like that Zigbee2MQTT will try to pair 2 devices, one with 0x0000000000000000 and the other with 0x00158d0003f0471c.

Also tryed yesterday evening to pair from scratch. --> remove all devices from `configuration.yaml` and deleted `database.db `and `state.json`

Device is still able to pair with KlikAanKlikUit.

Do you have some more suggestions to test?

Christian

buteo72, 2020-01-22:
sorry, wrong button :-)

Koenkk, 2020-01-23:
@buteo72 I havent seen the pair with `0x0000000000000000 ` behavior before.

- Can you try on a Linux system? Maybe this is a bug which only occurs on Windows
- I see you are living close to me, in case you want you can also give the device to me so I can debug further.

buteo72, 2020-01-23:
> @buteo72 I havent seen the pair with `0x0000000000000000 ` behavior before.
> 
> * Can you try on a Linux system? Maybe this is a bug which only occurs on Windows
Sorry, I don't have a Linux System. (yet)

> * I see you are living close to me, in case you want you can also give the device to me so I can debug further.
My thoughts exactly! No problem, I will write you an PM

buteo72, 2020-01-23:
After replacing the CC2531 with an other CC2531 it works again.
Thanks a lot for your intrests and help.

closed 2020-01-23 by buteo72

Koenkk, 2020-01-24:
@buteo72 thanks, added this to the FAQ (https://www.zigbee2mqtt.io/information/FAQ.html)

cnmarch6, 2020-12-21:
I am having the same problem when tried to pair:
* Xiaomi smart button
* Xiaomi temperature & humidity sensor
* Xiaomi door and window sensor

I was able to pair Xiaomi smart plug without any problem. 

Please help :(

buteo72, 2020-12-21:
Are you working with a CC253X , then reflash the stick. 
Please see also https://www.zigbee2mqtt.io/information/FAQ.html : 
"If device joins with 0x000000000000000 as ieeeAddress (you will see: Starting interview of '0x0000000000000000' in the Zigbee2MQTT log) your CC253X might be broken"

rtrinks, 2022-02-01:
Have you also tested to flash the software again on the CC253X?

m0wlheld, 2022-12-14:
I am seeing this on a CC2652P based coordinator (SONOFF Dongle Plus) with coordinator firmwares 20221102 (devel) and 20220219 (master)

[image]

hellcry37, 2023-03-13:
same issue here

S474N, 2023-07-16:
Same problem with Sonoff MINI Zigbee:
[image]

Router is SLZB-05.