Koenkk/zigbee2mqtt #2762: Zigbee Button (on/off) no reaction "first"...

njordan77, 2020-01-16:
Hi,
right now i'm with 1.8.0 on a Raspi3B+ with Stretch and a June19 USB dongle (CC2531). Works great, various vendors integrated (mostly IKEA stuff, some HUE, XIAOMI). Really in love - only for whatever reason - sometimes during the normal operation but definitely always after a restart of the machine the first button pressed (if its a zigbee button) does not react in anything...so a second press and the whole process works again....after that every first shot works, even after a whole day.....

Any similar experiences, is this normal? hope not, as it makes me now feel unsafe so i press every button now several times to make sure it worked, but this is a rather stupid solution i feel. 

Thanks a lot for any helpful comment.

bartekluk, 2020-01-18:
What are you integrating it with?

njordan77, 2020-01-18:
looking directly in the zigbee2mqtt logs + the related reaction within Openhab 2.5 (based on rules).

bartekluk, 2020-01-19:
Then the most likely culprit is the fact that when openhab starts none of the states are initialized (NULL) hence you press once state gets initialized and then it works normally.

closed 2020-03-26 by stale[bot]